Xilinx/xup_vitis_network_example #3: Ping cannot be established

xuwenquan0, 2020-11-13:
Hello, I have run the basic example in U280 using PYNQ,  where we connected U280 with a 100 G switch on port 11 by QSFP 28 fiber module, and a server  with 100 G ethernet NIC was also connected with the switch on port 9, and the U280 is also connected with the server by PCIe. Then the physical links on U280 and switch are both up, i.e., "Link interface 0 {'cmac_link': True};". However, the Ping cannot established between the server and U280 (server "10.0.0.10" ping U280 "10.0.0.1"), we checked the packet flow, and we found the packet was received by port 9 (server side) and sent out by port 11 (U280 side), but U280 didn't respond. We ensure that the setting of switch is normal, and we test two servers connected with the same switch by 100G NIC, and the Ping can be successfully established. As we cannot grab packets from U280 easily, we have no idea where is the problem. Could you give us some suggestions? Thx.

mariodruiz, 2020-11-13:
Hi @xuwenquan0,

By default every interface gets the default IP address `192.168.0.5`.

In this file https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py you will get helper funtions to check the current IP address of a given interface https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py#L249

You can also run an ARP discovery from the Alveo card and show ARP table https://github.com/Xilinx/xup_vitis_network_example/blob/1a7dad112b81948050bee3576a2df29dc7287d99/Notebooks/vnx_utils.py#L226 and https://github.com/Xilinx/xup_vitis_network_example/blob/1a7dad112b81948050bee3576a2df29dc7287d99/Notebooks/vnx_utils.py#L153

With these functions you will be able to check if the Alveo card is seeing the other devices in your network.

See an example how to use the functions here https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx-benchmark-rtt.ipynb

Mario

xuwenquan0, 2020-11-13:
Thank you for your reply. I've rebooted all devices, and reconfigured the network setting of server and U280 simultaneously (made then in same subnet of 192.168.0.1/24). Then it works. Thank you very much!

mariodruiz, 2020-11-13:
Glad that you could resolve the issue.

You can also update the IP address of the Alveo card using this function

https://github.com/Xilinx/xup_vitis_network_example/blob/1a7dad112b81948050bee3576a2df29dc7287d99/Notebooks/vnx_utils.py#L284

closed 2020-11-13 by mariodruiz