Xilinx/xup_vitis_network_example #5: How to realize none packet loss

WuMinkang, 2020-11-19:
Hello,
Using Vitis to realize network communication is an important task, we are interested in your work and try to recurrent it. In our experiment setup , we connect two alveo U280 cards with a switch . I mean the network topology is like U280 - switch - U280. And we did reproduce the results that the network bandwidth is close to 100Gbps. However , we found that some UDP packets were lost during our experiment.
We repeat the test on the ETH Zurich XACC, the network topology in your email is similar to ours, but this time all the UDP packets were captured, none of them were lost, why?
What is the model of your switch and how do you config it?  And we found that the XRT on the ETH Zurich XACC is different from ours. The XRT version on the ETH Zurich XACC is 2.5.309 and we didn't find a download link on Xilinx official website. Is it a test version? Ours is 2.6.655. Does it matter?

mariodruiz, 2020-11-19:
@WuMinkang,
The packet loss can depend on many things, cable, equipment, traffic in the switch, noise and so on. But, not XRT. UDP is a good option only if there is no packet loss or you do not care if you loss packets. To address that issue I suggest to use TCP/IP. ETH have just released their TCP/IP stack for Vitis. 

https://github.com/fpgasystems/Vitis_with_100Gbps_TCP-IP

The XRT version requirement is only for development, it should not be an issue to deploy the xclbin with an older version. Just to be clear, XRT version does not impact packet loss.

They have a Cisco switch.

WuMinkang, 2020-11-19:
In our experiment we have met another problem.
Our alveo card with QSFP28 cable connected would sometimes breakdown and can not be found as a pcie device using command lspci. After command dmesg it reports that health_check_cb: Card requires pci hot reset and warns that the card is reporting a temperature of -1C. This problem would occur after the cable is connected to the card, but the time between cable connection and the problem is not stable.
[image]
Have you met this before and if so how do you fix it?

mariodruiz, 2020-11-19:
Maybe the temperature is another factor in the packet loss.

I have not seen this issue before. Do you have access to the board? Is is hot? Is the Alveo card properly cooled? Do you have a different cable to try with?

WuMinkang, 2020-11-19:
We do have access to the board. And it is not hot at all. We believe the card is properly cooled because we have another card not connected with a cable and the problem didn't occur on the other card. We tried several QFSP28 cable but the problem keep occurring.Anyway, thank you for your time, we would continue to try to solve this problem.

closed 2021-01-05 by mariodruiz