Xilinx/xup_vitis_network_example #7: python error: signal only works in main thread when pynq.Overlay with dask device

zhaos, 2020-12-20:
Hi, 
I want to recurr the experiment of vnx-benchmark with two freshly installed ubuntu 18.04 machine.
when executing the function "pynq.Overlay(xclbin, device=)", it's ok with device "currentDevice = pynq.Device.devices[0]" in vnx-basic, but failed with device "daskdev_w0 = DaskDevice(client, workers[0])" in vnx-benchmark.
I think there are some issues with my version of dask and pynq. (dask==2020.12.0, pynq==2.5.4 as suggested by the first issue).
below is the err msg from notebook
`---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-26-058320bbebd2> in <module>
      3 
      4 xclbin = '../vnx_benchmark_if0.xclbin'
----> 5 ol_w0 = pynq.Overlay(xclbin, device=daskdev_w0 )
      6 #ol_w0.ip_dict
      7 

~/.local/lib/python3.6/site-packages/pynq/overlay.py in __init__(self, bitfile_name, dtbo, download, ignore_version, device)
    342 
    343         if download:
--> 344             self.download()
    345 
    346         self.__doc__ = _build_docstring(self._ip_map._description,

~/.local/lib/python3.6/site-packages/pynq/overlay.py in download(self, dtbo)
    401                 Clocks.set_pl_clk(i)
    402 
--> 403         super().download(self.parser)
    404         if dtbo:
    405             super().insert_dtbo(dtbo)

~/.local/lib/python3.6/site-packages/pynq/bitstream.py in download(self, parser)
    152 
    153         """
--> 154         self.device.download(self, parser)
    155 
    156     def remove_dtbo(self):

<ipython-input-13-87cd03401595> in download(self, bitstream, parser)
    101         with open(bitstream.bitfile_name, 'rb') as f:
    102             bitstream_data = f.read()
--> 103         self._call_dask(_download, bitstream_data)
    104         super().post_download(bitstream, parser)
    105 

<ipython-input-13-87cd03401595> in _call_dask(self, func, *args)
     77     def _call_dask(self, func, *args):
     78         future = self._dask_client.submit(func, *args, workers=self._worker, pure=False)
---> 79         return future.result()
     80 
     81     def invalidate(self, bo, offset, ptr, size):

~/.local/lib/python3.6/site-packages/distributed/client.py in result(self, timeout)
    222         if self.status == "error":
    223             typ, exc, tb = result
--> 224             raise exc.with_traceback(tb)
    225         elif self.status == "cancelled":
    226             raise result

~/.local/lib/python3.6/site-packages/distributed/utils.py in offload()
   1453 async def offload(fn, *args, **kwargs):
   1454     loop = asyncio.get_event_loop()
-> 1455     return await loop.run_in_executor(_offload_executor, lambda: fn(*args, **kwargs))
   1456 
   1457 

/usr/lib/python3.6/concurrent/futures/thread.py in run()
     54 
     55         try:
---> 56             result = self.fn(*self.args, **self.kwargs)
     57         except BaseException as exc:
     58             self.future.set_exception(exc)

~/.local/lib/python3.6/site-packages/distributed/utils.py in <lambda>()
   1453 async def offload(fn, *args, **kwargs):
   1454     loop = asyncio.get_event_loop()
-> 1455     return await loop.run_in_executor(_offload_executor, lambda: fn(*args, **kwargs))
   1456 
   1457 

~/.local/lib/python3.6/site-packages/distributed/worker.py in _deserialize()
   3291     """ Deserialize task inputs and regularize to func, args, kwargs """
   3292     if function is not None:
-> 3293         function = loads_function(function)
   3294     if args:
   3295         args = pickle.loads(args)

~/.local/lib/python3.6/site-packages/distributed/worker.py in loads_function()
   3282             result = cache_loads[bytes_object]
   3283         except KeyError:
-> 3284             result = pickle.loads(bytes_object)
   3285             cache_loads[bytes_object] = result
   3286         return result

~/.local/lib/python3.6/site-packages/distributed/protocol/pickle.py in loads()
     73             return pickle.loads(x, buffers=buffers)
     74         else:
---> 75             return pickle.loads(x)
     76     except Exception as e:
     77         logger.info("Failed to deserialize %s", x[:10000], exc_info=True)

~/.local/lib/python3.6/site-packages/cloudpickle/cloudpickle.py in subimport()
    560 # hack for __import__ not working as desired
    561 def subimport(name):
--> 562     __import__(name)
    563     return sys.modules[name]
    564 

~/.local/lib/python3.6/site-packages/pynq/__init__.py in <module>
     30 from .mmio import MMIO
     31 from .uio import UioController
---> 32 from .xlnk import Xlnk
     33 from .pl import PL
     34 from .ps import Register

~/.local/lib/python3.6/site-packages/pynq/xlnk.py in <module>
     50     Xlnk().xlnk_reset()
     51     sys.exit(127)
---> 52 signal.signal(signal.SIGSEGV, sig_handler)
     53 
     54 

/usr/lib/python3.6/signal.py in signal()
     45 @_wraps(_signal.signal)
     46 def signal(signalnum, handler):
---> 47     handler = _signal.signal(_enum_to_int(signalnum), _enum_to_int(handler))
     48     return _int_to_enum(handler, Handlers)
     49 

ValueError: signal only works in main thread
`

mariodruiz, 2020-12-21:
Hi @zhaos,

In my environment the dask version is `dask==dask==2.9.2`
What is the result of the 3rd cell in the Notebook (Basic Remote functions)? 

How do you start the dask cluster? Is `pynq` and `dask` in the path for both machines?

zhaos, 2020-12-22:
> Hi @zhaos,
> 
> In my environment the dask version is `dask==dask==2.9.2`
> What is the result of the 3rd cell in the Notebook (Basic Remote functions)?
> 
> How do you start the dask cluster? Are `pynq` and `dask` in the path for both machines?

Thanks for your immediate reply,

In vnx_basic, 3rd cell complete without errors and next step shows physical link is up, so I think pynq is ok.
[image]

and both pynq and dask is in path on the two host machines
[image]

I start dask cluster with following cmds:
1. host1$ dask-scheduler
2. host1$ dask-worker tcp://[ipv4 addr]:8786 --name worker_A
3. host2$ dask-worker tcp://[ipv4 addr]:8786 --name worker_B

I've also tried with dask==2.9.2, but execute cmd " dask-scheduler", just raise some error:

> Traceback (most recent call last):
>  File "/home/tian/.local/bin/dask-scheduler", line 7, in <module>
>    from distributed.cli.dask_scheduler import go
>  File "/home/tian/.local/lib/python3.6/site-packages/distributed/__init__.py", line 3, in <module>
>    from .actor import Actor, ActorFuture
>  File "/home/tian/.local/lib/python3.6/site-packages/distributed/actor.py", line 7, in <module>
>    from .client import Future, default_client
>  File "/home/tian/.local/lib/python3.6/site-packages/distributed/client.py", line 32, in <module>
>    from dask.utils import ensure_dict, format_bytes, funcname, stringify
ImportError: cannot import name 'stringify'

mariodruiz, 2020-12-22:
So, now you are able to download the overlay and check link. What about the rest of the notebook? Are you able to run it?

See all the packages and versions in my environment in the attached
[requirements.txt](https://github.com/Xilinx/xup_vitis_network_example/files/5728755/requirements.txt)
 file

zhaos, 2020-12-24:
Hi, @mariodruiz .
Yes, I can run vnx correctly. 
And I've run the steps in for loop in benchmark manually! 
It's just that the dask doesn't work in my setup, maybe it's a dask issue.

anqianqi1, 2021-01-18:
Hi, @zhaos @mariodru
I have met exactly the same issue running "vnx-benchmark-rtt.ipynb", can you explain how you solve the problem in detail?

I have those issues when using the latest versions.
dask (2021.1.0)
dask-labextension (3.0.0)
distributed (2021.1.0)
pynq (2.5.4)

Meanwhile, I can successfully run Basic remote 3rd cell in the Notebook.

When I downgrade package versions to the following:
dask (2.9.2)
dask-labextension (3.0.0)
distributed (2.9.3)
pynq (2.5.4)

I met this errors running block 3 in "vnx-benchmark-rtt.ipynb":

```
distributed.client - WARNING - Couldn't gather 1 keys, rescheduling {'verify_workers-046b96f2-2cb0-4d69-aff3-d0d769c4df9e': ('tcp://128.197.176.19:35243',)}
distributed.client - WARNING - Couldn't gather 1 keys, rescheduling {'verify_workers-046b96f2-2cb0-4d69-aff3-d0d769c4df9e': ('tcp://128.197.176.19:35243',)}
```

Could you help me with this? Thanks

mariodruiz, 2021-01-18:
How are you creating the dask cluster?

zhaos, 2021-01-19:
@anqianqi1 , sorry that I didn't solve the problem, but just work around it by interacting with pynq manually while not using dask.

anqianqi1, 2021-01-19:
> How are you creating the dask cluster?

I set up the dask cluster the same way as zhaos. One terminal as 'dask-scheduler' and two terminal as 'dask-worker'

anqianqi1, 2021-01-19:
> @anqianqi1 , sorry that I didn't solve the problem, but just work around it by interacting with pynq manually while not using dask.

Oh, so did you program the board with the same benchmark bit file without dask cluster?

mariodruiz, 2021-01-19:
Two questions,
1. Are the dask commands running in the same conda env as `pynq`?
2. Is public key authentication enabled in the servers?

anqianqi1, 2021-01-19:
> Two questions,
> 
> 1. Are the dask commands running in the same conda env as `pynq`?
> 2. Is public key authentication enabled in the servers?
1. we have one server with two boards plugged in and directly connected through QSFP, testing with benchmark rtt point to point. Here is the path:
```
which pynq-server
~/.local/bin/pynq-server
which dask-scheduler
~/.local/bin/dask-scheduler
which dask-worker
~/.local/bin/dask-worker
```

2. I am running the benchmark in one server with two boards plugged in. Everything else is in the server. Do I still need to enable the public key authentication?

anqianqi1, 2021-01-19:
Here is the output:
[image]

mariodruiz, 2021-01-19:
Please edit cell 5 in your notebook (`Download xclbin to workers`) with this,

```python
from vnx_utils import *
import pynq
import ctypes
import tempfile
import numpy as np

xclbin = <xclbin filename replace>
ol_w0 = pynq.Overlay(xclbin, device=pynq.Device.devices[0])
ol_w1 = pynq.Overlay(xclbin, device=pynq.Device.devices[1])
```

You should start from this cell, in your scenario forget everything that is before this cell.

anqianqi1, 2021-01-19:
@mariodruiz  Thank you so much! The issue is fixed!

mariodruiz, 2021-01-19:
Good to know. I will update the documentation to specify this. 
You can do a PR with the notebook for non dask execution. (If you want to)

commit aaf4db51 (2021-01-27): Add better documentation for running the experiments in a multiple Alveo cards in the same system scenario. Fixes #7
closed 2021-01-19 by mariodruiz