Xilinx/xup_vitis_network_example #8: I meet an error, when I use xilinx_u250_xdma_201830_2 platform

Microdent, 2020-12-22:
In your Readme.md, xilinx-u250-gen3x16-xdma-3.1-202020-1 is not supported by Vitis 2020.1. So I changed to the Xilinx_u250_xdma_201830_2 platform.

But I meet the following problem when using the Xilinx_u250_xdma_201830_2 platform:
[image]
[image]

So I try to use Vitis 2020.2 + xilinx-u250-gen3x16-xdma-3.1-202020-1 to run this example, but there are some tcl scripts were made for Vivado 2020.1.
Could you please help with me run this example on Alveo U250 Borad?

mariodruiz, 2020-12-22:
My apologies, the README is not clear. `xilinx-u250-gen3x16-xdma-3.1-202020-1` is supported by the project. The repository as it is does not support the platform.

To generate the xcbin for this platform I suggest to do the following.
1. Run the make command to create xclbin with Vitis 2020.1, this will fail because the platform is only supported in Vitis 2020.2. This is fine, as long as all the xo files are created.
2. Run the make command using Vitis `2020.2` and XRT `2.8.*`, this step should generate the xclbin without a problem.

In order for the platform to be fully supported, the synthesis process of the HLS IP cores of the network layer needs to be ported to Vitis HLS. This process may take some time.

Microdent, 2020-12-22:
@mariodruiz , the xclbin is generated successfully, thank you!

closed 2020-12-22 by Microdent

lycosdae, 2021-02-02:
Dear Sir,
I meet a similar error with u200. 
[image]
Will there be any changes to make this example support u200?

mariodruiz, 2021-02-02:
Alveo U200 is not supported because the current Alveo U200 shell does not exposes GT pins to the user logic. Until a new shell arrives, I am afraid the U200 is out of the picture for VNx. 

This table described what is supported https://github.com/Xilinx/xup_vitis_network_example#support