Xilinx/xup_vitis_network_example #10: Question about MAC address

heiyiren1999, 2021-01-24:
In the vnx-basic notebook, it mentions that the Alveo MAC address is 00:0A:35:02:9D:E5 by default but I don't find any function that can change the default value. Is there any api for changing MAC address? I think this function is important to my design. Thanks for any help!

mariodruiz, 2021-01-24:
Hi @heiyiren1999, the network layer where the MAC is defined is bound to this class https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py#L148

You can query the current network info using this function https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py#L297 

Let's consider that you have loaded the result of `pynq.Overlay()` is assigned to `ol`, you can query and update the MAC just by

```python
new_mac = <assign value>
current_mac = ol.networklayer_0.register_map.mac_address
ol.networklayer_0.register_map.mac_address = new_mac
```

This is a really nice functionality of `pynq`, you can go a step further and show the current value of all registers by just:

```python
ol.networklayer_0.register_map
```

Notice that when you update the IP address using [.updateIPAddress](https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py#L318)  the least significant octet of the MAC address is updated with the least significant octet of the IP address.

heiyiren1999, 2021-01-25:
Thanks very much! It's very convenient. ^_^
By the way, I wonder if there is an C/C++ version of this project? I'm more familiar with C/C++ host code in Vitis. And I think it should have better performance.

mariodruiz, 2021-01-25:
**There is no official support for OpenCL host code for this project**. One of our partners wrote the OpenCL code for configuring VNx, you can find it here https://github.com/inaccel/runtime/blob/Xilinx-MP/src/inaccel/runtime.c#L607-L760

Note that the host code is not for any of the examples in this repo. If you go in this direction is very likely that we won't be able to support you.

Mairo

closed 2021-01-25 by mariodruiz

heiyiren1999, 2021-01-25:
OK, fine. However, I still have some questions. I think the networklayer is a free running kernel because its hwControlProtocol is "ap_ctrl_none" ("/xup_vitis_network_example/NetLayers/kernel.xml"). Then I check the chapter of "free running kernel" in the ug1393 and find that free running kernel should be free of any control handshake (p100). But in this project the user has to set ti (their ip), tp (their port), mp (my port) and v (valid) for the networklayer kernel. So how does this free running kernel receive the control signals?

mariodruiz, 2021-01-25:
It has an AXI4-Lite interface. The _free of control handshake_ in the documentation refers to the control registers offset from 0x0 to 0xC. In any case, the network layer is an RTL kernel, even though most of the logic is synthesized with HLS. https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/devrtlkernel.html?hl=rtl%2Ckernel#xvi1504034323705

heiyiren1999, 2021-01-25:
It's very nice of you to tell me that and I find I become more familiar with kernel models. And I'll try to rewrite the host code in C/C++. Thanks very much! Have a nice day!