Xilinx/xup_vitis_network_example #9: 'cmac' object has no attribute '_call_struct'

heiyiren1999, 2021-01-24:
It's an amazing project and I want to implement it on my server. I follow your notebook to learn how to use it. However, when I run the **print("Link interface 0 {}; link interface 1 {}".format(ol.cmac_0.linkStatus(),ol.cmac_1.linkStatus()))**, it returned me an error that **'cmac' object has no attribute '_call_struct'**. The version of PYNQ on my server is 2.6.0, the operating system is 18.04 and the XRT is 202010.2.7.766_18.04-amd64-xrt (I think this error should not be related with the version of XRT). Now I don't know how to deal with this error. Would you like to give some advice? Thanks for any help!

heiyiren1999, 2021-01-24:
Oh, OK. I find that this error has been solved in one of the closed issues. 😂

closed 2021-01-24 by heiyiren1999

mariodruiz, 2021-01-24:
Yes, this is a known issue with the `pynq 2.6.0` version. For everyone's benefit the issue is solved here #6

commit 3dcbf9da (2021-02-06): Add code to be able to work on pynq 2.6.0. This fixes #9 completely