Xilinx/xup_vitis_network_example #11: What does the destination ID mean in the mm2s kernel?

heiyiren1999, 2021-01-27:
Sorry to bother again but I'm confused about the definition of destination ID. ( https://github.com/Xilinx/xup_vitis_network_example/blob/4955fa3c957b96254405255499a155d47b50bc82/Basic_kernels/src/krnl_mm2s.cpp#L54 ) 
Does it mean the index of the destination ip address in the socket table ? However, when I exchange the position of socket[0] and socket[1] in the table in the vnx-basic.ipynb and set dest 0 (e.g. mm2s_wh = mm2s.start(mm2s_buf,size, 0), instead of mm2s_wh = mm2s.start(mm2s_buf,size, 1)), the U280 cannot reply to NIC correctly.

mariodruiz, 2021-01-27:
Yes this is correct, the dest argument is the ID of the UDP table. 

The first part of your changes are correct but you also need to edit cell 9 and 12 to open the software port appropriately.

heiyiren1999, 2021-01-28:
I've changed socket[1] to socket[0] in the cell 9 and 12 but the U280 cannot reply to NIC correctly. When socket[0] is used, the project stalls and I find that the the number of packages sent to FPGA is less than 100 (so the s2mm kernel stalls). When I use socket[1] to send and receive packages, the project works well. 
Besides, when I use OpenCl to run the mm2s kernel, no matter what the value of dest I set, the ARP package always ask for 255.255.255.1 ip address. But I'm sure that I've written the correct ip address into socket table and s2mm kernel always works well while mm2s kernel has something wrong.

heiyiren1999, 2021-01-28:
I reorganized my problem here.
For python3, the vnx-basic.ipynb runs well but when I rewrite the code in test.py file instead of .ipynb, there is something wrong and the problems I find are all in test.py file.

First, I have to comment from Line 29 to 89 and run the test.py for the fisrt time, then run the whole file, otherwise it will stalls in the "use socket to send data from NIC to u280" stage (Line 29 - 63). I use wireshark to catch packages when it stalls and find thing below.
[image]

Second,  when it runs into "use socket to send data from u280 to NIC" stage. The processor stalls as well and the packages sent by u280 are as below.
[image]

It's really strange.

What's more, I rewrite the host code in OpenCL (test.cpp) and find that no matter in which position I store the socket information in the socket table, the ARP packages ask for "255.255.255.1" ip address instead of the NIC ip address ("10.0.0.6") when I run the mm2s kernel. However, I read and check all items in the socket table and find "10.0.0.6" is in the right position and there is no item has "255.255.255.1" ip address. I want to check the implementation about the socket table but the project is too large. Would you like to tell me where is the implementation about the the socket table 

[test.zip](https://github.com/Xilinx/xup_vitis_network_example/files/5884897/test.zip)

?

heiyiren1999, 2021-01-28:
I find the reason why the ARP packages ask for "255.255.255.1" ip address. It's because that I set the wrong gateway address.  The gateway address should be "10.0.0.1" but I set it "255.255.255.1". However, the NIC and u280 are connected directly so in fact their connection doesn't need gateway address. So why there are ARP packages asking for gateway address?

mariodruiz, 2021-01-28:
When you update the IP address you need to run ARP discovery, otherwise the first packet is lost.

Just after line 21, add this `ol.networklayer_0.arpDiscovery()`. In the companion notebook this is achieved by pinging from the CPU. This should solve your problem.

Regarding OpenCL code, as I mentioned in the other issue, I can't support you. My advice is to set up everything like in the pynq code

heiyiren1999, 2021-01-29:
Thanks, I'll try it !

closed 2021-01-29 by heiyiren1999