Xilinx/xup_vitis_network_example #12: Modifying Basic kernel into two board directly connected

anqianqi1, 2021-01-29:
Hi, I am trying to modify the Basic design into two alveo board directly connected through QSFP port. However, the data seems not delivering correctly, the receiver board is not getting data. 

The code is attached. The code tries to transfer an array twice. 
In the first transfer, line 75, here is the output I test in Jupyter(data is not matching):
[image]

In the second transfer, line 91, the data is not correctly delivered and also has this error:
[image]

[basic_point_to_point.zip](https://github.com/Xilinx/xup_vitis_network_example/files/5891683/basic_point_to_point.zip)

mariodruiz, 2021-01-29:
At first sight everything looks ok in the code. 

The runtime error you are getting may be the key. 

Are you getting link?

Can you provide the output of this after the first data transfer?

```python
ol_w0.networklayer_1.register_map
ol_w1.networklayer_1.register_map
```

anqianqi1, 2021-01-29:
Sorry! And here is the output of link status:
```
print("Link board 0 {}; link board 1 {}".format(ol_w0.cmac_1.linkStatus(),ol_w1.cmac_1.linkStatus()))
```
```
Link board 0 {'cmac_link': True}; link board 1 {'cmac_link': True}
```

Here is the output of 'networklayer_1.register_map':
```
ol_w0.networklayer_1.register_map
RegisterMap {
  CTRL = Register(AP_START=1, AP_DONE=1, AP_IDLE=1, AP_READY=1, AUTO_RESTART=1),
  mac_address = Register(value=43839036677),
  ip_address = Register(value=3232235525),
  gateway = Register(value=3232235521),
  ip_mask = Register(value=4294967040),
  constant_r = Register(value=1584862309),
  eth_inbound_cycles = Register(value=19033678779782),
  eth_inbound_bytes = Register(value=306060),
  eth_inbound_packets = Register(value=5101),
  pkth_out_cycles = Register(value=19033678779782),
  pkth_out_bytes = Register(value=306060),
  pkth_out_packets = Register(value=5101),
  arp_in_cycles = Register(value=19033678779782),
  arp_in_bytes = Register(value=306060),
  arp_in_packets = Register(value=5101),
  arp_out_cycles = Register(value=19033687755341),
  arp_out_bytes = Register(value=306060),
  arp_out_packets = Register(value=5101),
  icmp_in_cycles = Register(value=0),
  icmp_in_bytes = Register(value=0),
  icmp_in_packets = Register(value=0),
  icmp_out_cycles = Register(value=0),
  icmp_out_bytes = Register(value=0),
  icmp_out_packets = Register(value=0),
  ethhi_out_cycles = Register(value=0),
  ethhi_out_bytes = Register(value=0),
  ethhi_out_packets = Register(value=0),
  eth_outbound_cycles = Register(value=19033687755341),
  eth_outbound_bytes = Register(value=306060),
  eth_outbound_packets = Register(value=5101),
  udp_in_cycles = Register(value=0),
  udp_in_bytes = Register(value=0),
  udp_in_packets = Register(value=0),
  udp_out_cycles = Register(value=0),
  udp_out_bytes = Register(value=0),
  udp_out_packets = Register(value=0),
  udp_app_in_cycles = Register(value=0),
  udp_app_in_bytes = Register(value=0),
  udp_app_in_packets = Register(value=0),
  udp_app_out_cycles = Register(value=0),
  udp_app_out_bytes = Register(value=0),
  udp_app_out_packets = Register(value=0),
  debug_reset_counters = Register(value=0),
  debug_clock_freq = Register(value=250000000),
  debug_number_ports = Register(value=12),
  debug_settings = Register(value=11381997),
  udp_offset = Register(value=0),
  arp_mac_addr_offset = Register(value=0),
  arp_ip_addr_offset = Register(value=0),
  arp_valid_offset = Register(value=0),
  arp_discovery = Register(value=0)
}
```

```
ol_w1.networklayer_1.register_map
RegisterMap {
  CTRL = Register(AP_START=1, AP_DONE=1, AP_IDLE=1, AP_READY=1, AUTO_RESTART=1),
  mac_address = Register(value=43839036682),
  ip_address = Register(value=3232235530),
  gateway = Register(value=3232235521),
  ip_mask = Register(value=4294967040),
  constant_r = Register(value=1584862309),
  eth_inbound_cycles = Register(value=96724496090360),
  eth_inbound_bytes = Register(value=597000),
  eth_inbound_packets = Register(value=9950),
  pkth_out_cycles = Register(value=96724496090360),
  pkth_out_bytes = Register(value=597000),
  pkth_out_packets = Register(value=9950),
  arp_in_cycles = Register(value=96724496090360),
  arp_in_bytes = Register(value=597000),
  arp_in_packets = Register(value=9950),
  arp_out_cycles = Register(value=96724450971303),
  arp_out_bytes = Register(value=612300),
  arp_out_packets = Register(value=10205),
  icmp_in_cycles = Register(value=0),
  icmp_in_bytes = Register(value=0),
  icmp_in_packets = Register(value=0),
  icmp_out_cycles = Register(value=0),
  icmp_out_bytes = Register(value=0),
  icmp_out_packets = Register(value=0),
  ethhi_out_cycles = Register(value=0),
  ethhi_out_bytes = Register(value=0),
  ethhi_out_packets = Register(value=0),
  eth_outbound_cycles = Register(value=96724450971303),
  eth_outbound_bytes = Register(value=612300),
  eth_outbound_packets = Register(value=10205),
  udp_in_cycles = Register(value=0),
  udp_in_bytes = Register(value=0),
  udp_in_packets = Register(value=0),
  udp_out_cycles = Register(value=0),
  udp_out_bytes = Register(value=0),
  udp_out_packets = Register(value=0),
  udp_app_in_cycles = Register(value=0),
  udp_app_in_bytes = Register(value=0),
  udp_app_in_packets = Register(value=0),
  udp_app_out_cycles = Register(value=0),
  udp_app_out_bytes = Register(value=0),
  udp_app_out_packets = Register(value=0),
  debug_reset_counters = Register(value=0),
  debug_clock_freq = Register(value=250000000),
  debug_number_ports = Register(value=12),
  debug_settings = Register(value=11381997),
  udp_offset = Register(value=0),
  arp_mac_addr_offset = Register(value=0),
  arp_ip_addr_offset = Register(value=0),
  arp_valid_offset = Register(value=0),
  arp_discovery = Register(value=0)
}
```

mariodruiz, 2021-01-29:
Did you capture the registers after sending some data?

It is extrange that these register are empty 

```console
udp_app_in_cycles = Register(value=0),
udp_app_in_bytes = Register(value=0),
udp_app_in_packets = Register(value=0),
udp_app_out_cycles = Register(value=0),
udp_app_out_bytes = Register(value=0),
udp_app_out_packets = Register(value=0),
```

These should show some activity. I need to investigate this a bit deeper

anqianqi1, 2021-01-29:
Thanks!
And I capture the register right after sending data. Here is the code where I put `register_map`:
For sender:
```
mm2s_buf_ol0.sync_to_device()
mm2s_wh_ol0 = mm2s_ol0.start(mm2s_buf_ol0,size, 0)
ol_w0.networklayer_1.register_map
```
For receiver:
```
s2mm_wh_ol1.wait()
print(ol_w1.networklayer_1.register_map)
s2mm_buf_ol1.sync_from_device()
```

mariodruiz, 2021-01-30:
I found the issue, if you look how the datamover works

https://github.com/Xilinx/xup_vitis_network_example/blob/master/Basic_kernels/src/krnl_mm2s.cpp#L69

If `size < 64` hardware never starts moving from the memory, this is the reason why the embedded probes are empty. To avoid this issue I enhance the vnx_utils.py to verify that when the accelerator is started all the parameters are valid. https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py#L418-L453

Please pull changes and update size in your code

closed 2021-01-30 by mariodruiz

anqianqi1, 2021-02-01:
Thank you so much! the new jupyter notebook example works good!

commit e6b2f4db (2021-01-30): Enhancement, fixes #12. Add driver to verify that k2mm and s2mm are not configured with a size smaller than 64-Byte