Xilinx/xup_vitis_network_example #13: Is there any difference among traffic_generator_1_0, traffic_generator_1_1, traffic_generator_1_2, traffic_generator_1_3?

heiyiren1999, 2021-01-31:
I find that in vnx-benchmark-rtt.ipynb, you use traffic_generator_1_0 for 'LOOPBACK' and traffic_generator_1_2 for 'LATENCY'. But in vnx-benchmark-throughput.ipynb, you use traffic_generator_1_1 for 'CONSUMER' and traffic_generator_1_3 for 'PRODUCER'. So I wonder if there is any difference among these four traffic_generators. Does each of them match to only one mode?

heiyiren1999, 2021-01-31:
What's more, I also want to know what's the meaning of "The latency of each packet is 32-bit". Why the unit of latency is bit instead of s/ms/ns? Thanks for any help!
[image]

mariodruiz, 2021-01-31:
The traffic generator kernels are the same, see here that they are replicated https://github.com/Xilinx/xup_vitis_network_example/blob/master/config_files/connectivity_benchmark_if3.ini#L6-L7

These kernels are connected to a switch, as shown here

[image]

Each of them can send payload to any entry in the table, i.e. dest = 0 to 15. However, in the reception side the switch will route the payload to the corresponding kernel, so in this case

| dest | kernel                |
|------|-----------------------|
| 0    | traffic_generator_*_0 |
| 1    | traffic_generator_*_1 |
| 2    | traffic_generator_*_2 |
| 3    | traffic_generator_*_3 |

The selection of these kernels in the notebooks is arbitrary and aims at demonstrating the capabilities of the design.

Regarding the latency, I need to update the documentation to make it clear. In reality, with this design can only measure round trip time. The collector will perform (rx time - tx time) this a 32-bit result. This RTT is measure in clock cycles. The notebook transform cycles into microseconds in the cell *Compute some statistics on the results*. This transformation cannot be implemented on hardware easily as the frequency can change from implementation to implementation, and the design frequency can be updated at runtime as well. So, the easiest solution is to make the transformation in sw

Hope this clarifies your questions

heiyiren1999, 2021-02-01:
Thanks very much! And besides, do you know how to write a time cycle counter in HLS in Vitis? I want to write a simpler test-bench by myself (just for learning HLS and Vitis).

heiyiren1999, 2021-02-01:
Well...I also want to know how the time cycle counter works. Does the counter in the traffic generator kernel not consider the time that the kernel spends on waiting data? If the data doesn't arrive consecutively, does the counter pause when there is no data in the fifo?

mariodruiz, 2021-02-01:
The cycle counter in the traffic generator is always running, never pauses! Tx payload is provided with a timestamp and an ID number. Upon reception the payload is timestamped again and a summary of this information is sent to the collector. 

The collector computes RTT as explained above, the collector also counts for consecutive packets which can be retrieved later, if there is packet loss this counter is different from the total number of packets sent.

heiyiren1999, 2021-02-01:
Oh...so, does timestamp mean the index of time cycle in which the data is transported or received?

mariodruiz, 2021-02-01:
It's not the index, the tx packets include the whole 40-bit time counter. The counter is not synchronized with anything, for that reason we can only measure RTT

To measure RTT this is the piece of code generating tx payload https://github.com/Xilinx/xup_vitis_network_example/blob/master/Benchmark_kernel/src/segment_generator.v#L242-L244

and his piece of coce generates the summary for rx payload https://github.com/Xilinx/xup_vitis_network_example/blob/master/Benchmark_kernel/src/segment_generator.v#L341-L342

heiyiren1999, 2021-02-01:
Thanks, I got your idea. But how the timestamp works with throughput test-bench? Is the rx_time which is got from computeThroughputApp function calculated out by subtraction between the timestamp of the first package and the timestamp of the last package?
I want to check it so I read the traffic_generator.v. I find that bw_producer_i and bw_consumer_i modules are used in the traffic_generator.v. However, I search them in this repository but find there are no such modules.

mariodruiz, 2021-02-01:
RTT is computed (in clock cycles) for every single packet only when the kernels is configured to measure LATENCY.

The collector puts that info in a buffer in the global memory and the pynq code grabs that buffer, converts it to microsecond and compute statistics, such as average, max, min and std deviation. The code in the notebook is very clear about that.

The debug modules come from here https://github.com/Xilinx/xup_vitis_network_example/tree/master/NetLayers/src and are only used to count bytes, packets and channel active time between app and network layer. Nothing to do with latency.

closed 2021-02-01 by mariodruiz

heiyiren1999, 2021-02-01:
Thanks! And sorry to bother you so much. I seemed to mix the implementation on between calculating rx_time/tx_time in throughput test-bench and latency (between rx and tx payload) in rtt test-bench. But now I understand the differences. Thanks very much!

mariodruiz, 2021-02-01:
Remember that each mode has an unique usage. This image show the different modes:

[image]

One Alveo cards configures one mode an the other card configures another. The modes work in conjunction in a diagonal way 
`LATENCY & LOOPBACK` or `PRODUCER & CONSUMER`