Xilinx/xup_vitis_network_example #14: Throughput test-bench with only one u280

heiyiren1999, 2021-02-02:
Hello...me again. I have another some questions about test-bench implementation. 
I can only use one u280 recently but I still have to do the throughput test-bench. So I consider two methods. The first one is that sending packages by nic on the server instead of another u280 of "producer" mode. For this method, I still want to implement "consumer" mode on the only one u280 but I'm not clear about the construction of my packages. Is there any special bit in packages for "consumer" mode to work correctly?
The second method is to implement both "producer" and "consumer" on the same u280. However, when I link two interfaces to the switch, the link check for interface 1 is always false. I've exchanged the network lines and also implemented the benchmark_if3.xclbin on another u280, but the results are the same. So does only one interface work at the same time？

mariodruiz, 2021-02-02:
Hi, 
Regarding the first question you can send standard UDP packets, only to measure latency there is a custom protocol in the payload.

Both interfaces should work at the same time, but there was a bug that prevented interface 1 to get link, I addressed it 6 days ago https://github.com/Xilinx/xup_vitis_network_example/commit/b4f70155939de1c3cb58524bc5670591b308e0e7

Pull latest version and rebuild xclbin, that should solve the problem.

heiyiren1999, 2021-02-02:
Interface 1 can be linked now! Thanks very much!

closed 2021-02-02 by heiyiren1999