Xilinx/xup_vitis_network_example #15: Is traffic_generator a free running kernel?

heiyiren1999, 2021-02-02:
Hi * n times. Another some questions come to my mind.

First, because in all benchmark notebooks I find that traffic_generators are not started by .start() function or something like that, I think it should be a free-running kernel. However, when I check the kernel.xml (xup_vitis_network_example/Benchmark_kernel/kernel.xml), I find that the hwControlProtocol of the traffic_generator kernel is "ap_ctrl_hs", which means that it should not be a free-running kernel. 
So is the traffic_generator kernel a free-running kernel?

Second, in the computeThroughputApp function (vnx_utils.py), bytes, cycles and pkts are gotten from registers. However, I think this operation, namely reading from registers, is happened when the command self.register_map.xxx is executed. So how does it guarantee that the value which is read out from the register is the final correct result instead of an intermediate result during packages processing?

Third, because I implement both "latency" and "loopback" mode on the only one u280, I don't need to use Dask. But I don't know how to change the 13th cell in the vnx-benchmark-rtt-switch.ipynb to suit my implementation. I just delete it and find all the results (mean, std_dev, mode, max, min) are zeros. 

Appreciate any help!

mariodruiz, 2021-02-02:
Hi,

1. traffic generator is not a free running kernel. I am not using the .start() because many of the parameters in the signature are just for debug. I started the kernel manually, in some part of the notebook you should see `ol_w0_tg.register_map.CTRL.AP_START = 1` this is starting the kernel

1. There is no guarantee that the values are correct, but I waited for all the packets to be sent and an extra 800 ms to retrieve data. That should be enough for the data to be ready

```python
while int(ol_w0_tg.register_map.out_traffic_packets) != pkt:
    time.sleep(0.8)
```

1. You do not have to delete cell 13, replace `.invalidate` with `.sync_from_device `

closed 2021-02-02 by mariodruiz

heiyiren1999, 2021-02-03:
I got it. Thanks so much!

heiyiren1999, 2021-02-03:
Emmm... sorry, I have one more question. I'm puzzled about the aim of Benchmark Throughput experiment. It seems to test the bandwidth of "consumer" mode. However, in my opinion, to do a throughput test-bench on the "consumer" mode, I need to make sure that the throughput of the "producer"  mode wouldn't be the throughput bottleneck in the test-bench. But when I also print the tx_thr to check the throughput of "producer" mode, I find it is very close to the throughput of the "consumer"  mode. So what's the aim of Benchmark Throughput experiment? Does it really aim to test the throughput of the "consumer"  mode?

mariodruiz, 2021-02-03:
Both designs aim at showing the capabilities of providing networking in a Vitis shell. The benchmark notebooks shows that the design is able to reach theoretical maximum throughput at application and Ethernet level.

The traffic generator in producer and consumer mode is capable of larger throughput, but the limiting factor is the network which limits the communication to a 100 Gigabit Ethernet.

So the aim is to show that an user kernel can transmit and receive data from the network at full speed.

heiyiren1999, 2021-02-03:
OK! Appreciate your help very much!