Xilinx/xup_vitis_network_example #16: Steps to build for `xilinx_u250_gen3x16_xdma_3_1_202020_1` platform

mariodruiz, 2021-02-25:
The `xilinx_u250_gen3x16_xdma_3_1_202020_1` platform is only supported by Vitis 2020.2, however the network layer relies on Vivado HLS, which is deprecated in Vitis 2020.2. Therefore, to successfully build for this platform, you will need to use both Vitis 2020.1 and Vitis 2020.2.

These are the steps  I used to successfully reproduce the examples for this platform. Use XRT `2.8.743`

## Get a fresh copy of the repository 

Use this copy only to build for this platform. The goal is to avoid intermediate states

```sh
git clone https://github.com/Xilinx/xup_vitis_network_example.git --recursive vnx_u250
```

## Compile network layer

1. Source Vitis 2020.1 tools
2. `cd vnx_u250/NetLayers/`
3. Run make

```sh
make all DEVICE=<fullpath>/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm
```

## Generate xclbin

After the network layer is compile, we can use Vitis 2020.2 to generate the xclbin

1. Source Vitis 2020.2 tools
2. Run `make` to build the example as described [here](https://github.com/Xilinx/xup_vitis_network_example#generate-xclbin)

HFTrader, 2021-03-14:
I can confirm this works for xilinx_u50_gen3x16_xdma_201920_3

6357o31337, 2021-07-16:
When I compile network layer (make all DEVICE=<fullpath>/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm),
there was error
/bin/sh: 1: Syntax error: word unexpected (expecting ")")
Makefile:143: recipe for target 'check-design' failed
make: *** [check-design] Error 2

6357o31337, 2021-07-16:
When I 'Generate xclbin'(make all DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm INTERFACE=0 DESIGN=benchmark),

there are errors
/bin/sh: 1: Syntax error: word unexpected (expecting ")")
Makefile:143: recipe for target 'check-design' failed
make: *** [check-design] Error 2

How can I solve?

mariodruiz, 2021-07-16:
@6357o31337, this seems to be an issue with the shell you're using.

If you run `ps -p $$` you will the shell you're using. For your info, I am using `bash`

```sh
bash --version
```
```
GNU bash, version 4.4.20(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

```

6357o31337, 2021-07-21:
@mariodruiz , thank you.
Removing the 'check-design' and 'check-interface' targets helped me.

But when I try to 'Generate xclbin'(make all DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm INTERFACE=0 DESIGN=basic) again,
there is another error. (below imagaes)

How can I solve?

Thank you.

[image]

...

[image]

[image]

6357o31337, 2021-07-22:
For more information,

[image]

Thank you.

HFTrader, 2021-08-01:
I bet this is a memory issue. You got to add something like below in $HOME/.Xilinx/Vivado/Vivado_init.tcl

`set_param general.maxThreads 18`

mariodruiz, 2021-08-04:
@6357o31337 if you still have the problem please:
- Open another issue and use the VNx issues template to provide the required information. 
 - Use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

anipeg, 2021-08-04:
I got the same issue in Ubuntu.
That's because the Makefile default SHELL is /bin/sh.
I add "SHELL=/bin/bash" to Makefile first line and it work.

closed 2021-09-17 by mariodruiz