Xilinx/xup_vitis_network_example #17: What's the function of arpDiscovery()?

heiyiren1999, 2021-03-04:
Hello, how are you?
In https://github.com/Xilinx/xup_vitis_network_example/issues/11#issuecomment-768897746, you mentioned that the function of ol.networklayer_0.arpDiscovery() is achieved by pinging from the CPU. However, in my opinion, these two operations are done by different objects separately. The ol.networklayer_0.arpDiscovery() is done by FPGA and the pinging is from the CPU. So I don't understand why the function of the former can be achieved by the latter. And I also want to the function of arpDiscovery() in detail. Would you like to explain it? Thanks for any help!

heiyiren1999, 2021-03-04:
Today I find my problem as mentioned in #11(comment) didn't solved by adding ol.networklayer_0.arpDiscovery() after Line21. I reorganize the problem as below.

The problem is about sending data from U280 to NIC. I use **xbutil reset** command to reset the U280 board and then use **python3 test.py** to run the host code. When I run the test.py **the fisrt time** after reset the U280, the host process is stuck at Line59, which means it doesn't receive the data which should be sent from U280 to NIC. However, when I use Ctrl+C to quit the stuck host process and then use python3 test.py to run the host code again without reset the U280, the host can receive data correctly and print "Kernel sending data to the network and the host getting data from network was a: SUCCESS!. Total data received 140,800 bytes from ('10.0.0.233', 54321)"

It's very strange. How should I deal with this problem?

[test.zip](https://github.com/Xilinx/xup_vitis_network_example/files/6082031/test.zip)

mariodruiz, 2021-03-04:
Hi @heiyiren1999,

Let me put some context to your question and I think, it will be easier to answer it.

1. The ARP table translates the IP addresses to MAC addresses, this table is empty the first time you program the FPGA.
2. When sending a packet, the UDP module queries the ARP table to get corresponding MAC for a given IP
a. If the table is empty or the entry does not exist the UDP will drop the payload and the ARP module will issue a ARP discovery with such IP
b. If the entry exists the packet will be sent properly
3. To populate the ARP table, you could use two methods. 
- Ping from the CPU to FPGA IP, this will create the entry in the table
- run `.arpDiscovery()`, in this case the ARP module will issue an ARP request to all the IP addresses on its subnetwork. The entries in the table will be populate with those that replied

Now to answer your question, your problem seems to be that the table is not being populated properly the first time, so 2.a. happens, and you code keeps on waiting because at least one packet is not sent. The second time you run the code, the table is populated properly and no packet is discarded.

I don't have a clear solution to your problem. The ARP table should be populated after `ol.networklayer_0.arpDiscovery()` but does not seems to be the case. 

Debug steps: 
1. Read the ARP table after the `.arpDiscovery()` with `ol.networklayer_0.readARPTable()` and check that an entry exists for the sw IP.
2. Add a time out to the for loop to verify how much data you get the first time.  

If this is not the case, try to find out why. Possible cause: IP is not properly configure in the CPU the first time 

Hope this answer helps you. 

Mario

heiyiren1999, 2021-03-05:
Thanks for your answer!

I add ol.networklayer_0.readARPTable() after .arpDiscovery() in the test.py and it prints nothing. However, when I add time.sleep(60) before .arpDiscovery(), then ol.networklayer_0.readARPTable() can print all MAC addresses and IP addresses.

So I think the reason of my problem may be that the ARP Table needs some time to initialize itself. And this time may be longer than the time which the host takes to run to the line where .arpDiscovery() is.

mariodruiz, 2021-03-05:
Yes, it takes some time for the table to be populate, which mainly depends on the RTT. 

You can split your script into two parts, initialization and execution. Add around 2 seconds between each part and you should be fine.

Mario

closed 2021-03-05 by mariodruiz