Xilinx/xup_vitis_network_example #19: How to get more information like source address and destination address of a package?

heiyiren1999, 2021-03-06:
Hi. Me again. I meet some problems about the design of the project.

I have several clients sending packages to the only one FPGA, and the FPGA is considered to answer each client correctly. However, I find that the NetLayer kernel only sends the content of payload to the user kernel, which means I cannot get the source address of the package from its header. I don't expect to add address information in payload because it'll cause the payload seems overstaffed. So is there any method for user kernel to get more information of the package like source address and destination address of a package?

Besides, I find that the user kernel has to assign the index of SocketTable to pkt.dest to tell the NetLayer the destination of the package. In the krnl_mm2s (Basic_kernels example) , the index is set by host. However, now I want that the user kernel can realize which value the pkt.dest should be instead of getting the value from the host. I think that comparing each item in the SocketTable with the destination address of the package in a loop can solve this problem. However, the SocketTable is not in the user kernel so I don't know how to compare them. What should I do to solve this problem? Or is there any clever way to make the NetLayer realize the destination of the package?

Thanks for any help!

mariodruiz, 2021-03-06:
Hi @heiyiren1999,

Apart from the payload the network layer provides the index of the socket table in tdest. There are multiple ways to get this functionality.

1. Duplicate the socket table in your application and use the tdest to index. 
2. Implement multiple user applications per each connection, and use a switch to route them, like in the benchmark design
3. Modify the UDP module to provide the metadata, like IP address and destination and source port in the tuser signal. The plan is to add this in the future. 

In terms of complexity, 1 should be the easiest and you only need to edit your user kernel. 2, has an intermediate complexity as you need to add the switch in the middle and duplicate the application. 3, is more complex as you have to edit the network layer and the UDP module to achieve this. 

If you go for option 3, we would love your contribution as this is a really nice feature.

heiyiren1999, 2021-03-07:
It's very nice of you to tell me that and I'll use the method 1. to solve my problem recently. 

I'm also interested in method 3. However, I'm not much familiar with RTL kernels and with how to write configuration files. I have some questions about the NetLayers kernel. 
1. Under xup_vitis_network_example/NetLayers/, what's the relation between 100G-fpga-network-stack-core @ e4a4e96 file and src file?
2. What's the function of bd_network_layer.tcl, gen_xo.tcl and package_kernel.tcl? Where can I learn how to write .tcl files?

I want to improve myself so I try to understand each part of this project. And thanks for any help you would like to offer!

mariodruiz, 2021-03-07:
Hi,

1. [100G-fpga-network-stack-core ](https://github.com/hpcn-uam/100G-fpga-network-stack-core) provides the core functionality of the network later with HLS code: ARP, ICMP and UDP. The src folder holds HDL that is used in the network layer for debug purposes 
2. bd_network_layer.tcl is the tcl file to glue all the IP together in the network layer using Vivado IPI. gen_xo.tcl creates the Xilinx object (.xo) from the IPI, which is the file used in Vitis

[Vivado Design Suite Tcl Command Reference Guide](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug835-vivado-tcl-commands.pdf)

If you are not experienced with Vivado flow, my recommendation is not to try to use RTL kernels. But, you can find more info about RTL kernels [here](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/devrtlkernel.html)

heiyiren1999, 2021-03-08:
OK. Thanks for your suggestions very much!

closed 2021-03-08 by heiyiren1999

mariodruiz, 2021-03-10:
@heiyiren1999 in the last commit I added support for point 3 mentioned above. 

The metadata structure is described here https://github.com/Xilinx/xup_vitis_network_example/tree/master/NetLayers#network-layer-to-user-kernel-m_axis_nl2sk 

Note that even though I added support for it, there is no plan to include an application to use this functionality nor to modify the current ones to use it.

heiyiren1999, 2021-03-12:
Woooow! You're so nice! Thanks very much!