Xilinx/xup_vitis_network_example #20: question about getting this to show up as an eno device

gtrchchn, 2021-03-07:
First off, I wanted to say thank you for posting such a fantastic example.  This has really helped me understand the interactions between various types of kernels.  

I had a question about the [basic example](https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx-basic.ipynb) notebook you posted.  Specifically where the call to `!ifconfig enp115s0f1`.  I've built the basic design for if0 and have gotten up to this step successfully, but cannot run this particular command (enp115s0f1 is not listed when I run ifconfig).

Can I get more information about the expected operation of this command?  Am I to believe that this Alveo kernel will actually show up as an ethernet device within the system?  I can't seem to find anywhere within the example that's actually installing the Alveo card as an ethernet NIC, so I was hoping you could point me to the spot where it's actually installing this as an ethernet NIC in my system?

mariodruiz, 2021-03-07:
Hi @gtrchchn,

The network interfaces on the Alveo card are not exposed as network interfaces on the host, only kernels on the user logic on the Alveo card can access to such interfaces. Exposing these interfaces on the host is not the goal of the designs on this repository.

In the example notebook you point out, in the assumptions read 

1. You have an Alveo card configured with one of the supported shells
1. You have generated the xclbin file for the basic example
1. You have a 100 GbE capable NIC
1. Both the Alveo card and the 100 GbE capable NIC are in the same host
1. Alveo card is connected to the NIC, either directly or with a any network equipment, using any of the interfaces.

So, the `ifconfig` command is used for the 100G NIC, not for the Alveo card.

If you have a 100G NIC card, it is possible that the 100G interface is exposed with a different name.
1. Use `ifconfig -a` or `ip link show` to identify the 100G NIC interface
2. Update interface name accordingly in the notebook.

Mario

gtrchchn, 2021-03-07:
Thank you Mario! I thought you had somehow managed to have the Alveo card show up as an Ethernet NIC, and I was pretty impressed.  I must have misread that the first time I read the assumptions, but that makes perfect sense.  I'll be able to keep debugging from here using a loopback connection.  Thanks for your help

closed 2021-03-07 by gtrchchn

mariodruiz, 2021-03-08:
There are Vivado designs coming with that functionality from other team in Xilinx. Stay tuned

gtrchchn, 2021-03-08:
That's excellent actually! Really interested to stay informed - Is there a better spot to be signed up for information on releases for Alveo?  I'm part of a few Alveo head-start lounges, but the information seems to be sort of split between groups on what features are becoming supported.  Just looking for a better way to be involved

mariodruiz, 2021-03-10:
@gtrchchn 

These are the repo I was referring to:

https://github.com/Xilinx/open-nic-shell
https://github.com/Xilinx/open-nic 
https://github.com/Xilinx/open-nic-driver

For open source projects, keeping an eye on out GitHub is a good idea. You can also ask you FAE to keep you informed