Xilinx/xup_vitis_network_example #22: Readme image is wrong

andreaf2405, 2021-03-26:
I think the "refclk" shown in the image and used for the Alveo board is 161 MHz, not 156. Can you confirm?

mariodruiz, 2021-03-26:
Hi @andreaf2405,

Alveo U200, U250 and U280 use the 156.25 MHz reference clock.
Alveo U50 uses the 161.1328125 MHz reference clock.

Based on your comment I assume you are referring to the Alveo U50.  The readme includes a note under the image that states that the reference clock can change depending on the Alveo card. 

So, the image is not wrong. Only represents the most common scenario. Clock is properly configured depending on the card [here](https://github.com/Xilinx/xup_vitis_network_example/blob/master/Ethernet/cmac/scripts/xilinx_ips/cmac_uplus.tcl#L2-L5)

Mario

andreaf2405, 2021-03-26:
Oh, I see, thanks for clarifying! 
I am trying to adapt it to "hook in"  via TCL a 1-lane aurora core, do you know if that is possible? I am just experimenting, but I fear that the implementation will fail as the aurora will try and instantiate GTs which are already included in the shell.
Btw, you have done some great work here 

Andrea

mariodruiz, 2021-03-26:
It should be possible, but I have not tried anything different from the Ethernet Subsystem.

The shell does not uses any GT transceivers. These pins are directly exposed to user logic, https://developer.xilinx.com/en/articles/designing-a-transceiver-based-application-with-vitus.html

So, if you instantiate the Aurora IP and connect it to the proper GT pins it should work.

closed 2021-03-30 by mariodruiz