Xilinx/xup_vitis_network_example #23: Questions about gt_serial_port0

heiyiren1999, 2021-04-03:
Hello!
I build the project and find there is a bus-interface called "gt_serial_port0" in the component.xml file ("/Ethernet/packaged_kernel_cmac_0_xilinx_u280_xdma_201920_3/component.xml"). In the same .xml file, I also find other bus-interfaces (ap_clk, ap_rst_n, M_AXIS, S_AXIS, S_AXILITE). However, when I check the vivado.log ("/Ethernet/vivado.log"), I cannot find anything about "gt_serial_port0". I can only find something about other bus-interfaces which are shown in the image below.

[image]

So where can I check the generation of the bus-interface "gt_serial_port0"? I think that in this project, the bus-interface "gt_serial_port0" is generated by the tcl command ("/Ethernet/package_kernel.tcl") so there must be something printed as log. I know the component.xml verifies the existence of gt_serial_port0 but it's at a too late stage of the process. I want to check the log or something like that which is directly generated by package_kernel.tcl.

mariodruiz, 2021-04-06:
You could remove the `-notrace `from here https://github.com/Xilinx/xup_vitis_network_example/blob/master/Ethernet/Makefile#L74 and that will print everything.

I don't understand why you want that information, if the process to package to .xo takes only a few seconds.

heiyiren1999, 2021-04-09:
My partner is more familiar with Vitis GUI and Wizard to generate RTL kernel. However, he meets the problem that he cannot generate bus-interface "gt_serial_port0" by editing package_kernel.tcl. So... can user generate their own bus-interface by editing package_kernel.tcl in Wizard? 

ps: I want to trace the generation of bus-interface "gt_serial_port0" in the xup project to support my partner's work, so I ask the previous question.

mariodruiz, 2021-04-09:
Yes, it is possible to generate the gt_serial_port0 interface editing the tcl file. However, that is out of the scope of this repository and it is not recommended to do so for Vitis kernels.

heiyiren1999, 2021-04-10:
OK. Thanks for your help! I'll recommand him to use Makefile.

closed 2021-04-10 by heiyiren1999

mariodruiz, 2021-04-10:
@heiyiren1999 , you can search for the Vivado project that generates the RTL kernel (`Ethernet/tmp_cmac_0_xilinx_u280_xdma_201920_3/kernel_pack.xpr`), you should be able to open it and edit there if necessary. But, this is something that I cannot support.