Xilinx/xup_vitis_network_example #24: Got a error "No parts matched 'xcvu095-ffva2104-2-e" when building for U50

kangyifei, 2021-04-10:
### Build Issues

1. OS version 
Ubuntu 20.04.2 LTS
2. Vivado version 
Vivado v2020.1.1 (64-bit)
SW Build 2960000 on Wed Aug  5 22:57:21 MDT 2020
IP Build 2956692 on Thu Aug  6 01:41:30 MDT 2020
3. XRT version
XRT Build Version: 2.8.743
Build Version Branch: 2020.2
Build Version Hash: 77d5484b5c4daa691a7f78235053fb036829b1e9
Build Version Hash Date: Fri, 13 Nov 2020 07:51:30 -0800
Build Version Date: Mon, 16 Nov 2020 00:18:20 -0800
XOCL: 2.8.743,77d5484b5c4daa691a7f78235053fb036829b1e9
XCLMGMT: 2.8.743,77d5484b5c4daa691a7f78235053fb036829b1e9

When I build this repo for U50,I got a error "No parts matched 'xcvu095-ffva2104-2-e".When I installed Vitis,I have checked all devices.I search this part name on Google,it seems to be a Virtex device.But I still don't know how to get this part.

Any Help will be grateful!

mariodruiz, 2021-04-10:
Hi @kangyifei,

You need to get the board files from here https://www.xilinx.com/products/boards-and-kits/alveo/u50.html#vivado

Could you copy the full log?

Mario

mariodruiz, 2021-04-10:
Hi @kangyifei,

Please open Vivado and go to `Help > Add Designs Tools or Devices` 

If Virtex UltrasScale+ HMB is not installed, install it.

[image]

Additionally, in Vivado go to `Tools > XHub Stores..` , in the Boards tab download the Alveo U280 Data Center Accelerator Card. 

After this, close Vivado and in a new terminal try to rebuild. 

If the problem persist, I suggest to post in the forum https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo

kangyifei, 2021-04-12:
> Hi @kangyifei,
> 
> Please open Vivado and go to `Help > Add Designs Tools or Devices`
> 
> If Virtex UltrasScale+ HMB is not installed, install it.
> 
> [image]
> 
> Additionally, in Vivado go to `Tools > XHub Stores..` , in the Boards tab download the Alveo U280 Data Center Accelerator Card.
> 
> After this, close Vivado and in a new terminal try to rebuild.
> 
> If the problem persists, I suggest to post in the forum https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo
I reinstall Vitis 2020.1 on Ubuntu18.04 and this problem is fixed.

closed 2021-04-12 by kangyifei