Xilinx/xup_vitis_network_example #25: how to make in gui

fyflxl, 2021-04-19:
i'm a newcomer,i have make as the instruction on vitis 2020.2,and it
was  successfully built.but what should i do next?can i open it with a gui?
[image]

mariodruiz, 2021-04-19:
Hi @fyflxl,

What GUI do you want to open? 

The next step after xclbin generation is to download the xclbin onto the FPGA, to do this follow the Notebooks on the Notebook folder

fyflxl, 2021-04-20:
> Hi @fyflxl,
> 
> What GUI do you want to open?
> 
> The next step after xclbin generation is to download the xclbin onto the FPGA, to do this follow the Notebooks on the Notebook fol

> Hi @fyflxl,
> 
> What GUI do you want to open?
> 
> The next step after xclbin generation is to download the xclbin onto the FPGA, to do this follow the Notebooks on the Notebook folder

thank u 

> Hi @fyflxl,
> 
> What GUI do you want to open?
> 
> The next step after xclbin generation is to download the xclbin onto the FPGA, to do this follow the Notebooks on the Notebook folder

thanks a lot, i get it.and later i want to use 10/25 IP onto U50, can i follow this project to make it?

mariodruiz, 2021-04-20:
You can use the architecture VNx provides to use the 10/25 G IP, but that is something VNx does not support. VNx only support the 100G Integrated Ethernet Subsystem. 
If you want to use the 10/25 IP,  you will have to implement it.

fyflxl, 2021-04-20:
> You can use the architecture VNx provides to use the 10/25 G IP, but that is something VNx does not support. VNx only support the 100G Integrated Ethernet Subsystem.
> If you want to use the 10/25G IP, you will have to implement it.

i saw UltraScale+ Integrated 100G Ethernet Subsystem can be set as 10*10G,can i apply it onto U50?waiting for your reply.

mariodruiz, 2021-04-20:
Yes the Integrated 100G Ethernet Subsystem support 10 x 10G lanes, but the QSFP28 does not support such mode 4 x 25G lanes. Either way, you cannot use these lane independently.

To support 10/25 G the 10G/25G Ethernet Subsystem is needed.

fyflxl, 2021-04-20:
> Yes the Integrated 100G Ethernet Subsystem support 10 x 10G lanes, but the QSFP28 does not support such mode 4 x 25G lanes. Either way, you cannot use these lane independently.
> 
> To support 10/25 G the 10G/25G Ethernet Subsystem is needed.
okkk,thank u very much!

mariodruiz, 2021-04-20:
I suggest you discuss your options with your FAE.

closed 2021-04-26 by mariodruiz