Xilinx/xup_vitis_network_example #27: Migrate design to versal

anqianqi1, 2021-05-27:
Hello, 

We would like to migrate the design to the Xilinx versal platform. Could you give us some suggestions and directions about how to achieve that? 
Currently, we are trying to modify the Makefile and tcl scripts of each module to change the device. Is there anything else we need to take care of?
Thank you!

mariodruiz, 2021-05-27:
Hi,

There are substantial architectures changes between Alveo and Versal.

Changing the device is just the beginning, you need to map each module to its corresponding logic in the Versal architecture. 
This will also depends on what the platform provides.

There may be changes in the interfaces as well.

This is something I have not look at in depth, and I am afraid I cannot help at this point.

Mario

mariodruiz, 2021-05-27:
I am closing the issue as it does not apply to the current status of VNx. 
VNx won't be supporting Versal for the foreseeable future.

closed 2021-05-27 by mariodruiz