Xilinx/xup_vitis_network_example #28: Steps to build for xilinx_u200_gen3x16_xdma_1_202110_1 platform

mariodruiz, 2021-06-29:
The `xilinx_u200_gen3x16_xdma_1_202110_1` platform is only supported by Vitis 2021.1, however the network layer relies on Vivado HLS, which is deprecated in Vitis 2020.2. Therefore, to successfully build for this platform, you will need to use both Vitis 2020.1 and Vitis 2021.1.

These are the steps  I used to successfully reproduce the examples for this platform. Use XRT `2.11.634`

## Get a fresh copy of the repository 

Get the repository again and use it only to build for this platform. The goal is to avoid intermediate states

```sh
git clone https://github.com/Xilinx/xup_vitis_network_example.git --recursive vnx_u200
```

## Compile network layer

1. Source Vitis 2020.1 tools
2. Create Network layer internal IP

```sh
cd vnx_u200/NetLayers/100G-fpga-network-stack-core
make
```

## Generate xclbin

After the network layer is compile, we can use Vitis 2021.1 to generate the xclbin

1. Source Vitis 2021.1 tools
2. Run `make`, from the root directory i.e. vnx_u200, to build the example as described [here](https://github.com/Xilinx/xup_vitis_network_example#generate-xclbin)

For instance,

```sh
cd vnx_u200
make all DEVICE=<fullpath>/xilinx_u200_gen3x16_xdma_1_202110_1.xpfm INTERFACE=0
```

## Verify platform 

`xilinx_u200_gen3x16_xdma_1_202110_1.xpfm` features the [two stage platform](https://xilinx.github.io/XRT/2020.2/html/platforms_partitions.html#two-stage-platforms), therefore you need to make sure that the partition is programmed before trying to download the xclbin file. [UG1301 Getting Started with Alveo Accelerator Cards](https://www.xilinx.com/support/documentation/boards_and_kits/accelerator-cards/1_8/ug1301-getting-started-guide-alveo-accelerator-cards.pdf) page 24 step 11

hokim72, 2021-09-10:
I found cmac clocks aren't connected 
```
(Post-linking xilinx.com:au200:part0:1.3 QSFP GT pins Tcl hook):  connecting io_clk_qsfp_refclka_00 -> cmac_0/gt_refclk0
WARNING: [BD 41-1306] The connection to interface port </io_clk_qsfp_refclka_00_clk_n> is being overridden by the user with net <io_clk_qsfp_refclka_00_clk_n_1>. This port will not be connected as a part of interface connection </io_clk_qsfp_refclka_00>.
WARNING: [BD 41-1306] The connection to interface port </io_clk_qsfp_refclka_00_clk_p> is being overridden by the user with net <io_clk_qsfp_refclka_00_clk_p_1>. This port will not be connected as a part of interface connection </io_clk_qsfp_refclka_00>.

(Post-linking xilinx.com:au200:part0:1.3 QSFP GT pins Tcl hook):  connecting io_clk_qsfp_refclka_01 -> cmac_1/gt_refclk1
WARNING: [BD 41-1306] The connection to interface port </io_clk_qsfp_refclka_01_clk_n> is being overridden by the user with net <io_clk_qsfp_refclka_01_clk_n_1>. This port will not be connected as a part of interface connection </io_clk_qsfp_refclka_01>.
WARNING: [BD 41-1306] The connection to interface port </io_clk_qsfp_refclka_01_clk_p> is being overridden by the user with net <io_clk_qsfp_refclka_01_clk_p_1>. This port will not be connected as a part of interface connection </io_clk_qsfp_refclka_01>.
```
I can fix it with following patch
```
diff --git a/Ethernet/duplicate_patch.sh b/Ethernet/duplicate_patch.sh
index ad39b7b..34b344f 100755
--- a/Ethernet/duplicate_patch.sh
+++ b/Ethernet/duplicate_patch.sh
@@ -46,8 +46,10 @@ cp src/template_top.v src/cmac_top_1.v
 
 sed -i 's/module\ placeholder/module\ cmac_0/g' src/cmac_top_0.v
 sed -i 's/cmac_bd\ placeholder/cmac_bd\ cmac_bd_0_i/g' src/cmac_top_0.v
+sed -i 's/gt_refclkx/gt_refclk0/g' src/cmac_top_0.v
 sed -i 's/gt_placeholder_clk/gt_refclk0/g' src/cmac_top_0.v
 
 sed -i 's/module\ placeholder/module\ cmac_1/g' src/cmac_top_1.v
 sed -i 's/cmac_bd\ placeholder/cmac_bd\ cmac_bd_1_i/g' src/cmac_top_1.v
-sed -i 's/gt_placeholder_clk/gt_refclk1/g' src/cmac_top_1.v
\ No newline at end of file
+sed -i 's/gt_refclkx/gt_refclk1/g' src/cmac_top_1.v
+sed -i 's/gt_placeholder_clk/gt_refclk1/g' src/cmac_top_1.v
diff --git a/Ethernet/src/template_top.v b/Ethernet/src/template_top.v
index 5b5b8b9..99a17bb 100755
--- a/Ethernet/src/template_top.v
+++ b/Ethernet/src/template_top.v
@@ -84,7 +84,9 @@ module placeholder #(
   input  wire                        [3:0]    gt_rxn_in,
   output wire                        [3:0]    gt_txp_out,
   output wire                        [3:0]    gt_txn_out, 
+  (* X_INTERFACE_INFO = "xilinx.com:interface:diff_clock:1.0 gt_refclkx CLK_P" *)^M
   input  wire                                 gt_placeholder_clk_p,
+  (* X_INTERFACE_INFO = "xilinx.com:interface:diff_clock:1.0 gt_refclkx CLK_N" *)^M
   input  wire                                 gt_placeholder_clk_n
 );
 
@@ -136,4 +138,4 @@ module placeholder #(
   );
 
 
-endmodule
\ No newline at end of file
+endmodule^M
diff --git a/post_sys_link.tcl b/post_sys_link.tcl
index e8567cb..3de71fc 100755
--- a/post_sys_link.tcl
+++ b/post_sys_link.tcl
@@ -148,17 +148,15 @@ if {[dict exists $config_info kernels]} {
         connect_bd_intf_net [get_bd_intf_ports ${bd_gt_gtyquad_1}] ${__gt_intf}
       }
       # Look for a gt clock capable interface
-      set __refclk0_pins [get_bd_pins -of_objects [get_bd_cells ${__k_inst}] -filter {NAME =~ "gt_refclk0*"}]
-      if {[llength $__refclk0_pins] > 0} {
+      set __refclk0_intf [get_bd_intf_pins -of_objects [get_bd_cells ${__k_inst}] -filter {NAME =~ "gt_refclk0*"}]
+      if {[llength $__refclk0_intf] > 0} {
         puts "${__TCLID} connecting ${bd_gt_ref_clk_0_name_a} -> ${__k_inst}/gt_refclk0"
-        connect_bd_net [get_bd_ports ${bd_gt_ref_clk_0_name_a}_clk_n] [get_bd_pins ${__k_inst}/gt_refclk0_n]
-        connect_bd_net [get_bd_ports ${bd_gt_ref_clk_0_name_a}_clk_p] [get_bd_pins ${__k_inst}/gt_refclk0_p]
+        connect_bd_intf_net [get_bd_intf_ports ${bd_gt_ref_clk_0_name_a}] ${__refclk0_intf}
       }
-      set __refclk1_pins [get_bd_pins -of_objects [get_bd_cells ${__k_inst}] -filter {NAME =~ "gt_refclk1*"}]
-      if {[llength $__refclk1_pins] > 0} {
+      set __refclk1_intf [get_bd_intf_pins -of_objects [get_bd_cells ${__k_inst}] -filter {NAME =~ "gt_refclk1*"}]
+      if {[llength $__refclk1_intf] > 0} {
         puts "${__TCLID} connecting ${bd_gt_ref_clk_1_name_a} -> ${__k_inst}/gt_refclk1"
-        connect_bd_net [get_bd_ports ${bd_gt_ref_clk_1_name_a}_clk_n] [get_bd_pins ${__k_inst}/gt_refclk1_n]
-        connect_bd_net [get_bd_ports ${bd_gt_ref_clk_1_name_a}_clk_p] [get_bd_pins ${__k_inst}/gt_refclk1_p]
+        connect_bd_intf_net [get_bd_intf_ports ${bd_gt_ref_clk_1_name_a}] ${__refclk1_intf}
       }
       # Get Free runninc clock pin name and connection if any
       set __kernel_freerunclk_pins [get_bd_pins -of_objects [get_bd_cells ${__k_inst}] -filter {NAME =~ "clk_gt_freerun"}]
@@ -179,4 +177,4 @@ if {[dict exists $config_info kernels]} {
   }
 }
 
-puts "${__TCLID} QSFP GT pins Tcl hook DONE!"
\ No newline at end of file
+puts "${__TCLID} QSFP GT pins Tcl hook DONE!"
```

mariodruiz, 2021-09-10:
How did you come to that conclusion?

hokim72, 2021-09-10:
I confirmed it by seeing vivado project 
In vivado project, you can see disconnection of clocks before applying patch

mariodruiz, 2021-09-10:
I can assure you that the clocks are connected. If you select the clock pin in the cmac and then look at the properties you will see the net. 

I will verify that using differential interfaces won't give a problem with Vivado and other platforms. But, the current files are correct

hokim72, 2021-09-10:
There is one more patch for u200
I used it to prevent make from rebuilding  netlayer ips with vitis_hls
```
diff --git a/NetLayers/Makefile b/NetLayers/Makefile
index 833413c..62bf7e8 100755
--- a/NetLayers/Makefile
+++ b/NetLayers/Makefile
@@ -62,7 +62,6 @@ distcleanall: distclean
 
 $(TEMP_DIR)/${KRNL_NAME}.xo: kernel.xml package_kernel.tcl gen_xo.tcl src/*.v s
rc/*.vhd
        mkdir -p $(TEMP_DIR)
-       make -C $(SUBMODULENAME) $(SUBMODULETARGET)
        $(VIVADO) -mode batch -source gen_xo.tcl -notrace -tclargs $@ ${KRNL_NAM
E} $(XSA)
 
 check-devices:
@@ -74,4 +73,4 @@ endif
 check-vivado:
 ifndef XILINX_VIVADO
        $(error XILINX_VIVADO variable is not set, please set correctly and reru
n)
-endif
\ No newline at end of file
+endif

```

mariodruiz, 2021-09-10:
This should not be necessary. If the IP already exist, the makefiles shouldn't launch vitis_hls

hokim72, 2021-09-10:
Sorry for confusing you with wrong information
The patch is required to fix the following errors
```
mkdir -p synthesis_results_noHMB
cp Makefile.synthesis synthesis_results_noHMB/Makefile
make -C synthesis_results_noHMB -j4
make[3]: Entering directory '/home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB'
rm -rf ARP_hls_prj
rm -rf ETH_inserter_hls_prj
rm -rf ICMP_hls_prj
rm -rf PKT_HANDLER_prj
vivado_hls -f /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/..//scripts/arp_script.tcl -tclargs /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/../ ARP_hls_prj xcu250-figd2104-2L-e
make[3]: vivado_hls: Command not found
vivado_hls -f /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/..//scripts/ethernet_inserter_script.tcl -tclargs /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/../ ETH_inserter_hls_prj xcu250-figd2104-2L-e
Makefile:47: recipe for target 'ARP_hls_prj' failed
make[3]: *** [ARP_hls_prj] Error 127
make[3]: *** Waiting for unfinished jobs....
make[3]: vivado_hls: Command not found
Makefile:51: recipe for target 'ETH_inserter_hls_prj' failed
make[3]: *** [ETH_inserter_hls_prj] Error 127
vivado_hls -f /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/..//scripts/icmp_script.tcl -tclargs /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/../ ICMP_hls_prj xcu250-figd2104-2L-e
make[3]: vivado_hls: Command not found
Makefile:55: recipe for target 'ICMP_hls_prj' failed
make[3]: *** [ICMP_hls_prj] Error 127
vivado_hls -f /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/..//scripts/packet_handler_script.tcl -tclargs /home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core/synthesis_results_noHMB/../ PKT_HANDLER_prj xcu250-figd2104-2L-e
make[3]: vivado_hls: Command not found
Makefile:59: recipe for target 'PKT_HANDLER_prj' failed
make[3]: *** [PKT_HANDLER_prj] Error 127
make[3]: *** wait: No child processes.  Stop.
Makefile:5: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/hokim/work/vnx_u200/NetLayers/100G-fpga-network-stack-core'
Makefile:64: recipe for target '_x.xilinx_u200_gen3x16_xdma_1_202110_1/networklayer.xo' failed
make[1]: *** [_x.xilinx_u200_gen3x16_xdma_1_202110_1/networklayer.xo] Error 2
make[1]: Leaving directory '/home/hokim/work/vnx_u200/NetLayers'
Makefile:122: recipe for target 'NetLayers/_x.xilinx_u200_gen3x16_xdma_1_202110_1/networklayer.xo' failed
make: *** [NetLayers/_x.xilinx_u200_gen3x16_xdma_1_202110_1/networklayer.xo] Error 2
```
The original makefile deletes ips built by 2020.1 and then tries to use vivado_hls  which does exist in 2021.1

hokim72, 2021-09-10:
[image]
The above shows the connection after applying my patch 
But I couldn't see the connection line before applying my patch 
So I wrote my patch.
Even though bd file doesn't show the connection, there seems be no problem in working properly

mariodruiz, 2021-09-10:
You patch is not wrong, but is not necessary either, I explained above why. For now, I won't update anything because the flow is very robut

PR #36 (2021-09-17): Native support for Vitis HLS
- Update network layer to be implemented with Vitis HLS
- Add frame padding on the cmac kernel
- Bundle differential clock into differential interface in the cmac kernel
- Improve VNx utils
- Update notebooks with the results of this new implementation

Fix #28 #16
closed 2021-09-17 by mariodruiz