Xilinx/xup_vitis_network_example #29: make fails on Ubuntu (possibly all Debian-based) systems due to bash version

nickaj, 2021-08-05:
As noted in issue #16 make can fail with the below errors

Adding "SHELL=/bin/bash" to the top of the Makefile can restore functionality but would need some testing cross distro to check it doesn't cause fails on RH-based systems.

When I 'Generate xclbin'(make all DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm INTERFACE=0 DESIGN=benchmark),

there are errors
/bin/sh: 1: Syntax error: word unexpected (expecting ")")
Makefile:143: recipe for target 'check-design' failed
make: *** [check-design] Error 2

How can I solve?

_Originally posted by @6357o31337 in https://github.com/Xilinx/xup_vitis_network_example/issues/16#issuecomment-881124623_

mariodruiz, 2021-08-05:
Can you provide more details?
- OS
- Shell

I generate the xcbiln file in Ubuntu 18.04 without problems

nickaj, 2021-08-06:
Yes, certainly:
OS is Ubuntu 18.04.5 (LTS) with GNU bash, version 4.4.20(1)-release (x86_64-pc-linux-gnu) and GNU Make 4.1
SHELL is  xilinx_u280_xdma_201920_3
Vitis / Vivado version 2020.1
Compiling the "basic" example

I cut down the error to a short test which illustrates that something happens on Ubuntu-based systems in how bash handles the SHELL variable.

`
#SHELL=/bin/bash
DESIGN=A
check:
        @echo $(DESIGN)
        @if [[ ($(DESIGN) != "B" ) && ($(DESIGN) != "C" ) ]]; then echo Yes; else echo No; fi;
        @echo done
`

Output of above:
`
$ make check
A
/bin/sh: 1: Syntax error: word unexpected (expecting ")")
make: *** [Makefile:5: check] Error 2
`

Output with the first line NOT commented out
`
$ make check
A
Yes
done
`

I also tested this smaller example on Centos 7 with GNU bash, version 4.2.46(2)-release (x86_64-redhat-linux-gnu) and GNU Make 3.82 which gives the second output regardless of whether or not the SHELL= line is commented or uncommented.

mariodruiz, 2021-08-06:
This also works for my with Ubuntu 20.04 and zsh. Feel free to PR the solution

PR #30 (2021-08-06): Fixes #29
This fixes #29
commit a4aeca5a (2021-08-06): Merge pull request #30 from nickaj/master

Fixes #29
closed 2021-08-06 by mariodruiz