Xilinx/xup_vitis_network_example #31: compile error on U50

TimmonSha, 2021-08-23:
I am using a U50 board, vivado 2020.1 and vitis 2020.2。I set up all the environments including xrt, vitis and vivado.
when I tried to compile the project, I run the command: `make all DEVICE=xilinx_u50_gen3x16_xdma_201920_3 INTERFACE=0 DESIGN=basic`
but it shows the following errors: 

```
****** vpl v2020.2 (64-bit)
  **** SW Build (by xbuild) on 2020-11-18-05:13:29
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/sham/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_201920_3/link/int/kernel_info.dat'.
INFO: [VPL 74-74] Compiler Version string: 2020.2
INFO: [VPL 60-423]   Target device: xilinx_u50_gen3x16_xdma_201920_3
INFO: [VPL 60-1032] Extracting hardware platform to /home/sham/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_201920_3/link/vivado/vpl/.local/hw_platform
WARNING: /home/vitis/vitis_2020/Vitis/2020.2/tps/lnx64/jre9.0.4 does not exist.
[17:10:55] Run vpl: Step create_project: Started
Creating Vivado project.
[17:10:59] Run vpl: Step create_project: Completed
[17:10:59] Run vpl: Step create_bd: Started
[17:12:16] Run vpl: Step create_bd: RUNNING...
[17:12:41] Run vpl: Step create_bd: Completed
[17:12:41] Run vpl: Step update_bd: Started
[17:12:42] Run vpl: Step update_bd: Completed
[17:12:42] Run vpl: Step generate_target: Started
[17:13:57] Run vpl: Step generate_target: RUNNING...
[17:14:25] Run vpl: Step generate_target: Completed
[17:14:25] Run vpl: Step config_hw_runs: Started
[17:14:28] Run vpl: Step config_hw_runs: Completed
[17:14:28] Run vpl: Step synth: Started
[17:14:59] Block-level synthesis in progress, 0 of 15 jobs complete, 8 jobs running.
[17:15:30] Block-level synthesis in progress, 0 of 15 jobs complete, 8 jobs running.
[17:16:00] Block-level synthesis in progress, 0 of 15 jobs complete, 9 jobs running.
[17:16:30] Block-level synthesis in progress, 0 of 15 jobs complete, 9 jobs running.
[17:17:01] Block-level synthesis in progress, 0 of 15 jobs complete, 9 jobs running.
[17:17:31] Block-level synthesis in progress, 1 of 15 jobs complete, 9 jobs running.
[17:18:01] Block-level synthesis in progress, 7 of 15 jobs complete, 6 jobs running.
[17:18:32] Block-level synthesis in progress, 8 of 15 jobs complete, 6 jobs running.
[17:19:02] Block-level synthesis in progress, 8 of 15 jobs complete, 6 jobs running.
[17:19:32] Block-level synthesis in progress, 8 of 15 jobs complete, 6 jobs running.
[17:20:03] Block-level synthesis in progress, 9 of 15 jobs complete, 5 jobs running.
[17:20:29] Run vpl: Step synth: Failed
[17:20:30] Run vpl: FINISHED. Run Status: synth ERROR

===>The following messages were generated while  Compiling (synthesis checkpoint) kernel/IP: ulp_networklayer_0_0 Log file: /home/<user>/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_201920_3/link/vivado/vpl/prj/prj.runs/ulp_networklayer_0_0_synth_1/runme.log  :
ERROR: [VPL 8-448] named port connection 'ap_clk' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/<user>/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd5/src/networklayer.v:99]
ERROR: [VPL 8-448] named port connection 'ap_rst_n' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' 
...
ERROR: [VPL 8-6156] failed synthesizing module 'ulp_networklayer_0_0' [/home/<user>/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ip/ulp_networklayer_0_0/synth/ulp_networklayer_0_0.v:58]
```

I have no idea how to fix this error... How should I solve this problem? Thanks.

mariodruiz, 2021-08-23:
Why are you using one version of Vivado and a different one of Vitis?

For this platform I would suggest to use Vitis 2020.1 and XRT 2.6.555 as stated [here](https://github.com/Xilinx/xup_vitis_network_example#tools). Otherwise, you will have to do something like [this](https://github.com/Xilinx/xup_vitis_network_example/issues/16)

TimmonSha, 2021-08-24:
Thank you for your answer. I compiled successfully.

closed 2021-08-24 by TimmonSha

mariodruiz, 2021-08-24:
@TimmonSha can you elaborate what in the steps you took? Your answer might help someone else.