Xilinx/xup_vitis_network_example #32: How to set the link status ?

zhuofanzhang, 2021-09-08:
Hi, I am trying to run the basic example in U250 with the QDMA shell xilinx_u250_qdma_201920_1. However, the link status of both cmac are always FALSE no matter the card is connected with a NIC card or another card. I am wondering if there is any configuration I need to set up to make the link up?

mariodruiz, 2021-09-08:
Hi, 

Do you get link if you connect the QSFP to each other? 
This platform is not supported in this repository, what is more the platform is being deprecated. 

However, I think the problem can be how the reference clock is configured for the GT kernels. The answer to the question above can help answer this.

Mario

zhuofanzhang, 2021-09-09:
Hi, @mariodruiz 

The link is still down  if I connect the QSFP to each other. 

I will try if it works with the XDMA shell.

zhuofanzhang, 2021-09-11:
Hi, @mariodruiz

I am able to run the tests with the QDMA shell xilinx_u250_gen3x16_xdma_3_1_202020_1. However, I still have some link issues when the card is connected to different equipments. To be specific,

1) link up if card_0 is connected to card_1.
2) link up if card_0 is connected to a MELLANOX NIC.
3) link down if card_0 is connected to a INTED NIC.
4) link down if card_0 is connected to the switch.

It might related to the auto-negotiation mode of different network equipments.

In addition, I got a very large std_dev when run the RTT test, 
	mean    = 7403650.928 us
	std_dev = 5346700.192671 us
	mode    = 0.850 us, which appears 16 times
	max     = 14316339.200 us
	min     = 0.850 us 
Do you have any suggestion to improve the RTT performance?

Zhuofan

mariodruiz, 2021-09-13:
Hi,

As you mentioned there could be issues with autonegotiation and FEC. If you have the capability, disable them. This usually solves the problem with switches. I've never used an Intel NIC, so I can't help with that.

What is the setup in the experiment? What notebook are you running?

zhuofanzhang, 2021-09-22:
> Hi,
> 
> As you mentioned there could be issues with autonegotiation and FEC. If you have the capability, disable them. This usually solves the problem with switches. I've never used an Intel NIC, so I can't help with that.
> 
> What is the setup in the experiment? What notebook are you running?

I have two U250 cards (shell version: "xilinx_u250_gen3x16_xdma_shell_3_1") in different servers. cmac_1 of each card is connected by fiber. 

OS version
    LSB Version:	core-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
    Distributor ID:	Ubuntu
    Description:	Ubuntu 18.04.5 LTS
    Release:	18.04
    Codename:	bionic

XRT version 
    XRT Build Version: 2.9.317
    Build Version Branch: 2020.2_PU1
    Build Version Hash: b0230e59e22351fb957dc46a6e68d7560e5f630c
    Build Version Hash Date: Wed, 10 Mar 2021 21:43:11 -0800
    Build Version Date: Sat, 13 Mar 2021 05:10:45 -0800
    XOCL: 2.9.317,b0230e59e22351fb957dc46a6e68d7560e5f630c
    XCLMGMT: 2.9.317,b0230e59e22351fb957dc46a6e68d7560e5f630c

pynq version 
    PYNQ version 2.6.2
    Git Id: 28a1d5e970a6feff7fb7d975825fc58b6dd1f220

Jupyter Lab version
    jupyter core     : 4.7.1
    jupyter-notebook : 6.4.3
    qtconsole        : not installed
    ipython          : 7.16.1
    ipykernel        : 5.5.5
    jupyter client   : 7.0.2
    jupyter lab      : 3.1.10
    nbconvert        : 6.0.7
    ipywidgets       : not installed
    nbformat         : 5.1.3
    traitlets        : 4.3.3

Dask set up
   {'scheduler': {'host': {'python': '3.6.9.final.0',
   'python-bits': 64,
   'OS': 'Linux',
   'OS-release': '4.15.0-45-generic',
   'machine': 'x86_64',
   'processor': 'x86_64',
   'byteorder': 'little',
   'LC_ALL': 'None',
   'LANG': 'zh_CN.UTF-8'},
  'packages': {'python': '3.6.9.final.0',
   'dask': '2021.03.0',
   'distributed': '2021.03.0',
   'msgpack': '1.0.2',
   'cloudpickle': '1.6.0',
   'tornado': '6.1',
   'toolz': '0.8.2',
   'numpy': '1.19.5',
   'lz4': None,
   'blosc': None}},
 'workers': {'tcp://172.177.4.84:38383': {'host': {'python': '3.6.9.final.0',
    'python-bits': 64,
    'OS': 'Linux',
    'OS-release': '4.15.0-45-generic',
    'machine': 'x86_64',
    'processor': 'x86_64',
    'byteorder': 'little',
    'LC_ALL': 'None',
    'LANG': 'zh_CN.UTF-8'},
   'packages': {'python': '3.6.9.final.0',
    'dask': '2021.03.0',
    'distributed': '2021.03.0',
    'msgpack': '1.0.2',
    'cloudpickle': '1.6.0',
    'tornado': '6.1',
    'toolz': '0.8.2',
    'numpy': '1.19.5',
    'lz4': None,
    'blosc': None}},
  'tcp://172.177.4.91:31991': {'host': {'python': '3.6.9.final.0',
    'python-bits': 64,
    'OS': 'Linux',
    'OS-release': '4.15.0-156-generic',
    'machine': 'x86_64',
    'processor': 'x86_64',
    'byteorder': 'little',
    'LC_ALL': 'None',
    'LANG': 'C.UTF-8'},
   'packages': {'python': '3.6.9.final.0',
    'dask': '2021.03.0',
    'distributed': '2021.03.0',
    'msgpack': '1.0.2',
    'cloudpickle': '1.6.0',
    'tornado': '6.1',
    'toolz': '0.8.2',
    'numpy': '1.19.5',
    'lz4': None,
    'blosc': None}}},
 'client': {'host': {'python': '3.6.9.final.0',
   'python-bits': 64,
   'OS': 'Linux',
   'OS-release': '4.15.0-45-generic',
   'machine': 'x86_64',
   'processor': 'x86_64',
   'byteorder': 'little',
   'LC_ALL': 'None',
   'LANG': 'zh_CN.UTF-8'},
  'packages': {'python': '3.6.9.final.0',
   'dask': '2021.03.0',
   'distributed': '2021.03.0',
   'msgpack': '1.0.2',
   'cloudpickle': '1.6.0',
   'tornado': '6.1',
   'toolz': '0.8.2',
   'numpy': '1.19.5',
   'lz4': None,
   'blosc': None}}}

mariodruiz, 2021-09-22:
Everything looks good with the setup. What about the autonegotiation and FEC on the network equipment that does not work?

zhuofanzhang, 2021-09-24:
> Everything looks good with the setup. What about the autonegotiation and FEC on the network equipment that does not work?

As for the switch, I change the fec mode from RS to NONE, then I get link.
As for the intel NIC, still no link since I cannot change the antonego and fec mode of the NIC.

Besides, I got wrong results when I run the Basic application. The received message is different from the send message. Please see below for the result of transmitting 1 pkt of size 64.

`s2mm_wh.wait()
s2mm_buf.sync_from_device()
msg = "SUCCESS!" if np.array_equal(udp_message_global, s2mm_buf) else "FAILURE!"
print("Host sending data through the network and the host getting data from kernel \
was a: {}. Total data transmitted {:,} bytes to {}" .format(msg,size,(alveo_ipaddr, alveo_port)))
print("udp_message_global:\n", udp_message_global)
print("s2mm_buf:\n", s2mm_buf)`

`Host sending data through the network and the host getting data from kernel was a: FAILURE!. Total data transmitted 64 bytes to ('169.254.0.250', 62781)
udp_message_global:
 [[2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]
 [2]]
s2mm_buf:
 [[  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]
 [  0]
 [  0]
 [  0]
 [  0]
 [255]
 [255]
 [255]
 [255]]`

mariodruiz, 2021-09-24:
What is the setup for this?

Can you provide the output of `ol.cmac_1.getStats()`? 
Run this before starting the s2mm kernel, do not report report this result
Re-run `ol.cmac_1.getStats()` after the comparison, report this result

It also curious that the `udp_message_global` only takes 2 and 255 as values, not much randomness. Did you change this?

zhuofanzhang, 2021-09-26:
> What is the setup for this?
> 
> Can you provide the output of `ol.cmac_0.getStats()`?
> Run this before starting the s2mm kernel, do not report report this result
> Re-run `ol.cmac_0.getStats()` after the comparison, report this result
> 
> It also curious that the `udp_message_global` only have 2 as value, not much randomness. Did you change this?

I got the same wong results no matter
1) U250 board is connected to NIC directly,
2) U250 board and NIC are all connected to a 100GE switch.

Some rx states of `ol.cmac_0.getStats()`, e.g., packets, good_packets, bytes, good_bytes, keep changing even the s2mm kernel is not started yet.
Below is the `ol.cmac_0.getStats()` result after comparison. Please noted that I only send 1 pkt of size 64.

{'tx': {'packets': 0,
  'good_packets': 0,
  'bytes': 0,
  'good_bytes': 0,
  'packets_64B': 0,
  'packets_65_127B': 0,
  'packets_128_255B': 0,
  'packets_256_511B': 0,
  'packets_512_1023B': 0,
  'packets_1024_1518B': 0,
  'packets_1519_1522B': 0,
  'packets_1523_1548B': 0,
  'packets_1549_2047B': 0,
  'packets_2048_4095B': 0,
  'packets_4096_8191B': 0,
  'packets_8192_9215B': 0,
  'packets_large': 0,
  'packets_small': 0,
  'bad_fcs': 0,
  'pause': 0,
  'user_pause': 0},
 'rx': {'packets': 13,
  'good_packets': 13,
  'bytes': 1586,
  'good_bytes': 1586,
  'packets_64B': 0,
  'packets_65_127B': 13,
  'packets_128_255B': 0,
  'packets_256_511B': 0,
  'packets_512_1023B': 0,
  'packets_1024_1518B': 0,
  'packets_1519_1522B': 0,
  'packets_1523_1548B': 0,
  'packets_1549_2047B': 0,
  'packets_2048_4095B': 0,
  'packets_4096_8191B': 0,
  'packets_8192_9215B': 0,
  'packets_large': 0,
  'packets_small': 0,
  'packets_undersize': 0,
  'packets_fragmented': 0,
  'packets_oversize': 0,
  'packets_toolong': 0,
  'packets_jabber': 0,
  'bad_fcs': 0,
  'packets_bad_fcs': 0,
  'stomped_fcs': 0,
  'pause': 0,
  'user_pause': 0},
 'cycle_count': 7804510787}

As for the `udp_message_global `, I initialize this buffer with fixed value 2 to see if the wrong result is related to Big/Little Endian. 
`udp_message_global = np.random.randint(low=2, high=3, size=shape, dtype=np.uint8)`

mariodruiz, 2021-09-27:
Can you share the full jupyter notebook you are running?

zhuofanzhang, 2021-09-27:
> Can you share the full jupyter notebook you are running?

Please check the attached file. Thanks a lot.

[vnx-basic.txt](https://github.com/Xilinx/xup_vitis_network_example/files/7234487/vnx-basic.txt)

mariodruiz, 2021-09-27:
After you download the xclbin file onto the Alveo and run the code, can you run `xbutil query` in a terminal (use the appropriate card, if multiple)
There's going to be a section towards where the Compute Unit Status. Please report this full section

zhuofanzhang, 2021-09-27:
> After you download the xclbin file onto the Alveo and run the code, can you run `xbutil query` in a terminal (use the appropriate card, if multiple)
> There's going to be a section towards where the Compute Unit Status. Please report this full section

INFO: Found total 1 card(s), 1 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	4.15.0-45-generic
Version:	#48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019
Machine:	x86_64
CPU cores:	56
Memory:		128432 MB
Glibc:		2.27
Distribution:	Ubuntu 18.04.5 LTS
Now:		Mon Sep 27 08:58:35 2021 GMT
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.9.317
Git Hash:	b0230e59e22351fb957dc46a6e68d7560e5f630c
Git Branch:	2020.2_PU1
Build Date:	2021-03-13 05:10:45
XOCL:		2.9.317,b0230e59e22351fb957dc46a6e68d7560e5f630c
XCLMGMT:	2.9.317,b0230e59e22351fb957dc46a6e68d7560e5f630c

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Shell                           FPGA                            IDCode
xilinx_u250_gen3x16_xdma_shell_3_1                                0x4b57093
Vendor          Device          SubDevice       SubVendor       SerNum          
0x10ee          0x5005          0x000e          0x10ee          21330402S03P    
DDR size        DDR count       Clock0          Clock1          Clock2          
0 Byte          0               260             500             0               
PCIe            DMA chan(bidir) MIG Calibrated  P2P Enabled     OEM ID          
GEN 3x16        2               true            false           0x0(N/A)        
Interface UUID
f2f6c5e1273e78948f2c4806221462f2
Logic UUID
bd5fb8abab266c3265918257b5048e88
DNA                             CPU_AFFINITY    HOST_MEM size   Max HOST_MEM    
                                28-55           0 Byte          16 GB           
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
QSPI Write Protection Status
Primary: N/A
Recovery: N/A
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Temperature(C)
PCB TOP FRONT   PCB TOP REAR    PCB BTM FRONT   VCCINT TEMP     
38              36              38              N/A             
FPGA TEMP       TCRIT Temp      FAN Presence    FAN Speed(RPM)  
40              39              P               N/A             
QSFP 0          QSFP 1          QSFP 2          QSFP 3          
N/A             N/A             N/A             N/A             
HBM TEMP        
N/A             
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Electrical(mV|mA)
12V PEX         12V AUX         12V PEX Current 12V AUX Current 
12284           12040           1826            1216            
3V3 PEX         3V3 AUX         DDR VPP BOTTOM  DDR VPP TOP     
3358            3355            2500            2500            
SYS 5V5         1V2 TOP         1V8 TOP         0V85            
5534            1206            1839            859             
MGT 0V9         12V SW          MGT VTT         1V2 BTM         
903             11990           1207            1205            
VCCINT VOL      VCCINT CURR     VCCINT IO VOL   VCC3V3 VOL      
850             15798           N/A             N/A             
3V3 PEX CURR    VCCINT IO CURR  HBM1V2 VOL      VPP2V5 VOL      
N/A             N/A             N/A             N/A             
VCC1V2 CURR     V12 I CURR      V12 AUX0 CURR   V12 AUX1 CURR   
N/A             N/A             N/A             N/A             
12V AUX1 VOL    VCCAUX VOL      VCCAUX PMC VOL  VCCRAM VOL      
[... 99 lines cut ...]
                scalable: true
                stepable: false
        cmc
            qsfp_ctrl: enabled
            qsfp_interrupts: enabled
        dma
            num_channels: 2
            type: xdma
        ert
            command_queue_slots: tbd
            kernel_interrupts: enabled
        p2p
            maximum_memory_aperture: 64G
    partition_identifiers
        interface_uuids
            695718ec21a232e45e1afcb4e558e11f
                type: resolved
            f2f6c5e1273e78948f2c4806221462f2
                type: exposed
        logic_uuid: bd5fb8abab266c3265918257b5048e88
    partition_name: xdma-shell
    partition_type: shell
    partition_vendor: xilinx
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
INFO: xbutil query succeeded.

mariodruiz, 2021-09-27:
Thanks, I think I know where the problem is. 
Can you install pynq 2.7 https://github.com/Xilinx/xup_vitis_network_example/issues/34? 

This should solve the problem.

For info, a version of XRT change the way the CU index is computed. This produces a mix up, which can be seen below, the kernel that starts is `krnl_mm2s_1` instead of `krnl_s2mm_0`, therefore what you read from memory is not valid data. 

```console
Compute Unit Status                      Addr              Status        Usage
CU[  4]: krnl_mm2s:krnl_mm2s_1           @0x1440000         (IDLE)        1  
```

Upgrading pynq will solve the problem.

zhuofanzhang, 2021-09-27:
> Thanks, I think I know where the problem is.
> Can you install pynq 2.7? follow these steps #34
> After upgrading pynq, restart the notebook (either by restarting the kernel, or stopping and starting the notebook again) for the changes to take effect.
> 
> For info, a version of XRT change the way the CU index is computed. This produces a mix up, which can be seen below, the kernel that starts is `krnl_mm2s_1` instead of `krnl_s2mm_0`, therefore what you read from memory is not valid data.
> 
> ```
> Compute Unit Status                      Addr              Status        Usage
> CU[  4]: krnl_mm2s:krnl_mm2s_1           @0x1440000         (IDLE)        1  
> ```
> 
> Upgrading pynq will solve the problem.

Great! Transmission from NIC to Alveo Card succeed after upgrading pynq.

But packets loss occurs when moving data from Alveo Card to NIC. Only 73 packets received at while 100 packets are send. Seems the card is sending data too fast then cause packet loss.

`rx app: total payload packets 100	bytes 140,800	 throughput 2.645 Gbps
tx app: total payload packets 100	bytes 140,800	 throughput 149.920 Gbps`

zhuofanzhang, 2021-09-27:
> Great! Transmission from NIC to Alveo Card succeed after upgrading pynq.
>
> But packets loss occurs when moving data from Alveo Card to NIC. Only 73 packets received at while 100 packets are send. [...]

I found the issue. The packets loss is caused by the NIC. IRQ affinity should be mapped specially for the NIC to achieve high throughput transmission. 

Thanks again.

I will close the issue.

zhuofanzhang, 2021-09-27:
Seems pynq 2.7 has some compatible problem with DASK. Got the follow error when run Benchmark Round Trip time experiment (Switch).

`print("Link worker 0 {}; link worker 1 {}".format(ol_w0.cmac_0.linkStatus(),ol_w1.cmac_0.linkStatus()))`
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-d8628b68571c> in <module>
----> 1 print("Link worker 0 {}; link worker 1 {}".format(ol_w0.cmac_0.linkStatus(),ol_w1.cmac_0.linkStatus()))

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in __getattr__(self, key)
    367         """
    368         if self.is_loaded():
--> 369             return getattr(self._ip_map, key)
    370         else:
    371             raise RuntimeError("Overlay not currently loaded")

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in __getattr__(self, key)
    885             ipdescription = self._description['ip'][key]
    886             try:
--> 887                 driver = ipdescription['driver'](ipdescription)
    888             except UnsupportedConfiguration as e:
    889                 warnings.warn(

/home/local_z00417915_hz84/Vitis/xup_vitis_network_example-master_XDMA/Notebooks/vnx_utils.py in __init__(self, description)
     87 
     88     def __init__(self, description):
---> 89         super().__init__(description=description)
     90         self._fullpath = description['fullpath']
     91         self.start = self.start_sw = self.start_none = \

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in __init__(self, description)
    715             self._registers = None
    716         if 'index' in description:
--> 717             cu_index = self.device.open_contex(description)
    718             self.cu_mask = 1 << cu_index
    719             self._setup_packet_prototype()

AttributeError: 'DaskDevice' object has no attribute 'open_contex'

mariodruiz, 2021-09-27:
Since the original issue is solved I am closing this issue.
Please open a new issue, otherwise it is difficult to keep track.

closed 2021-09-27 by mariodruiz