Xilinx/xup_vitis_network_example #33: One of the two network interfaces of U280 is down

jijinfanhua, 2021-09-09:
Hi,
Sorry for not replying to @mariodruiz for a long time as there is a power problem with U280. I can't send a post on https://discuss.pynq.io/ because the filter Akismet always hides the post. So I make this issue.

 After I download .xclbin of XUP to U280 (two interfaces are connected to the switch via two qsfp28 Tansceiver modules and a cable), and use the _vnx-basic.ipynb_ to configure the FPGA, there is only one of the two network interfaces is UP.

The _make_ command is:
`make all DEVICE=xilinx_u280_xdma_201920_3 INTERFACE=3 DESIGN=basic
`

The result of _vnx-basic.ipynb_ is:
`Link interface 0 {'cmac_link': True}; link interface 1 {'cmac_link': False}`

I adjust the connecting scheme from 1 to 2:
[image]
and the result of _vnx-basic.ipynb_ doesn't change. So there is no problems with modules and the cable.
Are there some solutions?

Thanks a lot for your help.

mariodruiz, 2021-09-09:
Are the switch port configured the in same way? 
Are you using the same cables?

I assume that if you connect the two Alveo ports to each other you get link in both of them?

jijinfanhua, 2021-09-09:
> Are the switch port configured the in same way?
> Are you using the same cables?
> 
> I assume that if you connect the two Alveo ports to each other you get link in both of them?

The switch ports are configured in the same way, and the cables are the same.
I connect two ports of U280 with a cable, and the result is 'False, False'.

mariodruiz, 2021-09-09:
The problem seems to be be with interface 1 in the Alveo card.

What commit are you building against? 

You problem seems somehow related to this one https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/The-LED-becomes-red-when-a-qsfp28-Tansceiver-module-connects-to/td-p/1284735

Can you also confirm the SC that is running on the Alveo card?

jijinfanhua, 2021-09-09:
Yeah, that's my question too. And I'm upgrading the platform.

mariodruiz, 2021-09-09:
I am afraid that the problem is with the platform and not this repository. For this reason I am closing the issue.

closed 2021-09-09 by mariodruiz