Xilinx/xup_vitis_network_example #35: AttributeError: module 'pynq.pl_server' has no attribute 'xclbin_parser'

HuangKai0225, 2021-09-17:
I tried this setup on the open-cloud umass server, configured everything from scratch and ran the experiments but I got an error. 

I installed anaconda amd created the environment with the environment.yml from https://pynq.readthedocs.io/en/latest/getting_started/alveo_getting_started.html#alveo-extended-setup 

Then I ran 'pip install pynq' to install pynq and the version is 2.6.2. 

[image]

I think two or three months ago, I tried this setup, the same configuration, it works fine but this time it isn't. I want to know whether it is due to version maintenance.

mariodruiz, 2021-09-17:
Can you please provide this information?

1. OS version `lsb_release -a`
1. XRT version `xbutil version`

Are you able to download the xclbin without using Dask?

HuangKai0225, 2021-09-17:
I double checked the xrt setup. I think it is due to that I did not source xrt.setup.sh for the newly opened terminal to run ipython. I need to source the setup.sh for each terminal. I sourced the xrt setup.sh for the dask-scheduler and dask-workers terminals but missed one.

I also checked our lab server. It should be the same reason.

It should be fine now.  
Thank you so much. I will close this.

Can I ask one question in this thread?
[image]

Sometimes, the link is true for cmac_0 and false for cmac_1. However, sometimes, it is opposite. Cmac_0 link is false and cmac_1 link is true. Is this fine?

Thanks.

mariodruiz, 2021-09-17:
Yes indeed, you need the same environment in all the workers and scheduler, otherwise it will fail. 

I don't know your setup exactly. But once the FPGA is programmed with the xclbin if the link goes up it should never go down unless something changes in the setup.

closed 2021-09-20 by HuangKai0225