Xilinx/xup_vitis_network_example #37: Network setup for bidirectional data transfer test with benchmark kernel

HuangKai0225, 2021-09-20:
Hi,

I am trying to do a bi-directional data transfer test with the benchmark kernel. My mode setting is: 
k00 latency  k01 loopback k10 loopback k11 latency
k00-> k10->k00
k11 -> k01 ->k11

When I run my codes, the latency numbers from the collector are all zero. I think there is still something wrong.

The code snippet is:
```
print(ol_w1.networklayer_1.updateIPAddress('192.168.0.10', debug=True))
# 2
ol_w1.networklayer_1.sockets[0] = ('192.168.0.5', 62177, 60512, True)
ol_w1.networklayer_1.sockets[1] = ('192.168.0.5', 62173, 60511, True)
ol_w1.networklayer_1.populateSocketTable()
#3 
ol_w1.networklayer_1.arpDiscovery()
#4
ol_w1.networklayer_1.readARPTable()

print(ol_w0.networklayer_1.getNetworkInfo())
#2
ol_w0.networklayer_1.sockets[2] = ('192.168.0.10', 60512, 62177, True)
ol_w0.networklayer_1.sockets[3] = ('192.168.0.10', 60511, 62173, True)
ol_w0.networklayer_1.populateSocketTable()
#3 
ol_w0.networklayer_1.arpDiscovery()
#4
ol_w0.networklayer_1.readARPTable()

send_packets   = 2 ** 10
shape          = (send_packets,1)
rtt_cycles0     = pynq.allocate(shape, dtype=np.uint32, target=ol_w0.HBM0)
pkt0            = pynq.allocate(1,     dtype=np.uint32, target=ol_w0.HBM0)

#allocate FPGA 0 kernel 0 recv
collector_h00 = ol_w0.collector_1_0.start(rtt_cycles0,pkt0)

rtt_cycles1     = pynq.allocate(shape, dtype=np.uint32, target=ol_w1.HBM0)
pkt1            = pynq.allocate(1,     dtype=np.uint32, target=ol_w1.HBM0)

#allocate FPGA 1 kernel 1 recv
collector_h11 = ol_w1.collector_1_1.start(rtt_cycles1,pkt1)

#connfigure FPGA 1 kernel 0  
ol_w10_tg = ol_w1.traffic_generator_1_0
ol_w10_tg.register_map.mode = benchmark_mode.index('LOOPBACK')
ol_w10_tg.register_map.dest_id = 0 # Use connection in position 0 to reflect
ol_w10_tg.register_map.CTRL.AP_START = 1

#connfigure FPGA 0 kernel 1  
ol_w01_tg = ol_w0.traffic_generator_1_1
ol_w01_tg.register_map.mode = benchmark_mode.index('LOOPBACK')
ol_w01_tg.register_map.dest_id = 1 # Use connection in position 1
ol_w01_tg.register_map.CTRL.AP_START = 1

#configure FPGA 0 kernel 0
send_pkts = send_packets
ol_w00_tg = ol_w0.traffic_generator_1_0
ol_w00_tg.register_map.debug_reset = 1
ol_w0.networklayer_1.register_map.debug_reset_counters = 1
ol_w00_tg.register_map.mode = benchmark_mode.index('LATENCY')
ol_w00_tg.register_map.number_packets = send_pkts
ol_w00_tg.register_map.time_between_packets = 50
ol_w00_tg.register_map.number_beats = 1
ol_w00_tg.register_map.dest_id = 0

#configure FPGA 1 kernel 1
send_pkts = send_packets
ol_w11_tg = ol_w1.traffic_generator_1_1
ol_w11_tg.register_map.debug_reset = 1
ol_w1.networklayer_1.register_map.debug_reset_counters = 1
ol_w11_tg.register_map.mode = benchmark_mode.index('LATENCY')
ol_w11_tg.register_map.number_packets = send_pkts
ol_w11_tg.register_map.time_between_packets = 50
ol_w11_tg.register_map.number_beats = 1
ol_w11_tg.register_map.dest_id = 1

# start the two kernels
ol_w00_tg.register_map.CTRL.AP_START = 1
ol_w11_tg.register_map.CTRL.AP_START = 1
```

I have the following questions:
1. If I need to have two network connections, is the socket setup correct? I mean the IP address is the same, right? The port number should be OK.
2. A general question is, if I use only one AXI-stream port for both receiving and sending at the same time, it should work, right? How is the AXI-stream interface mapped to the hardware socket? 

Thanks.
Kai

mariodruiz, 2021-09-20:
Hi,

1. It does not seem to be set up properly. The configuration on the UDP table for worker 0 should be. You could also change the dest_id. See below answer below. 
```python
ol_w0.networklayer_1.sockets[0] = ('192.168.0.10', 60512, 62177, True)
ol_w0.networklayer_1.sockets[1] = ('192.168.0.10', 60511, 62173, True)
```
2. Yes, it should work. The description of how that's mapped is here https://github.com/Xilinx/xup_vitis_network_example/tree/master/NetLayers#udp. Basically, the dest in the traffic generator is the index of the socket table.

You can use the embedded probes on the network layer to verify how far the packets are getting. For instance, `ol_w0.networklayer_1.getDebugProbes`. In your current configuration, you will see that the payload from the application FPGA0 arrives to the UDP module but there is no output from the UDP module, this is because tdest is set to 0 and 1, and these entries are not configured in the socket table.

HuangKai0225, 2021-09-20:
Hi Mario,

OK. My understanding is that the problem for the above code setup is the dest_id right? The dest id is the socket number. Instead I thought it is the index of kernel.

Best,
Kai

HuangKai0225, 2021-09-20:
I changed the socket ID and used your fix above.

In this case, k00 dest id: 0   k 10 dest id: 0   k01 dest id: 1 and k11 dest id: 1. Is this correct?
I checked the output latency from collectors, they are just zeros:

```
In [21]: rtt_cycles1.sync_from_device() 
    ...: rtt_cycles1                                                                                                                                                                                        
Out[21]: 
PynqBuffer([[0],
            [0],
            [0],
            ...,
            [0],
            [0],
            [0]], dtype=uint32)

In [22]: rtt_cycles0.sync_from_device() 
    ...: rtt_cycles0                                                                                                                                                                                        
Out[22]: 
PynqBuffer([[0],
            [0],
            [0],
            ...,
            [0],
            [0],
            [0]], dtype=uint32)
```

I also printed the debug probe, but I do not know how to interpret it:
```
In [24]: ol_w0.networklayer_0.getDebugProbes                                                                                                                                                                
Out[24]: 
{'tx_path': {'arp_packets': 0,
  'arp_bytes': 0,
  'arp_cycles': 0,
  'icmp_packets': 0,
  'icmp_bytes': 0,
  'icmp_cycles': 0,
  'ethernet_header_inserter_packets': 2048,
  'ethernet_header_inserter_bytes': 122880,
  'ethernet_header_inserter_cycles': 52749,
  'ethernet_packets': 2048,
  'ethernet_bytes': 122880,
  'ethernet_cycles': 52749,
  'udp_from_app_packets': 2048,
  'udp_from_app_bytes': 94208,
  'udp_from_app_cycles': 52749,
  'udp_packets': 2048,
  'udp_bytes': 36864,
  'udp_cycles': 52749},
 'rx_path': {'ethernet_packets': 1027,
  'ethernet_bytes': 61620,
  'ethernet_cycles': 22603694377,
  'packet_handler_packets': 1024,
  'packet_handler_bytes': 47104,
  'packet_handler_cycles': 52173,
  'arp_packets': 0,
  'arp_bytes': 0,
  'arp_cycles': 0,
  'icmp_packets': 0,
  'icmp_bytes': 0,
  'icmp_cycles': 0,
  'udp_packets': 1024,
  'udp_bytes': 47104,
  'udp_cycles': 52173,
  'udp_to_app_packets': 1024,
  'udp_to_app_bytes': 18432,
  'udp_to_app_cycles': 52173}}

In [26]: ol_w1.networklayer_0.getDebugProbes                                                                                                                                                                
Out[26]: 
{'tx_path': {'arp_packets': 0,
  'arp_bytes': 0,
  'arp_cycles': 0,
  'icmp_packets': 0,
  'icmp_bytes': 0,
  'icmp_cycles': 0,
  'ethernet_header_inserter_packets': 1024,
  'ethernet_header_inserter_bytes': 61440,
  'ethernet_header_inserter_cycles': 52172,
  'ethernet_packets': 1024,
  'ethernet_bytes': 61440,
  'ethernet_cycles': 52172,
  'udp_from_app_packets': 1024,
  'udp_from_app_bytes': 47104,
  'udp_from_app_cycles': 52172,
  'udp_packets': 1024,
  'udp_bytes': 18432,
  'udp_cycles': 52172},
 'rx_path': {'ethernet_packets': 2052,
  'ethernet_bytes': 123120,
  'ethernet_cycles': 32033390492,
  'packet_handler_packets': 2048,
  'packet_handler_bytes': 94208,
  'packet_handler_cycles': 52748,
  'arp_packets': 0,
  'arp_bytes': 0,
  'arp_cycles': 0,
  'icmp_packets': 0,
  'icmp_bytes': 0,
  'icmp_cycles': 0,
  'udp_packets': 2048,
  'udp_bytes': 94208,
  'udp_cycles': 52748,
  'udp_to_app_packets': 2048,
  'udp_to_app_bytes': 36864,
  'udp_to_app_cycles': 52748}}

``` 

I think if it sends some data, the arp_packets should not be zero? Correct me if I am wrong. 
Is the dest_id setup correct? Why it's still not working?

Thank you so much.
Kai

HuangKai0225, 2021-09-20:
Thanks. Mario.
I just got the test with benchmark kernel working.

Best,
Kai

closed 2021-09-20 by HuangKai0225

mariodruiz, 2021-09-21:
@HuangKai0225 can you please share the solution? It may be useful for others

HuangKai0225, 2021-09-21:
You mean reopen this issue? I think the issue is the dest_id. It should match the socket index.

mariodruiz, 2021-09-21:
No, I mean to comment what made the experiment work. Was this suggestion https://github.com/Xilinx/xup_vitis_network_example/issues/37#issuecomment-923175344?

HuangKai0225, 2021-09-21:
Yes. I just commented in my previous post.

mariodruiz, 2021-09-21:
I am confused, you said here that it did not work https://github.com/Xilinx/xup_vitis_network_example/issues/37#issuecomment-923334567 

Did you make any other change?

HuangKai0225, 2021-09-21:
I think it is due to I try to test on two different servers. However, one of them is CMAC_0 but the other one is CMAC_1. I need to  change accordingly for the setup. After I double checked the kernel settings, the experiment works. 

I think the only issue is the dest id. If that's correct, it works.

I am also customizing the benchmark and test with other hardware. Thank you so much.