Xilinx/xup_vitis_network_example #38: Using struct for axi stream pkt in Basic kernel

anqianqi1, 2021-09-20:
Hello, 

I met some issues when modifying the Basic mm2s kernel(using the original hls kernel the communication works well). Could you please help me if possible? Thank you!

I am trying to use 'struct' for 'hls::stream<pkt> &k2n' in Basic mm2s kernel. Making 'k2n.data' has several struct members.
Here is an example:

```
typedef struct{
  struct{
    ap_int<3>         data_id;
    ap_int<PAYLOAD_WIDTH>     payload;   
  }data;
  bool              last;
  bool              keep;
  ap_uint<TDWIDTH>        dest;
} pkt;
```

Our setup is using two Alveo U280 boards with cable directly connected.
However, the Netlayers kernel seems not recognizing the '.data' field.  From 'networklayer_1.register_map', the sender board is not sending out any packets.

Could you please give any suggestions on how to change axi stream packets into a struct axi packets?
Thank you so much!

mariodruiz, 2021-09-21:
What interface is being created on the mm2s kernel? 
My understanding is that Vitis HLS does not support custom structs for input/output streams. If you want to map that to an AXI4-Stream you need typedef `ap_axiu<DWIDTH, 96, 1, TDWIDTH> pkt`;

So my suggestion is to create your payload structure and cast it to 512-bit of the data signal.

mariodruiz, 2021-09-21:
You can read about this here https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/managing_interface_synthesis.html?#doq1539734222381 
Customs struct are aggregated and cannot be dissagregated. The recommendation is to use the one of the struct defined in `ap_axi_sdata.h`

anqianqi1, 2021-09-22:
Hi mario,

Thanks for the reply. 
I am using the same AXI stream face as the original mm2s kernel.
Thank you for your suggestion, I'll try to cast the payload into 512 bit data signal.