Xilinx/xup_vitis_network_example #39: AttributeError: 'DaskDevice' object has no attribute 'open_contex'

mariodruiz, 2021-09-28:
pynq version `2.7`

After installing pynq `2.7` dask returns the following error 

```python
print("Link worker 0 {}; link worker 1 {}".format(ol_w0.cmac_0.linkStatus(),ol_w1.cmac_0.linkStatus()))
AttributeError Traceback (most recent call last)
in
----> 1 print("Link worker 0 {}; link worker 1 {}".format(ol_w0.cmac_0.linkStatus(),ol_w1.cmac_0.linkStatus()))

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in getattr(self, key)
367 """
368 if self.is_loaded():
--> 369 return getattr(self._ip_map, key)
370 else:
371 raise RuntimeError("Overlay not currently loaded")

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in getattr(self, key)
885 ipdescription = self._description['ip'][key]
886 try:
--> 887 driver = ipdescription'driver'
888 except UnsupportedConfiguration as e:
889 warnings.warn(

/home/local_z00417915_hz84/Vitis/xup_vitis_network_example-master_XDMA/Notebooks/vnx_utils.py in init(self, description)
87
88 def init(self, description):
---> 89 super().init(description=description)
90 self._fullpath = description['fullpath']
91 self.start = self.start_sw = self.start_none = \

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in init(self, description)
715 self._registers = None
716 if 'index' in description:
--> 717 cu_index = self.device.open_contex(description)
718 self.cu_mask = 1 << cu_index
719 self._setup_packet_prototype()

AttributeError: 'DaskDevice' object has no attribute 'open_contex'
```

mariodruiz, 2021-09-28:
Temporary workaround until this PR is merge https://github.com/Xilinx/PYNQ/pull/1252 

Add these functions to the [`DaskDevice` class](https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/dask_pynq.py#L93-L159)

```python
def open_contex(self, description, shared=True):
    return pynq.Device.active_device.open_contex(description, shared)

def close_contex(self, cu_name):
    pynq.Device.active_device.close_contex(cu_name)
```

zhuofanzhang, 2021-09-28:
> Temporary workaround until this PR is merged [Xilinx/PYNQ#1252](https://github.com/Xilinx/PYNQ/pull/1252)
> 
> Add these functions to the [`DaskDevice` class](https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/dask_pynq.py#L93-L159)
> 
> ```python
> def open_contex(self, description, shared=True):
>     return pynq.Device.active_device.open_contex(description, shared)
> 
> def close_contex(self, cu_name):
>     pynq.Device.active_device.close_contex(cu_name)
> ```

Got the following error after add above functions to DaskDevice class,

`print("Link worker 0 {}; link worker 1 {}".format(ol_w0.cmac_0.linkStatus(),ol_w1.cmac_0.linkStatus()))
`
```
/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/pl_server/device.py:81: UserWarning: No devices found, is the XRT environment sourced?
  UserWarning)
```
```
RuntimeError                              Traceback (most recent call last)
<ipython-input-6-d8628b68571c> in <module>
----> 1 print("Link worker 0 {}; link worker 1 {}".format(ol_w0.cmac_0.linkStatus(),ol_w1.cmac_0.linkStatus()))

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in __getattr__(self, key)
    367         """
    368         if self.is_loaded():
--> 369             return getattr(self._ip_map, key)
    370         else:
    371             raise RuntimeError("Overlay not currently loaded")

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in __getattr__(self, key)
    885             ipdescription = self._description['ip'][key]
    886             try:
--> 887                 driver = ipdescription['driver'](ipdescription)
    888             except UnsupportedConfiguration as e:
    889                 warnings.warn(

/home/local_z00417915_hz84/Vitis/xup_vitis_network_example-master_XDMA/Notebooks/vnx_utils.py in __init__(self, description)
     87 
     88     def __init__(self, description):
---> 89         super().__init__(description=description)
     90         self._fullpath = description['fullpath']
     91         self.start = self.start_sw = self.start_none = \

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/overlay.py in __init__(self, description)
    715             self._registers = None
    716         if 'index' in description:
--> 717             cu_index = self.device.open_contex(description)
    718             self.cu_mask = 1 << cu_index
    719             self._setup_packet_prototype()

/home/local_z00417915_hz84/Vitis/xup_vitis_network_example-master_XDMA/Notebooks/dask_pynq.py in open_contex(self, description, shared)
    161 
    162     def open_contex(self, description, shared=True):
--> 163         return pynq.Device.active_device.open_contex(description, shared)
    164 
    165     def close_contex(self, cu_name):

/usr/local/lib/python3.6/dist-packages/pynq-2.7.0-py3.6-linux-x86_64.egg/pynq/pl_server/device.py in active_device(cls)
     92         if not hasattr(cls, '_active_device'):
     93             if len(cls.devices) == 0:
---> 94                 raise RuntimeError("No Devices Found")
     95             cls._active_device = cls.devices[0]
     96         return cls._active_device

RuntimeError: No Devices Found
```

mariodruiz, 2021-09-28:
Did you update pynq in both machines? 
Are you using the same environment in both machines?

I tried this work around and I worked for me.

zhuofanzhang, 2021-09-28:
> Did you update pynq in both machines? Are you using the same environment in both machines?
> 
> I tried this work around and I worked for me.

Yes, I update pynq in both machines.

This application works fine if I downgrade to pynq 2.6.2.

mariodruiz, 2021-09-28:
That is because the benchmark notebooks are not using the index to start the application. For the other notebook, you will find the same error.

After you upgrade pynq to 2.7, are you able to run the basic notebook in both machines?

zhuofanzhang, 2021-09-28:
My mistake. I did not source XRT environment before starting jupyter notebook. 
Issue resolved.
Sorry for the misleading.

commit e07b6463 (2021-10-08): Add open and close context methods. Fixes #39