Xilinx/xup_vitis_network_example #47: How to realize none packet loss by using s2mm kernel ?

zhuofanzhang, 2021-11-26:
Hi Mario,

I have a UDP client keep sending packets (8192B/pkt) to a U250 board. In order to receive packets continuously, I keep enqueueing s2mm kernel. However, as shown in the following timeline figure, I found there are some packets dropped during two calls.

[image]

The following shows the value of some debug port registers of the networklayer. As you can see, the values of 'udp_in_packets' and 'udp_app_out_packets' are different.

```
UDP Debug INFOs: 
'udp_in_cycles'        : 5453975         
'udp_in_bytes'         : 82233344        
'udp_in_packets'       : 10000           
'udp_app_in_cycles'    : 0               
'udp_app_in_bytes'     : 0               
'udp_app_in_packets'   : 0               
'udp_out_cycles'       : 0               
'udp_out_bytes'        : 0               
'udp_out_packets'      : 0               
'udp_app_out_cycles'   : 5452065         
'udp_app_out_bytes'    : 81920128        
'udp_app_out_packets'  : 9996   
```

Any suggestion on realizing none packet loss using s2mm kernel ?  Thanks.

mariodruiz, 2021-11-26:
This kernel is used to show basic functionality, it is not meant to be used to receive packets continuously. 
You could use a larger size here https://github.com/Xilinx/xup_vitis_network_example/blob/master/Basic_kernels/src/krnl_s2mm.cpp#L51, perhaps change the type to `long long int`.

An alternative, is to enable auto restart in the kernel. But, this cannot be done with the enqueueTask OpenCL API. You would need to write the registers manually using the appropriate API.

Another alternative, is to get Vitis to add a FIFO between the Network layer and the kernel.  

Ideally, you will create your own kernel to meet your needs.

zhuofanzhang, 2021-12-06:
Hi Mario,

Thanks for your quick reply.

Alternative 1 will not help a lot since it still needs to keep enqueueing s2mm kernel.

I will dig deeper into alternative 2. I was intended to write a free-running s2mm kernel by using QDMA platform. However, QDMA platform is not supported in this repository. I will try writing a Auto-Restarting kernels as you suggested.

As for alternative 3, isn't the AXI streaming interface between the Network layer and s2mm kernel a FIFO ?

mariodruiz, 2021-12-06:
Hi,

For 3, it is not a FIFO, it a direct stream connection.

closed 2021-12-15 by zhuofanzhang