Xilinx/xup_vitis_network_example #48: How could I replace NetLayers with my own customized kernel?

IskandarZhang, 2021-12-01:
Hi Mario. In my last issue, you have mentioned that re-design NetLayers would be helpful to minimize the gap between 'actual' and 'thero' throughout. I suggest it is an inspiring idea and I'd like to make a start-up with some simple replacement. To start with, I want to focus on implementation first rather than improving performance. So I want to start with implementing a simple RTL kernel, which communicate with `cmac` kernel through AXIS and connect with `benchmark` kernel. I'd like to ask how the existing `NetLayers` is connected to the `benchmark` kernel and `cmac` kernel? It would be very helpful if you could provide some guidance.

Best.

mariodruiz, 2021-12-01:
Vitis does the linking (connections) using a configuration file. https://github.com/Xilinx/xup_vitis_network_example/blob/master/config_files/connectivity_benchmark_if0.ini#L23-L44 

You should get familiar with Vitis flow and how to link streaming kernels, these are good resoures to start with 
- https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/buildingdevicebinary.html#yha1565541199876
- https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/devrtlkernel.html

For any other questions regarding Vitis flow and designing kernels I would refer you to the Xilinx forums or documentation

IskandarZhang, 2021-12-02:
Thanks for your guiding, I'd like to start with those resources mentioned above.

Best.

closed 2022-04-22 by mariodruiz