Xilinx/xup_vitis_network_example #49: Question about the AXI4-Lite interface of the 100G Ethernet Subsystem ip.

LaceFern, 2021-12-30:
Hello. I find that the 100G Ethernet Subsystem ip (cmac_uplus_0) you use in the Ethernet RTL Kernel has an AXI4-Lite interface (Fig. 1) because you enable the IP option "Include AXI4-Lite Control and Statistics Interface" (Fig. 2).
[image]
(Fig. 1)
[image]
(Fig. 2)
However, I can not find where you assign values to this interface because I'm not sure which module directly coorperates with the Ethernet RTL Kernel. Would you like to tell me where you assign values to the AXI4-Lite interface of the Ethernet RTL Kernel? I want to check these values. Thanks in advance!

mariodruiz, 2021-12-30:
Hi, 
The register map is in the [UltraScale+ Devices
Integrated 100G Ethernet
Subsystem Product Guide](https://www.xilinx.com/products/intellectual-property/cmac_usplus.html#documentation). The IP is mapped to offset 0x0 within the kernel. When Vitis links the kernels, it will assign an offset address automatically. Using this address you can access the kernel from the host side.

Hope this answers your question.

LaceFern, 2021-12-31:
I check the notebook "Benchmark Throughput experiment (point to point)" and find that there are no necessary values must be set up to the cmac for reaching line rate. So does the performance of the cmac in this project equals the performance of the cmac with AXI4-Lite interface unenabled?

mariodruiz, 2022-01-02:
The AXI4-Lite interface is for monitoring, the AXI4-Lite interface has no bearing in the performance. 
The speed and operation mode is configured from Vivado and cannot be changed at runtime. This is explained in the PG 203

LaceFern, 2022-01-03:
Thanks for your help very much!

closed 2022-01-03 by LaceFern