Xilinx/xup_vitis_network_example #50: The

LaceFern, 2022-01-04:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

If you still want to raise an issue here, please give us as much detail as 
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version `lsb_release -a`
1. Vivado version `vivado -version`
1. XRT version `xbutil version`

### Run Time Issues

1. OS version `lsb_release -a`
1. XRT version `xbutil version`
1. pynq version `pynq version`
1. Jupyter Lab and Dask version if applicable

LaceFern, 2022-01-07:
I test the Layer 2 bandwidth with a naked CMAC IP and draw the picture below. I calculate the Layer 2 packet size (the size of the pink space about **Layer 2 Ethernet frame** in the pic. above) in your project, then use the formuler **UDP payload bandwidth / UDP payload size * Layer 2 packet size** to calculate the Layer 2 bandwidth of your project. The theoretical bandwidth is calculated by the formular **100 gbps / Layer 1 packet&IPG size * Layer 2 packet size**. Layer 1 packet&IPG size is the size of the pink space about **Layer 1 Ethernet packet & IPG** in the pic. above) 

[image]

More details are in the chart following.

[image]

This problem obstructs my work so I'm concern about it. Thanks for your help in advance!

LaceFern, 2022-01-07:
Sorry, I forgot to add 4B of CRC which is generated by CMAC into L2 Packet Size in Naked CMAC Bandwidth Test experiment. So the true experiment results are as below.

[image]

[image]

mariodruiz, 2022-01-10:
Hi,
You are right, I am not including the inter packet gap (12-Byte).

To compute the goodput at L2, you should count
- Inter-packet gap , 12-Byte
- Ethernet header, 14-Byte
- IP header, 20-Byte
- UDP header, 8-Byte

So, for a 64-Byte payload, the goodput is, 64/(64+12+14+20+8) = 64/118 = 0.5423. 54.23 Gbps it is a slightly slower than reported. I will double check this in the future.

How is this difference preventing your work?

mariodruiz, 2022-01-11:
Hi,
You are right, I am not including the inter packet gap (12-Byte) nor the preamble and frame delimiter in my calculations. 
I need to look at this into more details.

How is this problem obstructing your work?

LaceFern, 2022-01-12:
I need to implement network module with UDP protocol on my Vivado project for U280. 
At first, I used the network module in the Limago[https://github.com/hpcn-uam/Limago](url). However, I find the L2 bandwidth of this network module (the **orange** line with the legend **Vivado Udp** in the pic. below) is far lower than the theoretical bandwidth when the L2 Packet Size is small, namely, when the UDP payload is small.

[image]
[image]

However, in my work, the UDP payload is always small, so I need a network design with higher bandwidth for small UDP payload. That's the reason why I'm concern about this problem. 

I have tryed to read the code of Limago and yours but found it's hard to start with so many lines and files. I read the paper about Limago but it has little details about UDP. In fact, in my work, I don't need the whole TCP/IP stack but only functions about UDP. Would you like to offer some suggestions for me?  

Thanks in advance!

mariodruiz, 2022-01-12:
There's an overhead for small packets that cannot be avoided with the current code. The Network layer is not optimized for small packets.

Limago is archived and not longer supported. 

If you want to improve the goodput for small packets, you will have to redesign the network layer completely. 
You can start with the [UDP code here](https://github.com/hpcn-uam/100G-fpga-network-stack-core/blob/vnx-vitis-hls/hls/UDP/udp.cpp). 

We do not have the bandwidth to support this, this repository is provided as-is and serves as an example or starting point only.

IskandarZhang, 2022-03-12:
> 

Hello, The work you have done is very instructive to me, I also want to test a naked CMAC IP bandwidth to compare with this example project, I would be very grateful if you could give some guidance on how to carry out the test.

Best!

mariodruiz, 2022-03-14:
The Xilinx® UltraScale+™ Devices Integrated 100G Ethernet IP subsystem provides a high-performance, low-latency 100 Gb/s
Ethernet port that allows for a wide range of user customization and statistics gathering. For more information refer to the user guide. https://www.xilinx.com/content/dam/xilinx/support/documentation/ip_documentation/cmac_usplus/v3_1/pg203-cmac-usplus.pdf 

There's always an overhead of adding any layer on top of the physical signals. You can compute this theoretically, no need to run any experiment.

commit e526d10c (2022-06-20): Compute theoretical throughput properly. Fix #50
closed 2022-06-20 by mariodruiz