Xilinx/xup_vitis_network_example #53: Place_design error while running kernel_pack.xpr of Ethernet folder

IskandarZhang, 2022-04-08:
Hello Mario, me again. After `.xclbin` file generated, I found that some Vivado project files `kernel_pack.xpr` generated in `tmp_<ip name>_xilinx_u50_gen3x16_xdma_201920_3` these folders. Since I have always wanted to obtain some synthesis data of these projects such as power consumption and timing, I chose to run the implementation but it failed as the figure below shows.

[image]

I'm wondering this method of getting synthesis results is correct or not. In fact, I'd rather get the synthesis results of the entire project, including the Ethernet, Networklayers, and Benchmark/Basic kernels, but I can't find a way available to that. I really hope you can give some guidance. Thanks in advance.

mariodruiz, 2022-04-08:
If you want to measure power, you can use the power rails available on the board. PYNQ allows you to do this. 

There should also be a project for the full design. 

I am closing this as it does not really apply here.

closed 2022-04-08 by mariodruiz

mariodruiz, 2022-04-08:
You can find the full project on \<design\>\.<interface\>.\<platform_name\>/link/vivado/vpl/prj/prj.xpr