Xilinx/xup_vitis_network_example #56: Add entry to ARP table

108anup, 2022-04-22:
Is it possible to manually add an entry to the arp table?

I see that the `vnx_utils.py` defines a function that just does MMIO reads for reading the ARP table, from this I understand which address/offsets need to be updated. However, I wasn't able to use MMIO writes to update the ARP table. Specifically, I perform an MMIO write on a particular offset in ARP table, no error is returned, but when I read from the same offset I get an old value, not the newly written value.

I can do MMIO writes to the socket table. On reading the same offset in the socket table, I get the newly written entry.

Do I need to somehow update the HLS code annotations to make the ARP table memory writable? Since some other H/W module writes to the ARP table as well, is it then not at all possible to write to this table from the host CPU?

108anup, 2022-04-22:
Also in `vnx_utils.py` I noticed in the socket table writes, there is no endianness change while writing. While for ARP table reading, there is an endianness change function used. Is it just that the hardware reading the two tables uses different endianness?

mariodruiz, 2022-04-25:
Hi @108anup,

It should be possible to write to the ARP table. I have not tried myself. However, the [invalidateARPTable](invalidateARPTable) clears the valid bit. The MAC address is stored in little endian, that's why we need to reorder the bytes. 

What code have you tried?

108anup, 2022-04-25:
The arp_server.cpp chooses table idx based on last 8 bits of IP. Here is a small example following that convention, but this also does not work:
```
import pprint
myip = ipaddress.IPv4Address('10.0.0.65')
little_int_ip = _byteOrderingEndianess(int(myip))
print(myip, int(myip), little_int_ip)

my_mac_hex = 0x40a6b722ab88
my_mac = _byteOrderingEndianess(my_mac_hex, 6)
my_mac_msb = my_mac // (1<<32)
my_mac_lsb = my_mac % (1<<32)
print(my_mac_hex, my_mac, my_mac_msb, my_mac_lsb)
print("")

ol_w0.networklayer_0.invalidateARPTable()
ol_w0.networklayer_0.arpDiscovery()
pprint.pprint(ol_w0.networklayer_0.readARPTable())
print("")

i = 65
print("Writing to i={}".format(i))
old_ip = ol_w0.networklayer_0.read(ip_addr_offset + (i * 4))
print("Old IP {}".format(ipaddress.IPv4Address(_byteOrderingEndianess(old_ip))))
ol_w0.networklayer_0.write(ip_addr_offset + (i * 4), little_int_ip)
ol_w0.networklayer_0.write(mac_addr_offset + (i * 2 * 4), my_mac_lsb)
ol_w0.networklayer_0.write(mac_addr_offset + ((i * 2 + 1) * 4), my_mac_msb)
ol_w0.networklayer_0.write(valid_addr_offset + (i // 4) * 4, 1 << ((i % 4) * 8)) # valid
pprint.pprint(ol_w0.networklayer_0.readARPTable())
print("")

ol_w0.networklayer_0.write(ip_addr_offset + (i * 4), little_int_ip)
print(ol_w0.networklayer_0.read(ip_addr_offset + (i * 4)))

ol_w0.networklayer_0.write(ip_addr_offset + (i * 4), int(myip))
print(ol_w0.networklayer_0.read(ip_addr_offset + (i * 4)))
```

```
10.0.0.65 167772225 1090519050
71084781251464 150268603246144 34987 582460992

{53: {'IP address': '10.0.0.53', 'MAC address': '00:0a:35:04:40:b5'},
 55: {'IP address': '10.0.0.55', 'MAC address': '00:0a:35:04:40:b5'}}

Writing to i=65
Old IP 0.0.0.0
{53: {'IP address': '10.0.0.53', 'MAC address': '00:0a:35:04:40:b5'},
 55: {'IP address': '10.0.0.55', 'MAC address': '00:0a:35:04:40:b5'},
 65: {'IP address': '0.0.0.0', 'MAC address': '40:a6:b7:22:ab:88'}}

0
0
```

mariodruiz, 2022-04-25:
Hi @108anup,

I was looking at the generated HDL code and there is no logic to write the IP address from the host as you indicated, only logic for reading is implemented. 

Can you elaborate on your use case for this?

Mario

108anup, 2022-04-25:
I was using VNx as a packet generator and testing a custom NIC that does not perform ARP. So I wanted to add an entry manually. If I don't then VNx just broadcasts to all MACs.
Is there another project which might be suited to implement packet generation/benchmarking on Xilinx FPGA?

Also, I wanted to know why is there then logic/functionality to update MAC address? Also in HLS, by default all global arrays should be writable right? Is there a way to get HLS to generate code for writing?

mariodruiz, 2022-04-25:
> Is there another project which might be suited to implement packet generation/benchmarking on Xilinx FPGA?

OpenNIC https://github.com/Xilinx/open-nic.

> Also, I wanted to know why is there then logic/functionality to update MAC address?

The board mac address? Yes, this is possible, there is no direct method to do update the mac, but it is done indirectly here https://github.com/Xilinx/xup_vitis_network_example/blob/master/Notebooks/vnx_utils.py#L483  

> Also in HLS, by default all global arrays should be writable right? Is there a way to get HLS to generate code for writing?

It should be the case, but it's not doing it. I have an idea how to solve it. Would you be able to test if I provide a patch?

108anup, 2022-04-25:
Open NIC does not have a generator module inbuilt. I could port the kernels to run on Open NIC. Fundamentally Open NIC uses the host network stack and acts as a dumb NIC. Ideally I wanted something that never has to interact with host to generate packets (to avoid PCIe bottleneck).

No, I meant the MAC addresses in the ARP table. Why am I able to update MAC addresses but not IP addresses. I thought the HLS code does not specify either of these behaviors, so why is it that part of the state can be updated from host and part can't be.

Yes I can test any patch you provide, I can write the patch myself also, if you give me an idea of the solution you have in mind.

mariodruiz, 2022-04-25:
Apply these changes. Run `make distclean` in the `NetLayers` and `100G-fpga-network-stack-core` and then you can rebuild 

```diff
diff --git a/hls/arp_server/arp_server.cpp b/hls/arp_server/arp_server.cpp
index 968117a..f3bdc85 100755
--- a/hls/arp_server/arp_server.cpp
+++ b/hls/arp_server/arp_server.cpp
@@ -240,7 +240,7 @@ void arp_table(
                        arpRequestOut.write(query_ip);  // send ARP request
                }
                // Send response to an IP to MAC association
-               macIpEncode_rsp.write(arpTableReply(currEntry.macAddress, currEntry.valid));
+               macIpEncode_rsp.write(arpTableReply(currEntry.macAddress, currEntry.ipAddress, currEntry.valid));
        }
 
 }
diff --git a/hls/arp_server/arp_server.hpp b/hls/arp_server/arp_server.hpp
index 200a8e4..b933c08 100755
--- a/hls/arp_server/arp_server.hpp
+++ b/hls/arp_server/arp_server.hpp
@@ -63,10 +63,11 @@ const uint8_t       noOfArpTableEntries     = 8;
 struct arpTableReply
 {
        ap_uint<48> macAddress;
+       ap_uint<32> ipAddress;
        bool            hit;
        arpTableReply() {}
-       arpTableReply(ap_uint<48> macAdd, bool hit)
-                       :macAddress(macAdd), hit(hit) {}
+       arpTableReply(ap_uint<48> macAdd, ap_uint<32> ip, bool hit)
+                       :macAddress(macAdd), ipAddress(ip), hit(hit) {}
 };
 
 struct arpTableEntry {
@@ -143,4 +144,4 @@ void arp_server(
                ap_uint<32>&                    gatewayIP,
                ap_uint<32>&                    networkMask);
 
-#endif
\ No newline at end of file
+#endif

```

mariodruiz, 2022-04-25:
@108anup,

Please, do not test this. It is not going to work. I need to check the changes in more detail

108anup, 2022-04-25:
In general, what causes HLS to generate the axi lite control port for global variables?

From this: https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/pragma-HLS-interface
I would imagine, this line in arp_server.cpp should create control interface for the arpTable. Right?
`#pragma HLS INTERFACE s_axilite port=arpTable bundle=s_axilite`

Would moving this annotation outside the scope of the arp_server function help?

108anup, 2022-04-25:
Since arpTable is disaggregated, we might need to add axilite interface to each element separately.
https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/pragma-HLS-disaggregate

It would be something like below, maybe not exactly like it. The examples in the above link don't show array of structs, but show a struct with arrays.

```
#pragma HLS INTERFACE s_axilite port=arpTable[0].macAddress bundle=s_axilite
#pragma HLS INTERFACE s_axilite port=arpTable[0].ipAddress bundle=s_axilite
#pragma HLS INTERFACE s_axilite port=arpTable[0].valid bundle=s_axilite
```

mariodruiz, 2022-04-26:
Hi, the directives in the code are correct. 
The IP address in the table is a bit of an overhead, because only IP within the subnetwork will be added. The index in the table will be enough to identify an IP. Regardless, it is nice to have the IP address there.

These changes make the IP entries writable. Can you please test and let me know?

https://github.com/hpcn-uam/100G-fpga-network-stack-core/commit/fa8c5b8c75d87900d7e4355247463c91726d7cff

Mario

108anup, 2022-04-26:
Thanks I shall try it out.

I still don't fully understand what prevents/allows some state in HLS to be writable (given that it is already marked with an axilite INTERFACE directive).

Also should I cherry pick the last commit from the arp_write_ip branch, or should I directly use that entire branch from the network stack.

mariodruiz, 2022-04-26:
The compiler is optimizing away the writable operation from the host side, I am not sure why is doing it.

>Also should I cherry pick the last commit from the arp_write_ip branch, or should I directly use that entire branch from the network stack.

As you prefer, if the changes work I am going to merge the changes in the appropriate branch

108anup, 2022-04-27:
I am in process for trying the patch.
I seem to be having compilation issues with error: "caught Tcl error: ERROR: '2204271649' is an invalid argument."
Seems to be related to https://support.xilinx.com/s/question/0D52E00006uzPZuSAM/caught-tcl-error-error-2201031626-is-an-invalid-argument-please-specify-an-integer-value?language=en_US

I am using Vitis 2021.2

108anup, 2022-04-27:
I am getting error msgs which look like below.
I am rechecking with a clean repo with/without arp-write patch to double check everything. But if below issue seems genuine please help out.
```
===>The following messages were generated while  Compiling (synthesis checkpoint) kernel/IP: pfm_dynamic_networklayer_0_0 Log file: /home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.runs/pfm_dynamic_networklayer_0_0_synth_1/runme.log  :
ERROR: [VPL 8-448] named port connection 'ap_clk' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:99]
ERROR: [VPL 8-448] named port connection 'ap_rst_n' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:100]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tvalid' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:102]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:103]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:104]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:105]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:106]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tvalid' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:108]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:109]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:110]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:111]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:112]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:114]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:115]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:116]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilin [... cut ...]
ERROR: [VPL 8-6156] failed synthesizing module 'networklayer' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/6fd3/src/networklayer.v:34]
ERROR: [VPL 8-6156] failed synthesizing module 'pfm_dynamic_networklayer_0_0' [/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ip/pfm_dynamic_networklayer_0_0/synth/pfm_dynamic_networklayer_0_0.v:58]
ERROR: [VPL 17-69] Command failed: Synthesis failed - please see the console or run log file for details
```

mariodruiz, 2022-04-28:
Did you patch Vivado for [the Y2K22 bug](https://support.xilinx.com/s/article/76960?language=en_US) as described [README](https://github.com/Xilinx/xup_vitis_network_example#support)?

108anup, 2022-04-28:
Yes I did apply the Y2K22 patch. So this error is only coming with the arp-write patch. Without that commit, the latest vanilla VNx compiles.

mariodruiz, 2022-04-28:
What Vitis version are you using?

108anup, 2022-04-28:
I am using 2021.2

108anup, 2022-04-28:
The generated network_layer_bd.v file is basically an empty module:

`/home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/benchmark.intf3.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.gen/my_rm/bd/bd/ipshared/3b35/src/network_layer_bd.v`

```
//Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.
//--------------------------------------------------------------------------------
//Tool Version: Vivado v.2021.2 (lin64) Build 3367213 Tue Oct 19 02:47:39 MDT 2021
//Date        : Thu Apr 28 13:40:22 2022
//Host        : neptune5 running 64-bit Ubuntu 18.04.6 LTS
//Command     : generate_target network_layer_bd.bd
//Design      : network_layer_bd
//Purpose     : IP block netlist
//--------------------------------------------------------------------------------
`timescale 1 ps / 1 ps

(* CORE_GENERATION_INFO = "network_layer_bd,IP_Integrator,{x_ipVendor=xilinx.com,x_ipLibrary=BlockDiagram,x_ipName=network_layer_bd,x_ipVersion=1.00.a,x_ipLanguage=VERILOG,numBlks=0,numReposBlks=0,numNonXlnxBlks=0,numHierBlks=0,maxHierDepth=0,numSysgenBlks=0,numHlsBlks=0,numHdlrefBlks=0,numPkgbdBlks=0,bdsource=USER,synth_mode=OOC_per_IP}" *) (* HW_HANDOFF = "network_layer_bd.hwdef" *)
module network_layer_bd
   ();

endmodule
```

108anup, 2022-04-28:
From the log, the issue seems to be some missing IPs (xilinx.com:hls:ethernet_header_inserter:1.0 xilinx.com:hls:packet_handler:1.0 xilinx.com:hls:udp:1.0 xilinx.com:hls:arp_server:1.0 xilinx.com:hls:icmp_server:1.0), do I need some other licenses? I have the CMAC IP license.

```
INFO: [BD::TCL 103-2003] Currently there is no design <network_layer_bd> in project, so creating one...
Wrote  : </home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.srcs/sources_1/bd/network_layer_bd/network_layer_bd.bd>
INFO: [BD 41-2613] The output directory /home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd for network_layer_bd cannot be found.
INFO: [BD::TCL 103-2004] Making design <network_layer_bd> as current_bd_design.
INFO: [BD::TCL 103-2005] Currently the variable <design_name> is equal to "network_layer_bd".
INFO: [BD::TCL 103-2011] Checking if the following IPs exist in the project's IP catalog:
xilinx.com:ip:axis_register_slice:1.1 xilinx.com:ip:axis_switch:1.1 xilinx.com:hls:ethernet_header_inserter:1.0 xilinx.com:hls:packet_handler:1.0 xilinx.com:ip:smartconnect:1.0 xilinx.com:hls:udp:1.0 xilinx.com:ip:xlconstant:1.1 xilinx.com:hls:arp_server:1.0 xilinx.com:hls:icmp_server:1.0  .
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
ERROR: [BD::TCL 103-2012] The following IPs are not found in the IP Catalog:
  xilinx.com:hls:ethernet_header_inserter:1.0 xilinx.com:hls:packet_handler:1.0 xilinx.com:hls:udp:1.0 xilinx.com:hls:arp_server:1.0 xilinx.com:hls:icmp_server:1.0

Resolution: Please add the repository containing the IP(s) to the project.
INFO: [BD::TCL 103-2020] Checking if the following modules exist in the project's sources:
axi4stream_sinker bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg interface_settings performance_debug_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg  .
WARNING: [BD::TCL 103-2023] Will not continue with creation of design due to the error(s) above.
Wrote  : </home/ubuntu/Projects/StaRR-NIC/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.srcs/sources_1/bd/network_layer_bd/network_layer_bd.bd>
```

mariodruiz, 2022-04-28:
I think the HLS IP are not being generated properly. 

If you run this in the command line within the `100G-fpga-network-stack-core` directory, you should see 10 files.

```sh
find -name *.zip
```

If the output is empty, the HLS IP are not generated. To understand, the problem better, always within the  `100G-fpga-network-stack-core` directory run

```sh
# Remove existing IP
make distclean
```

```sh
# Generate HLS IP for U280
make hbm > log.txt
```

When, finished search for this line within the ``log.txt` file

```console
INFO: Applying HLS Y2K22 patch v1.2 for IP revision
```

If this line is not found, the Y2K22 bug is not applied to Vitis HLS.

108anup, 2022-04-28:
`find -name "*.zip"` gave empty result.
So I did `make distclean` and then `make hbm > log.txt`
I do see the line saying applying patch. I also see the zip files now with
`find -name "*.zip"`

Could there be some issue with the makefile?

```
./synthesis_results_HMB/ETH_inserter_hls_prj/ultrascale_plus/impl/ip/xilinx_com_hls_ethernet_header_inserter_1_0.zip
./synthesis_results_HMB/ETH_inserter_hls_prj/ultrascale_plus/impl/export.zip
./synthesis_results_HMB/PKT_HANDLER_prj/ultrascale_plus/impl/ip/xilinx_com_hls_packet_handler_1_0.zip
./synthesis_results_HMB/PKT_HANDLER_prj/ultrascale_plus/impl/export.zip
./synthesis_results_HMB/UDP_prj/ultrascale_plus/impl/ip/xilinx_com_hls_udp_1_0.zip
./synthesis_results_HMB/UDP_prj/ultrascale_plus/impl/export.zip
./synthesis_results_HMB/ICMP_hls_prj/ultrascale_plus/impl/ip/xilinx_com_hls_icmp_server_1_0.zip
./synthesis_results_HMB/ICMP_hls_prj/ultrascale_plus/impl/export.zip
./synthesis_results_HMB/ARP_hls_prj/ultrascale_plus/impl/ip/xilinx_com_hls_arp_server_1_0.zip
./synthesis_results_HMB/ARP_hls_prj/ultrascale_plus/impl/export.zip
```

mariodruiz, 2022-04-28:
Don't think so. 

Before rebuilding the design run `make distclean` in the NetLayer folder just in case. 

Then rebuild. Let me know the outcome.

108anup, 2022-04-28:
Ideally the root Makefile calls `make -C 100G-fpga-network-stack-core hbm`. 
I don't know why the zips were absent. 
I had done `git clean -xfd` before calling `make all DEVICE=xilinx_u280_xdma_201920_3 INTERFACE=3 DESIGN=benchmark`.

Anyway, after manually running `make hbm` inside `100G-network-stack-core`, the IP error does not show up, the generated network_layer_bd.v is non-empty. Block synthesis jobs are done. The optimization phases are running... Hopefully this will complete compilation.

108anup, 2022-04-28:
I am able to write to the IP address field in the ARP table. I also had to do `xbutil reset`. For some reason, pynq did not load new xclbin on instantiating overlay.

Can add this function to the `vnx_utils` (#57):
```
    def writeARPEntry(self, mac, ip):
        """Add an entry to the ARP table

        Parameters
        ----------
        mac: str
            MAC address in the format XX:XX:XX:XX:XX:XX
        ip: str
            IP address in the format XXX.XXX.XXX.XXX
        """

        if not isinstance(mac, str):
            raise ValueError("MAC address must be a string.")
        elif not isinstance(ip, str):
            raise ValueError("IP address must be a string.")

        mac_int = int("0x{}".format(mac.replace(":", "")), 16)
        little_mac_int = _byteOrderingEndianess(mac_int, 6)
        mac_msb = little_mac_int // (2 ** 32)
        mac_lsb = little_mac_int % (2 ** 32)

        ip_int = int(ipaddress.IPv4Address(ip))
        little_ip_int = _byteOrderingEndianess(ip_int, 4)

        mac_addr_offset = self.register_map.arp_mac_addr_offset.address
        ip_addr_offset = self.register_map.arp_ip_addr_offset.address
        valid_addr_offset = self.register_map.arp_valid_offset.address

        i = ip_int % 256
        self.write(ip_addr_offset + (i * 4), little_ip_int)
        self.write(mac_addr_offset + (i * 2 * 4), mac_lsb)
        self.write(mac_addr_offset + ((i * 2 + 1) * 4), mac_msb)

        # Valid
        old_valid_entry = self.read(valid_addr_offset + (i // 4) * 4)
        this_valid = 1 << ((i % 4) * 8)
        self.write(valid_addr_offset + (i // 4) * 4, old_valid_entry | this_valid)
```

108anup, 2022-04-28:
Keeping issue open for merging arp write functionality into upstream. Please close if needed.

closed 2022-04-29 by mariodruiz

108anup, 2022-04-29:
BTW, thanks for prompt action on this issue @mariodruiz!

commit 7d5251ee (2022-04-29): Make IP entries in the ARP modules writable from host. Fix #56