Xilinx/xup_vitis_network_example #61: not able to generate .xclbin file for Ip in "xup_vitis_network_example" for network interface in ALVEO U280

SPala121, 2022-05-17:
I am trying to build IP for network interface that is in the github(https://github.com/Xilinx/xup_vitis_network_example) for ALVEO U280 FPGA. I am running make file provided in the code. 

make all Device=xilinx_u280_xdma_201920_3 INTERFACE=0 DESIGN=basic

It is failing at Block-level synthesis with following error:
'''
[16:56:41] Run vpl: Step create_project: Started
Creating Vivado project.
[16:57:05] Run vpl: Step create_project: Completed
[16:57:05] Run vpl: Step create_bd: Started
[17:00:14] Run vpl: Step create_bd: RUNNING...
[17:02:01] Run vpl: Step create_bd: RUNNING...
[17:04:27] Run vpl: Step create_bd: RUNNING...
[17:05:43] Run vpl: Step create_bd: RUNNING...
[17:06:25] Run vpl: Step create_bd: Completed
[17:06:25] Run vpl: Step update_bd: Started
[17:06:29] Run vpl: Step update_bd: Completed
[17:06:29] Run vpl: Step generate_target: Started
[17:07:52] Run vpl: Step generate_target: RUNNING...
[17:08:47] Run vpl: Step generate_target: Completed
[17:08:47] Run vpl: Step config_hw_runs: Started
[17:08:57] Run vpl: Step config_hw_runs: Completed
[17:08:57] Run vpl: Step synth: Started
[17:10:02] Block-level synthesis in progress, 0 of 18 jobs complete, 8 jobs running.
[17:12:50] Block-level synthesis in progress, 0 of 18 jobs complete, 8 jobs running.
[17:33:45] Block-level synthesis in progress, 0 of 18 jobs complete, 8 jobs running.
[17:33:45] Run vpl: FINISHED. Run Status: synth ERROR
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/media/settysha/Softwares/xup_vitis_network_example/basic.intf0.xilinx_u280_xdma_201920_3/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [17:34:56] Run run_link: Step vpl: Failed
Time (s): cpu = 00:05:39 ; elapsed = 00:42:35 . Memory (MB): peak = 2064.742 ; gain = 0.000 ; free physical = 1106 ; free virtual = 2727
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:110: basic.intf0.xilinx_u280_xdma_201920_3/vnx_basic_if0.xclbin] Error 1

'''

1. OS version Ubuntu 20.04.4 LTS
2. Vivado version v2021.2
3. XRT version 2.12.427
4. Vitis IDE v2021.2.0
5. Platform xilinx_u280_xdma_201920_3

mariodruiz, 2022-05-17:
Hi @SPala121,

How much memory do you have in your system?
It is likely that you ran out of memory.

SPala121, 2022-05-17:
Hi @mariodruiz ,

System has 16GB memory

mariodruiz, 2022-05-17:
That's not enough. You need 32GB free for Vivado. Memory requirements here https://www.xilinx.com/products/design-tools/vivado/vivado-ml.html#memory. Virtex Ultrascale+  device similar to XCVU11P

closed 2022-05-17 by mariodruiz

mariodruiz, 2022-05-17:
@SPala121, I closed the issue as I documented the memory requirements on the main README

SPala121, 2022-05-17:
Hi @mariodruiz 

Or will issue be not having license for UltraScale+ Integrated 100G Ethernet Subsystem IP as mentioned in read.md?
Because I am able to synthesis basic designs in my existing Vivado version(that comes with Vitis SDK). 
Also, do i need to have Vivado ML version to synthesis this IP? Because in read.md, there is no specification for Vivado versions

mariodruiz, 2022-05-17:
The problem you are facing is with memory. You may also have a problem to generate the xclbin if no license for the UltraScale+ Integrated 100G Ethernet Subsystem is available. 

When you install Vitis, Vivado is also installed and used to implement the design (this is transparent for the user)

commit b15cc333 (2022-05-17): Include memory requirements. Fix #61