Xilinx/xup_vitis_network_example #63: Is it possible to use VNx example for connecting two cards on the same host?

sergey-gra, 2022-05-20:
I have two Alveo U50 cards on the same host and want to evaluate data transfer between them through QSFP interface. Can I use VNx example to reproduce that? If yes, what I need to change in sources?

mariodruiz, 2022-05-20:
Hi @sergey-gra ,

> Can I use VNx example to reproduce that?
Yes

> If yes, what I need to change in sources?

Small changes on the notebooks as described here https://github.com/Xilinx/xup_vitis_network_example/tree/master/Notebooks#multiple-alveo-cards-in-the-same-system

Mario

mariodruiz, 2022-05-20:
To clarify, no modifications on the build process, and you can reuse the same `xclbin` file for both boards.

sergey-gra, 2022-05-20:
Thank you for your response @mariodruiz . I'll follow your advice.

closed 2022-05-20 by sergey-gra