Xilinx/xup_vitis_network_example #64: Version error

Qiang-Flute, 2022-05-20:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

If you still want to raise an issue here, please give us as much detail as 
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version `lsb_release -a`
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.7 LTS
Release:	16.04
Codename:	xenial
```
1. Vivado version `vivado -version`
```
Vivado v2021.1 (64-bit)
SW Build 3247384 on Thu Jun 10 19:36:07 MDT 2021
IP Build 3246043 on Fri Jun 11 00:30:35 MDT 2021
Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.
```

1. XRT version `xbutil version`
```
       XRT Build Version: 2.11.634
    Build Version Branch: 2021.1
      Build Version Hash: 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
 Build Version Hash Date: Tue, 8 Jun 2021 20:06:48 -0700
      Build Version Date: Tue, 08 Jun 2021 22:09:01 -0700
                    XOCL: 2.11.634,5ad5998d67080f00bca5bf15b3838cf35e0a7b26
                 XCLMGMT: 2.11.634,5ad5998d67080f00bca5bf15b3838cf35e0a7b26
```

### Run Time Issues
Here are the log with error when source to vitis 2021.1 when do make all operation
```
source /tools/Xilinx/Vitis/2021.1/settings64.sh 
 source /opt/xilinx/xrt/setup.sh
make all DEVICE=xilinx_u280_xdma_201920_3 INTERFACE=0 DESIGN=basic
```
```
make: Warning: File 'Makefile' has modification time 30801214 s in the future
cp config_files/connectivity_basic_if0.ini configuration_basic_if0.tmp.ini
echo "" >> configuration_basic_if0.tmp.ini
echo "" >> configuration_basic_if0.tmp.ini
echo "[advanced]" >> configuration_basic_if0.tmp.ini
echo "param=compiler.userPostSysLinkOverlayTcl=/home/shouqicao/workspace/xup/xup_vitis_network_example/Ethernet/post_sys_link.tcl" >> configuration_basic_if0.tmp.ini
#echo "param=compiler.worstNegativeSlack=-2" >> configuration_basic_if0.tmp.ini
#echo "param=compiler.compiler.errorOnHoldViolation=false" >> configuration_basic_if0.tmp.ini
echo "[vivado]" >> configuration_basic_if0.tmp.ini
echo "prop=run.impl_1.strategy=Performance_NetDelay_low" >> configuration_basic_if0.tmp.ini
make -C Basic_kernels/ all DEVICE=xilinx_u280_xdma_201920_3 -j3
make[1]: Entering directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/Basic_kernels'
make[1]: Warning: File 'Makefile' has modification time 30801214 s in the future
/tools/Xilinx/Vitis/2021.1/bin/v++ -t hw --platform xilinx_u280_xdma_201920_3 --save-temps -k krnl_s2mm -c -o _x.xilinx_u280_xdma_201920_3/krnl_s2mm.xo src/krnl_s2mm.cpp
/tools/Xilinx/Vitis/2021.1/bin/v++ -t hw --platform xilinx_u280_xdma_201920_3 --save-temps -k krnl_mm2s -c -o _x.xilinx_u280_xdma_201920_3/krnl_mm2s.xo src/krnl_mm2s.cpp
Option Map File Used: '/tools/Xilinx/Vitis/2021.1/data/vitis/vpp/optMap.xml'
[... 227 lines cut ...]
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Fri May 21 04:14:09 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Fri May 21 04:14:09 2021
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/shouqicao/workspace/xup/xup_vitis_network_example/Basic_kernels/_x/reports/krnl_mm2s/v++_compile_krnl_mm2s_guidance.html', at Fri May 21 04:14:11 2021
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/shouqicao/workspace/xup/xup_vitis_network_example/Basic_kernels/_x/reports/krnl_s2mm/v++_compile_krnl_s2mm_guidance.html', at Fri May 21 04:14:11 2021
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u280_xdma_201920_3/xilinx_u280_xdma_201920_3.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_u280_xdma_201920_3/hw/xilinx_u280_xdma_201920_3.xsa'
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u280_xdma_201920_3/xilinx_u280_xdma_201920_3.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_u280_xdma_201920_3/hw/xilinx_u280_xdma_201920_3.xsa'
INFO: [v++ 74-78] Compiler Version string: 2022.1
INFO: [v++ 74-78] Compiler Version string: 2022.1
ERROR: [v++ 60-1299] The specified platform is not supported. Platform 'xilinx_u280_xdma_201920_3.xpfm' (version 2019.2) is not supported by the current tool version (2022.1). By policy, platforms are supported for the remainder of the calendar year release plus the following calendar year release
ERROR: [v++ 60-592] Failed to finish compilation
ERROR: [v++ 60-1299] The specified platform is not supported. Platform 'xilinx_u280_xdma_201920_3.xpfm' (version 2019.2) is not supported by the current tool version (2022.1). By policy, platforms are supported for the remainder of the calendar year release plus the following calendar year release
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:40: recipe for target '_x.xilinx_u280_xdma_201920_3/krnl_mm2s.xo' failed
make[1]: *** [_x.xilinx_u280_xdma_201920_3/krnl_mm2s.xo] Error 1
make[1]: *** Waiting for unfinished jobs....
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:43: recipe for target '_x.xilinx_u280_xdma_201920_3/krnl_s2mm.xo' failed
make[1]: *** [_x.xilinx_u280_xdma_201920_3/krnl_s2mm.xo] Error 1
make[1]: Leaving directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/Basic_kernels'
Makefile:113: recipe for target 'Basic_kernels/_x.xilinx_u280_xdma_201920_3/krnl_mm2s.xo' failed
make: *** [Basic_kernels/_x.xilinx_u280_xdma_201920_3/krnl_mm2s.xo] Error 2
```

mariodruiz, 2022-05-21:
Hi,

The error is clear 

```bash
Application Exception: The XO file 'NetLayers/_x.xilinx_u280_xdma_201920_3/networklayer.xo' was created in version 2022.1, which is later than the software version you are currently running: 2021.1. Forward compatibility is not supported
```

It appears that you have run the makefile flow first with 2022.1 and thus generating the all the xo files. Then you may have encounter that the platform is not supported in 2022.1 and moved to 2021.1. Hence, hitting this error because the `*.xo` files are generated with a new version of the tool. 

The Makefiles are not smart enough to generate `xo` files for different Vivado/Vitis versions. So, the easiest solution is to get a fresh copy of the repository and use Vitis 2021.1.

Qiang-Flute, 2022-05-23:
thank you. The error about version was solved since I get the fresh copy. But there are some new error about IP.
```
make[2]: Leaving directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/100G-fpga-network-stack-core'
/tools/Xilinx/Vivado/2021.1/bin/vivado -mode batch -source package_netlayer.tcl -notrace -tclargs _x.xilinx_u280_xdma_201920_3/networklayer.xo networklayer xilinx_u280_xdma_201920_3

****** Vivado v2021.1 (64-bit)
  **** SW Build 3247384 on Thu Jun 10 19:36:07 MDT 2021
  **** IP Build 3246043 on Fri Jun 11 00:30:35 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source package_netlayer.tcl -notrace
INFO: xoname-> _x.xilinx_u280_xdma_201920_3/networklayer.xo
      krnl_name-> networklayer
      device-> xilinx_u280_xdma_201920_3

INFO: [BD::TCL 103-2003] Currently there is no design <network_layer_bd> in project, so creating one...
Wrote  : </home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.srcs/sources_1/bd/network_layer_bd/network_layer_bd.bd> 
INFO: [BD::TCL 103-2004] Making design <network_layer_bd> as current_bd_design.
INFO: [BD::TCL 103-2005] Currently the variable <design_name> is equal to "network_layer_bd".
INFO: [BD::TCL 103-2011] Checking if the following IPs exist in the project's IP catalog:  
xilinx.com:ip:axis_register_slice:1.1 xilinx.com:ip:axis_switch:1.1 xilinx.com:hls:ethernet_header_inserter:1.0 xilinx.com:hls:packet_handler:1.0 xilinx.com:ip:smartconnect:1.0 xilinx.com:hls:udp:1.0 xilinx.com:ip:xlconstant:1.1 xilinx.com:hls:arp_server:1.0 xilinx.com:hls:icmp_server:1.0  .
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
WARNING: [Coretcl 2-175] No Catalog IPs found
ERROR: [BD::TCL 103-2012] The following IPs are not found in the IP Catalog:
  xilinx.com:hls:ethernet_header_inserter:1.0 xilinx.com:hls:packet_handler:1.0 xilinx.com:hls:arp_server:1.0 xilinx.com:hls:icmp_server:1.0

Resolution: Please add the repository containing the IP(s) to the project.
INFO: [BD::TCL 103-2020] Checking if the following modules exist in the project's sources:  
axi4stream_sinker bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg interface_settings performance_debug_reg bandwidth_reg bandwidth_reg bandwidth_reg bandwidth_reg  .
WARNING: [BD::TCL 103-2023] Will not continue with creation of design due to the error(s) above.
Wrote  : </home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.srcs/sources_1/bd/network_layer_bd/network_layer_bd.bd> 
VHDL Output written to : /home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd/synth/network_layer_bd.v
VHDL Output written to : /home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd/sim/network_layer_bd.v
VHDL Output written to : /home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd/hdl/network_layer_bd_wrapper.v
Exporting to file /home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd/hw_handoff/network_layer_bd.hwh
Generated Block Design Tcl file /home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd/hw_handoff/network_layer_bd_bd.tcl
Generated Hardware Definition File /home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/tmp_networklayer_xilinx_u280_xdma_201920_3/kernel_pack.gen/sources_1/bd/network_layer_bd/synth/network_layer_bd.hwdef
INFO: [BD 41-1637] Generated targets are already up-to-date for block design 'network_layer_bd' - hence not re-generating.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/6to3_reducer.vhd'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/counter_64_7_v3.vhd'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/bandwidth_reg.v'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/axi4stream_constant.v'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/axi4stream_sinker.v'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/interface_settings.v'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers/src/performance_debug_reg.v'.
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_clk' of definition 'xilinx.com:signal:clock:1.0' (from X_INTERFACE_INFO parameter from HDL file).
[... cut ...]
    (file "package_cmac.tcl" line 73)
INFO: [Common 17-206] Exiting Vivado at Sun May 23 00:04:14 2021...
Makefile:52: recipe for target '_x.xilinx_u280_xdma_201920_3/cmac_0.xo' failed
make[1]: *** [_x.xilinx_u280_xdma_201920_3/cmac_0.xo] Error 1
make[1]: Leaving directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/Ethernet'
Makefile:119: recipe for target 'Ethernet/_x.xilinx_u280_xdma_201920_3/cmac_0.xo' failed
make: *** [Ethernet/_x.xilinx_u280_xdma_201920_3/cmac_0.xo] Error 2
```

mariodruiz, 2022-05-23:
Hi,

You need a license to use the CMAC

```bash
WARNING: [IP_Flow 19-2162] IP 'cmac_bd_cmac_uplus_0_0' is locked:
* IP 'cmac_bd_cmac_uplus_0_0' requires one or more mandatory licenses but no valid licenses were found. However license checkpoints may prevent use of this IP in some tool flows.
```

How to get a license is described here: https://github.com/Xilinx/xup_vitis_network_example/blob/master/Ethernet/README.md#license

Qiang-Flute, 2022-05-24:
```
shouqicao@tensor-r7201:~/.Xilinx$ ls
9_Vivado_license_2037.lic  registry  Vitis      Vivado      xinstall
Common                     v++       vitis_hls  Xilinx.lic
```
I got the Xilinx.lic license for this example, and I load the license by vivado2021.1. I also check the directory .Xilinx, the license copy to this directory correctly. But when I try to remake the example, the IP&License error still occur.
```
INFO: [Common 17-206] Exiting Vivado at Mon May 24 00:06:59 2021...
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Leaving directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/NetLayers'
make -C Ethernet/ all DEVICE=xilinx_u280_xdma_201920_3 INTERFACE=3
make[1]: Entering directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/Ethernet'
make[1]: Warning: File 'Makefile' has modification time 31595023 s in the future
mkdir -p _x.xilinx_u280_xdma_201920_3
./duplicate_patch.sh
/tools/Xilinx/Vivado/2021.1/bin/vivado -mode batch -source package_cmac.tcl -notrace -tclargs _x.xilinx_u280_xdma_201920_3/cmac_0.xo cmac_0 xilinx_u280_xdma_201920_3 0

****** Vivado v2021.1 (64-bit)
  **** SW Build 3247384 on Thu Jun 10 19:36:07 MDT 2021
  **** IP Build 3246043 on Fri Jun 11 00:30:35 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source package_cmac.tcl -notrace
INFO: xoname-> _x.xilinx_u280_xdma_201920_3/cmac_0.xo
      krnl_name-> cmac_0
      device-> xilinx_u280_xdma_201920_3
      interface-> 0
Wrote  : </home/shouqicao/workspace/xup/xup_vitis_network_example/Ethernet/tmp_cmac_0_xilinx_u280_xdma_201920_3/kernel_pack.srcs/sources_1/bd/cmac_bd/cmac_bd.bd> 
Generating IPI for cmac_uplus_0 with GT clock running at 156250000 Hz
INFO: [Device 21-403] Loading part xcu280-fsvh2892-2L-e
WARNING: [IP_Flow 19-2162] IP 'cmac_bd_cmac_uplus_0_0' is locked:
* IP 'cmac_bd_cmac_uplus_0_0' requires one or more mandatory licenses but no valid licenses were found. However license checkpoints may prevent use of this IP in some tool flows.
ERROR: [IP_Flow 19-2322] Failed to set user parameter 'gt_txusrclk2_out_port.CLK_DOMAIN' value 'cmac_bd_cmac_uplus_0_0_gt_txusrclk2' for IP 'cmac_uplus_0'.
ERROR: [BD 41-145] Parameter gt_txusrclk2_out_port.CLK_DOMAIN not found on block cmac_uplus_0 
ERROR: [IP_Flow 19-2322] Failed to set user parameter 'gt_ref_clk_out_port.CLK_DOMAIN' value 'cmac_bd_cmac_uplus_0_0_gt_ref_clk_out' for IP 'cmac_uplus_0'.
ERROR: [BD 41-145] Parameter gt_ref_clk_out_port.CLK_DOMAIN not found on block cmac_uplus_0 
ERROR: [IP_Flow 19-2322] Failed to set user parameter 'gt_rxrecclkout_port.CLK_DOMAIN' value 'cmac_bd_cmac_uplus_0_0_gt_rxrecclkout' for IP 'cmac_uplus_0'.
ERROR: [BD 41-145] Parameter gt_rxrecclkout_port.CLK_DOMAIN not found on block cmac_uplus_0 
ERROR: [IP_Flow 19-2322] Failed to set user parameter 'gt_rxusrclk2_out_port.CLK_DOMAIN' value 'cmac_bd_cmac_uplus_0_0_gt_rxusrclk2' for IP 'cmac_uplus_0'.
ERROR: [BD 41-145] Parameter gt_rxusrclk2_out_port.CLK_DOMAIN not found on block cmac_uplus_0 
create_bd_cell: Time (s): cpu = 00:00:10 ; elapsed = 00:00:12 . Memory (MB): peak = 3200.422 ; gain = 552.363 ; free physical = 165025 ; free virtual = 578272
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv xilinx.com:ip:cmac_usplus ${cmac_name}"
    invoked from within
"set cmac_ip [create_bd_cell -type ip -vlnv xilinx.com:ip:cmac_usplus ${cmac_name}]"
    (file "/home/shouqicao/workspace/xup/xup_vitis_network_example/Ethernet/bd_cmac.tcl" line 120)

    while executing
"source ${root_dir}/bd_cmac.tcl"
    (file "package_cmac.tcl" line 73)
INFO: [Common 17-206] Exiting Vivado at Mon May 24 00:07:28 2021...
Makefile:52: recipe for target '_x.xilinx_u280_xdma_201920_3/cmac_0.xo' failed
make[1]: *** [_x.xilinx_u280_xdma_201920_3/cmac_0.xo] Error 1
make[1]: Leaving directory '/home/shouqicao/workspace/xup/xup_vitis_network_example/Ethernet'
Makefile:120: recipe for target 'Ethernet/_x.xilinx_u280_xdma_201920_3/cmac_0.xo' failed
```

mariodruiz, 2022-05-24:
Did you check that the host ID matches? 
[image]

Qiang-Flute, 2022-05-24:
I check the host ID, Host IDs match table show Yes.

mariodruiz, 2022-05-24:
Ok, that rules ok the license. I am not able to reproduce the error myself so a few things to consider.

1. 
```bash
make[1]: Warning: File 'Makefile' has modification time 31595023 s in the future
```
This concerns me. Did you change the system date instead of applying the patch I mentioned here? https://github.com/Xilinx/xup_vitis_network_example/issues/62#issuecomment-1132749471

2.

By any chance your locale are not in English? https://www.tecmint.com/set-system-locales-in-linux/

It would be good if you can share the output of  `locale`

3. After installing the license, did you open a new terminal to rebuild the design? Perhaps, the old state is cached

HFTrader, 2022-05-24:
31595023 is almost exactly one year, I just wanted to point this out.

mariodruiz, 2022-06-17:
@Qiang-Flute, did you make any progress?

mariodruiz, 2022-07-29:
Closing this due to inactivity.