Xilinx/xup_vitis_network_example #66: VLNV Warning

victornvq30, 2022-05-26:
Hi Everyone,
While building CMAC IP with a command:
make all DEVICE=xilinx_u50_gen3x16_xdma_201920_3 INTERFACE=0
all went though ok except that I got the following warning:

WARNING: [Vivado 12-12407] VLNV in kernel.xml does not match VLNV in any of the IPs specified with the ip_directory option: xilinx.com:kernel:cmac_0:1.0

What we was find out that is related to string in kernel_0.xml that is

 <kernel name="cmac_0" language="ip_c" vlnv="xilinx.com:kernel:cmac_0:1.0" attributes="" preferredWorkGroupSizeMultiple="0" workGroupSize="1" interrupt="false" hwControlProtocol="ap_ctrl_none">

Our question: how seriously the warning is and how to fix it ?
Thank you !

zhiyisun, 2022-05-31:
Hello @victornvq30 , 

I think the above warning can be ignored. Your building command is generating cmac IP with VLNV "xilix.com:kernel:cmac_0:1.0". This new cmac IP you generated can't be found from Vivado/Vitis's default IP directory. That's expected.  Because we didn't set the current directory into the Vivado/Vitis's IP directory. After all IP generated, in our case, those .xo files defined in Makefile $LIST_XO, we use the relative path (e.g. $(CMACDIR)$(TEMP_DIR)/cmac_0.xo) to find that xo file and generate the xclbin eventually.

victornvq30, 2022-05-31:
Hi zhiyi,
That is clear, thank you !

mariodruiz, 2022-05-31:
This warning can be safely ignored.

closed 2022-05-31 by mariodruiz

victornvq30, 2022-05-31:
I got it thank you !