Xilinx/xup_vitis_network_example #67: Improve build time

victornvq30, 2022-05-26:
Hi Everyone,
I'm experienced that build the .xo files doesn't take long, however the linking process to create xclbin file takes quite long. For etc
[root@fpga xup_vitis_network_example]# /tools/Xilinx/Vitis/2021.2/bin/v++ -t hw --platform xilinx_u50_gen3x16_xdma_201920_3 --save-temps --config configuration_basic_if0.tmp.ini --temp_dir ./basic.intf0.xilinx_u50_gen3x16_xdma_201920_3 -l -o'basic.intf0.xilinx_u50_gen3x16_xdma_201920_3/vnx_basic_if0_260522.xclbin' NetLayers/_x.xilinx_u50_gen3x16_xdma_201920_3/networklayer.xo Ethernet/_x.xilinx_u50_gen3x16_xdma_201920_3/cmac_0.xo Basic_kernels/_x.xilinx_u50_gen3x16_xdma_201920_3/krnl_mm2s.xo Basic_kernels/_x.xilinx_u50_gen3x16_xdma_201920_3/krnl_s2mm.xo  --user_ip_repo_paths /root/xup_vitis_network_example/NetLayers/100G-fpga-network-stack-core/synthesis_results_HMB -j 8

it takes about **2 hours to complete the vnx_basic_if0.xclbin file**

My question is how to improve this build time ? because will make changes the NetLayers frequences while developing a new IP kernel to handle network traffic.
My hardware is Alveo U50, Dell server Dual Silver Xeon with 16 cores/ 32 threads, 64 GB RAM running on CentOS 7.9.2009, kernel 3.10.0-1160.el7

Thank you for your hint !
Best Regards,

zhiyisun, 2022-05-31:
Hello @victornvq30 ,

You probably can try  "--hls.jobs" (https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/hls-Options. ) to specify the number of parallel jobs. But basically, it will take time for FPGA xclbin image generation. Vitis/Vivado provides simulation to help developers to speed up their development. (https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Vitis-Application-Development-Flow) You can use software emulation and hardware emulation to debug/verify your code as much as possible before building the FPGA image. With that, developers can fix most of function (and even performance) issue before running the program on hardware. I recommend to read the Vitis Unified Software Platform Documentation: Application Acceleration Development (UG1393) for more info. 

Hope it helps.

Regards,
Zhiyi

victornvq30, 2022-05-31:
Hi Zhiyi,
We will try that way your recommended. Thank you !

mariodruiz, 2022-05-31:
Hi @victornvq30,

Even using more jobs as @zhiyisun suggested won't speed up much the generation of the `xclbin`. Vivado still has to place and route the design.

What changes do you need to do? Each IP in the network layer has it own HLS testbench, if you can simulate at that level the development process should be much faster. Alternatively, you can do hardware emulation and use the Python code for this as well.

victornvq30, 2022-05-31:
Hi Mario, 
As your suggestion, I understand that we have to prepare data for simulation seriously. **Our project is to analyzing the package data from Telecom network, modify the packages, send to host**, which we are re-writing/modifying the NetLayer kernels. Thus we are going to dump the raw network data into PCAP file. Subsequently we will dump raw package from PCAP file for NetLayer kernels when running the simulation.
Please advise, if the way that we make data for simulation is correct ? or may you have a better way to suggest us ?
Thanks and Best Regards,

victornvq30, 2022-06-01:
Hi Mario,

We went through the OpenNIC documentation you suggested. As for conclusion, yes, this could be more suitable for our project as we  need only for development a plugin for box 250 MHz, as it is presented herein 
https://github.com/Xilinx/open-nic-shell/tree/main/plugin/p2p

The only problem is that the sample programmed System Verilog, while we are not family with this as we prefer using vitis HLS with C/C++.

Would you give us a guideline or simple example for this type of plugin using Vitis HLS C/C++ ?

Thanks you and best regards,
Quang Nguyen

mariodruiz, 2022-06-01:
One alternative you can try is the project NetFPGA https://netfpga.org/NetFPGA-PLUS.html, there are a few example designs there. 

You can take inspiration from the HLS code used for the network layer. https://github.com/hpcn-uam/100G-fpga-network-stack-core/tree/vnx-vitis-hls/hls

I would suggest you ask your question in the OpenNIC repository for suggestion how to write the plug in boxes.  An alternative is that you use P4 for your plug-ins. For this please contact your FAE for more information.

closed 2022-06-01 by mariodruiz