Xilinx/xup_vitis_network_example #71: 'xrt_core::error'  what():  No IP matching 'cmac_0': Invalid argument while running 'basic' design through C++ host code

SPala121, 2022-06-09:
I am trying to run 'basic' design of xup_vitis_network_ example IP through C++ host code (from branch host_xrt). 

./rx vnx_basic_if0.xclbin

 I am getting following error:

Device created: xilinx_u280_xdma_201920_3
Bitfile loaded 02f89ce5-f48a-e281-ebef-bc274cf47d97
terminate called after throwing an instance of 'xrt_core::error'
  what():  No IP matching 'cmac_0': Invalid argument
Aborted (core dumped)

Build summary of IP is attached:
[vnx_basic_if0.xclbin.txt](https://github.com/Xilinx/xup_vitis_network_example/files/8869598/vnx_basic_if0.xclbin.txt)

mariodruiz, 2022-06-09:
@fpgafais

fpgafais, 2022-06-17:
Hi @SPala121,
I have just tried to run host_xrt on a single link vnx_basic_if0.xclbin and works fine. CMAC_0 is exactly the instance that should be there, also looking at your .txt file. When running  Please make sure you are using latest host_xrt (I see you dind't specify the Allveo ID when running ./rx) and that the xclbin is exactly the one with network cores on link 0.

SPala121, 2022-06-20:
Hi @fpgafais ,
I ran the command with ALVEO ID(but I forgot to mention in the command)
./rx /home/settysha/vnx_basic_if0.xclbin  xilinx_u280_xdma_201920_3

Also. I have latest host_xrt and xclbin has network cores on link 0
But still I am facing same issue.

mariodruiz, 2022-06-20:
@SPala121 the second argument is an unsigned integer 

https://github.com/Xilinx/xup_vitis_network_example/blob/211300afd384808c29d3a02c41a174309c27eee6/host_xrt/exec/alveo_rx.cpp#L33-L37

You should run it like this
```sh
./rx /home/settysha/vnx_basic_if0.xclbin 0
```

SPala121, 2022-06-20:
Hi @mariodruiz 

I ran this command as well. But it throws same error.

chenb1981, 2022-06-23:
56line :   auto l1 = AlveoVnxLink(u280, 1);       if a single link, I think 2nd param should be 0

SPala121, 2022-06-23:
@fpgafais here is my alveo_rx.cpp. Actually I have set it to 0.

[alveo_rx.txt](https://github.com/Xilinx/xup_vitis_network_example/files/8965242/alveo_rx.txt)

mariodruiz, 2022-06-23:
@SPala121 did you verify the xclbin with PYNQ?

mariodruiz, 2022-06-23:
Did you use any of the existing notebook? Or created your own? It seems that you are not importing the vnx_utils

SPala121, 2022-06-23:
@mariodruiz oh my bad. I created my own notebook and didnot import vnxutils.
But getting error while populating socket table.

kernel.networklayer_0.populateSocketTable()
 "Socket list length 16 is bigger than the      number of sockets in hardware 0"

Is this a separate issue? Or am I missing something?

mariodruiz, 2022-06-23:
Please check this issue https://github.com/Xilinx/xup_vitis_network_example/issues/73#issue-1269063908

SPala121, 2022-06-23:
@fpgafais I checked the status of compute units on U280 with the following command:
sudo xbutil examine -d 0000:81:00.1

This is the output:
Xclbin UUID
  AADDFE86-33C9-B5AB-5E45-71E8890FDF80

Compute Units
  PL Compute Units
    Index   Name                                              Base_Address    Usage   Status  
    0       cmac_0:cmac_0                                     0x1800000       0       (--)    
    1       krnl_mm2s:krnl_mm2s_0                             0x1810000       0       (IDLE)  
    2       krnl_s2mm:krnl_s2mm_0                             0x1820000       0       (IDLE)  
    3       networklayer:networklayer_0                       0x1830000       0       (--) 

We can see cmac_0 present in the FPGA. But the 'xrt_core::error' still there.

fpgafais, 2022-06-23:
@SPala121 how many boards you have in your system? Could it be that the bitfile is loaded to a different card?

SPala121, 2022-06-27:
@fpgafais I have only one board connected to the host

SPala121, 2022-07-05:
@fpgafais @mariodruiz 
Issue of 'xrt_core::error' got resolved! Basically, we should give IP name as 'cmac_0:{cmac_0}'. In the code it was considering only 'cmac_0'.

closed 2022-07-05 by SPala121