Xilinx/xup_vitis_network_example #78: Netlayer customization

byeongkeonLee, 2022-07-19:
### Build Issues

1. OS version, e.g. `lsb_release -a`
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

2. Vitis version `vitis -version`
    Vitis v2021.2 (64-bit) + Y2K22 is applied

3. XRT version `xbutil version`
Version              : 2.12.427
Branch               : 2021.2
XOCL                 : 2.12.427
XCLMGMT              : 2.12.427

---
Hi, Mario

This is interesting work and I am trying to learn and modify Netlayer to customize. 

#48  is a related question, but I can not get enough information so I leave an issue here.

First, I removed ARP and ICMP modules since I want to start from one simple module: the UDP module.
I'm familiar with C++, so I can easily get rid of the source code of some modules. 
(There are only udp.cpp, udp.hpp, common_utilities.cpp, common_utilities.hpp with additional data structure)

However, in the synthesis step, an error occurs and it might seem due to tcl files that are not familiar. 

Could you give me some hints on how to modify tcl file to link IP correctly?

Error message is following:

```
===>The following messages were generated while  Compiling (synthesis checkpoint) kernel/IP: ulp_networklayer_0_0 Log file: /home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.runs/ulp_networklayer_0_0_synth_1/runme.log  :
ERROR: [VPL 8-448] named port connection 'ap_clk' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:99]
ERROR: [VPL 8-448] named port connection 'ap_rst_n' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:100]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tvalid' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:102]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:103]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:104]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:105]
ERROR: [VPL 8-448] named port connection 'S_AXIS_eth2nl_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:106]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tvalid' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:108]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:109]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:110]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:111]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2eth_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:112]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:114]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:115]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:116]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:117]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tdest' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:118]
ERROR: [VPL 8-448] named port connection 'S_AXIS_sk2nl_tvalid' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:119]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2sk_tdata' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:121]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2sk_tkeep' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:122]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2sk_tlast' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:123]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2sk_tready' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' [/home/bklee/tutorial/re/test_xup_vitis_network_example/basic.intf0.xilinx_u200_gen3x16_xdma_1_202110_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/679f/src/networklayer.v:124]
ERROR: [VPL 8-448] named port connection 'M_AXIS_nl2sk_tdest' does not exist for instance 'network_layer_bd_i' of module 'network_layer_bd' .....

```

Thanks in advance!

mariodruiz, 2022-07-19:
Hi,

You need to modify the NetworkLayer kernel to remove these modules. The TCL file is doing this.
The networklayer kernel is an RTL kernel, which is an advance flow. 
It is not necessary not to compile the rest of the HLS IP. 

You can find more information here:

https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/RTL-Kernels
https://docs.xilinx.com/v/u/2019.2-English/ug835-vivado-tcl-commands

You will need to get familiar with Vivado IP as well if you want to modify this kernel.

Mario

byeongkeonLee, 2022-07-19:
Oh, I successfully generate the tcl file with vivado IDE, and the synthesis step has been passed.

I just found out that the network_bd.tcl file can be modified with vivado IDE easily. 

Further, your Xilinx documents are also helpful. 

Thank you for your kindness!

closed 2022-07-19 by byeongkeonLee