Xilinx/xup_vitis_network_example #80: Does this repository support QDMA?

byeongkeonLee, 2022-07-28:
Hi, Mario. I always thank you for your kind reply.

I have a question about the QDMA of this project.

[image]

In this figure, there is a word {X | Q}DMA for communication between PCIe and FPGA global memory. 

However, it seems that there is no official information about QDMA in this repository.

Could you let me know whether I can apply QDMA to this project, and if possible, how? (brief hints are appreciated!)

Best regards,

mariodruiz, 2022-07-28:
Hi,

Good catch. I need to update the image.
QDMA platforms are not longer available. 

Mario

closed 2022-07-28 by mariodruiz

byeongkeonLee, 2022-07-28:
Thank you for the quick reply, Mario!

I have one more question.

If you would not mind, I would like to know the brief reason QDMA is not available anymore.
(Because I'm still confused it is impossible due to a reason such as no Xilinx support)

Thank you!

mariodruiz, 2022-07-28:
Well, the platforms are not supported anymore and they never were publicly available anyways. 
New features in the XDMA shells such as Host Memory Access are a good replacement for QDMA.

byeongkeonLee, 2022-07-28:
I see!  Thank you!