Xilinx/xup_vitis_network_example #83: Failed generating target for U50 while creating FPGA bitstream

fujimotokua, 2022-08-08:
Hello,

Currently, I'm trying to build VNx on Fedora36, but I see the below errors during the build process.
I installed Vitis v2022.1, xilinx_u50_gen3x16_xdma_5_202210_1 and UltraScale+ Integrated 100G Ethernet Subsystem 

I'm not sure how to debug from here with the below error messages. 
Would you be able to point out what to do next?

## Environment 

### The result of lsb_release
```
LSB Version:    :core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID: Fedora
Description:    Fedora release 36 (Thirty Six)
Release:        36
Codename:       ThirtySix
```

### Vitis Version
```
Autocomplete enabled for the xbutil command
Autocomplete enabled for the xbmgmt command
XILINX_XRT        : /opt/xilinx/xrt
PATH              : /opt/xilinx/xrt/bin:/home/.local/bin:/home/bin:/usr/lib64/qt3.3/bin:/usr/lib64/ccache:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin
LD_LIBRARY_PATH   : /opt/xilinx/xrt/lib
PYTHONPATH        : /opt/xilinx/xrt/python

****** Xilinx Vitis Development Environment
****** Vitis v2022.1 (64-bit)
  **** SW Build 3524922 on 2022-04-14-18:00:18
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
```

### Execute Command
```
make all DEVICE=/opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_5_202210_1/xilinx_u50_gen3x16_xdma_5_202210_1.xpfm INTERFACE=0 DESIGN=basic
```

### Error Message 
```
ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified
ERROR: [VPL 19-3477] Update of parameter 'PARAM_VALUE.CHANNEL_ENABLE' failed for IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt'. ERROR: [Common 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified

ERROR: [VPL 19-3428] Failed to create Customization object ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt
ERROR: [VPL 17-70] Application Exception: Failed to create subcore IP 'cmac_bd_cmac_uplus_0_0_gt'. Create IP failed with errors
ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs:
ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified
ERROR: [VPL 19-3477] Update of parameter 'PARAM_VALUE.CHANNEL_ENABLE' failed for IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt'. ERROR: [Common 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified

ERROR: [VPL 19-3428] Failed to create Customization object ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt
ERROR: [VPL 17-70] Application Exception: Failed to create subcore IP 'cmac_bd_cmac_uplus_0_0_gt'. Create IP failed with errors
ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs:
ERROR: [VPL 19-98] Generation of the IP CORE failed.
Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs:
Could not configure the upgraded 'IP cmac_bd_cmac_uplus_0_0'.
If the problem persists, please contact Xilinx support.
Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs:
ERROR: [VPL 41-1030] Generation failed for the IP Integrator block cmac_0
ERROR: [VPL 60-773] In '/home/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/runme.log', caught Tcl error:  ERROR: [Common 17-39] 'generate_target' failed due to earlier errors.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, Step failed: generate_target An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/home/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [17:01:55] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:15 ; elapsed = 00:02:34 . Memory (MB): peak = 2417.367 ; gain = 0.000 ; free physical = 82788 ; free virtual = 102337
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
```

mariodruiz, 2022-08-08:
Hi @fujimotokua,

Fedora isn't a OS that is supported by Vitis https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Installation-Requirements. Therefore, anything can be happening.

Please, try one on one of the supported OS.

Mario

yangxiaomaomao, 2022-08-26:
@mariodruiz hi，I encounter almost the same question as @fujimotokua when I build the kernel as Readme says, the same vitis version, the same Alveo card U50, the same execute command, and the same error. I have also get license of  UltraScale+ Integrated 100G Ethernet Subsystem. You say it is a OS issue, but the following is my lsb_release -a result

[image]

Though Ubuntu 20.04.4 LTS is not included in the supported OS list, Ubuntu 20.04.3/2/1 is all included, I wonder whether it is a OS version issue or other reasons and what to do next, hope your reply. 

Thanks!

mariodruiz, 2022-08-26:
@yangxiaomaomao 

Please open a new issue providing as much details as you can.

Mario

closed 2024-08-01 by mariodruiz