Xilinx/xup_vitis_network_example #84: Failed to generate bitstream after passing synthesis and implementation [Vivado 2021.1, Alveo U250]

charles-typ, 2022-08-23:
### Build Issues

1. OS version, e.g. `lsb_release -a`
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal
1. Vitis version `vitis -version`
   1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
   Version is 2021.1, the patch is applied
1. XRT version `xbutil version`
2.12.427 branch 2021.2

Was able to finish synthesis and implementation, but gets the following error when I tried to generate the bitstream.
```
[Common 17-69] Command failed: This design contains one or more cells for which bitstream generation is not permitted:
bd_wrapper_i/bd_i/Interfaces/cmac_wrapper_0/inst/cmac_connector_i/cmac_wrapper_i/cmac_0/inst/i_cmac_uplus_0_top (<encrypted cellview>)
If a new IP Core license was added, in order for the new license to be picked up, the current netlist needs to be updated by resetting and re-generating the IP output products before bitstream generation
```
Seems to be an IP issue, but I was able to setup the IP (which is also used in synthesis and implementation I suppose).
[image]
[image]

mariodruiz, 2022-08-23:
Hi @charles-typ,

If you installed the license of the CMAC after the first run, you need to regenerate the cmac kernel.
Remove any *.xo file inside the `Ethernet` directory and run make again.

charles-typ, 2022-09-03:
Hi @mariodruiz,

Thanks for the help. I did regenerate the cmac kernel multiple times but still got the same error.

The solution that worked for me was to use the latest version of Vivado instead (2022.01), closing this issue.

closed 2022-09-03 by charles-typ