Xilinx/xup_vitis_network_example #85: W

Sarika124, 2022-08-24:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

**Usign Vitis 2021.2 or older? Make sure the [Y2K22 patch is applied](
https://support.xilinx.com/s/article/76960?language=en_US)**

If you still want to raise an issue here, please give us as much detail as
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version, e.g. `lsb_release -a`
1. Vitis version `vitis -version`
   1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
1. XRT version `xbutil version`

### Run Time Issues

1. OS version `lsb_release -a`
1. XRT version `xbutil version`
1. pynq version `pynq version`
1. JupyterLab and Dask version if applicable

mariodruiz, 2022-08-25:
Yes, but you will have to adapt the notebooks to do this.

Sarika124, 2022-08-30:
Hi,
For that need to edit anything in design? 
I tried this same design but getting error like this.
[image]

Notebooks: 
[image]

mariodruiz, 2022-08-30:
If you are using the design with the two interfaces, you do not need to change anything. 

For the platform you are using you have to make sure the second partition of the shell is programmed as described here https://support.xilinx.com/s/article/75975?language=en_US. This is also indicated here https://github.com/Xilinx/xup_vitis_network_example#alveo-cards

You need to make sure that `xbutil validate` works, otherwise VNx will not work

Sarika124, 2022-09-14:
Hi,
I had issue in the xbutil validate and able to validate the card successfully.  Now above issue rectified. Thank you for the guidance.

Now by checking the physical link, it shows "cmac_link - false ". Connected using 100G cable one QSFP port to second QSF port of same U250 machine. Is there any method to verify the physical connection given was proper or not?

[image]

Sarika124, 2022-09-14:
Is it possible to see FPGA  NIC interface in FPGA plugged server through ifconfig???

Sarika124, 2022-09-16:
just for clarification, We are using100G QSFP28 Active Optical Cable. Is this the right interconnect to use??

[image]

mariodruiz, 2022-09-16:
It should work, but I have not tested that particular cable. Did you verify the cable with something else?

I typically use Direct Attached Copper cables.

Sarika124, 2022-09-16:
No.We didn't verified the cable with any other experiments.

mariodruiz, 2022-10-17:
I think you should start by making sure the cable works.

rajeshkrmeena, 2023-01-05:
> **For Vivado questions, please use [Vivado forum](https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**
> 
> **For Vitis questions, please use [the Vitis forum](https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**
> 
> **For Vitis questions, please use [the Vitis forum](https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**
> 
> **For pynq questions, please use [the PYNQ discussion forum](https://discuss.pynq.io/).**
> 
> **Usign Vitis 2021.2 or older? Make sure the [Y2K22 patch is applied](https://support.xilinx.com/s/article/76960?language=en_US)**
> 
> If you still want to raise an issue here, please give us as much detail as possible to the issue you are seeing. We have listed some helpful fields below.
> 
> * Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.
> 
> ### Build Issues
> 1. OS version, e.g. `lsb_release -a`
> 2. Vitis version `vitis -version`
>    
>    1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
> 3. XRT version `xbutil version`
> 
> ### Run Time Issues
> 1. OS version `lsb_release -a`
> 2. XRT version `xbutil version`
> 3. pynq version `pynq version`
> 4. JupyterLab and Dask version if applicable

Yes... you have to connect the ports using a QSFP cable.

mariodruiz, 2024-04-27:
Closing at no activity has been seen in a long time.

closed 2024-04-27 by mariodruiz