Xilinx/xup_vitis_network_example #86: Fail to create FPGA bitstream

yangxiaomaomao, 2022-08-26:
When I build the VNx as the README.md says, I encounter the following problems. I want to know the reason and the way to solve it.

### Build Issues
1. change into the directory `xup_vitis_network_example` and input the following command, 
`make all DEVICE=xilinx_u50_gen3x16_xdma_5_202210_1 INTERFACE=0 DESIGN=basic`after a few minutes, it reports the `ERROR` as follows:
```
****** vpl v2022.1 (64-bit)
  **** SW Build 3524075 on 2022-04-13-17:42:45
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/yangxiaomao/fpga/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/int/kernel_info.dat'.
INFO: [VPL 74-78] Compiler Version string: 2022.1
INFO: [VPL 60-423]   Target device: xilinx_u50_gen3x16_xdma_5_202210_1
INFO: [VPL 60-1032] Extracting hardware platform to /home/yangxiaomao/fpga/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/.local/hw_platform
[18:52:59] Run vpl: Step create_project: Started
Creating Vivado project.
[18:53:04] Run vpl: Step create_project: Completed
[18:53:04] Run vpl: Step create_bd: Started
[18:53:38] Run vpl: Step create_bd: Completed
[18:53:38] Run vpl: Step update_bd: Started
[18:53:39] Run vpl: Step update_bd: Completed
[18:53:39] Run vpl: Step generate_target: Started
[18:54:54] Run vpl: Step generate_target: Failed
[18:54:55] Run vpl: FINISHED. Run Status: generate_target ERROR

===>The following messages were generated while  creating FPGA bitstream. Log file: /home/yangxiaomao/fpga/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/runme.log :
ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified
ERROR: [VPL 19-3477] Update of parameter 'PARAM_VALUE.CHANNEL_ENABLE' failed for IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt'. ERROR: [Common 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified

ERROR: [VPL 19-3428] Failed to create Customization object ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt
ERROR: [VPL 17-70] Application Exception: Failed to create subcore IP 'cmac_bd_cmac_uplus_0_0_gt'. Create IP failed with errors
ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified
ERROR: [VPL 19-3477] Update of parameter 'PARAM_VALUE.CHANNEL_ENABLE' failed for IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt'. ERROR: [Common 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified

ERROR: [VPL 19-3428] Failed to create Customization object ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt
ERROR: [VPL 17-70] Application Exception: Failed to create subcore IP 'cmac_bd_cmac_uplus_0_0_gt'. Create IP failed with errors
ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
ERROR: [VPL 19-98] Generation of the IP CORE failed.
Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
Could not configure the upgraded 'IP cmac_bd_cmac_uplus_0_0'.
If the problem persists, please contact Xilinx support.
Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
ERROR: [VPL 41-1030] Generation failed for the IP Integrator block cmac_0 
ERROR: [VPL 60-773] In '/home/yangxiaomao/fpga/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/vivado.log', caught Tcl error:  ERROR: [Common 17-39] 'generate_target' failed due to earlier errors.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, Step failed: generate_target An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/home/yangxiaomao/fpga/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [18:54:55] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:13 ; elapsed = 00:02:13 . Memory (MB): peak = 2212.250 ; gain = 0.000 ; free physical = 21108 ; free virtual = 78169
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:106: basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/vnx_basic_if0.xclbin] Error 1
```

### Software Version
1. OS version, the result of  `lsb_release -a` is as follows:
```
LSB Version:    core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.4 LTS
Release:        20.04
Codename:       focal
```
2. Vitis version. The result of `vitis -version` is as follows:
```
****** Xilinx Vitis Development Environment
****** Vitis v2022.1 (64-bit)
  **** SW Build 3524922 on 2022-04-14-18:00:18
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
```
3. XRT version. The result of  `xbutil version` is as follows:
```
Version              : 2.13.466
Branch               : 2022.1
Hash                 : f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
Hash Date            : 2022-04-14 17:43:11
XOCL                 : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
XCLMGMT              : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
```
4. Other Versions. The result of `xbutil examine` is as follows:
```
System Configuration
  OS Name              : Linux
  Release              : 5.15.0-46-generic
  Version              : #49~20.04.1-Ubuntu SMP Thu Aug 4 19:15:44 UTC 2022
  Machine              : x86_64
  CPU Cores            : 48
  Memory               : 64050 MB
  Distribution         : Ubuntu 20.04.4 LTS
  GLIBC                : 2.31
  Model                : X640 G30

XRT
  Version              : 2.13.466
  Branch               : 2022.1
  Hash                 : f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
  Hash Date            : 2022-04-14 17:43:11
  XOCL                 : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776
  XCLMGMT              : 2.13.466, f5505e402c2ca1ffe45eb6d3a9399b23a0dc8776

Devices present
BDF             :  Shell                           Platform UUID                         Device ID       Device Ready*  
[0000:8a:00.1]  :  xilinx_u50_gen3x16_xdma_base_5  44654095-25B4-C06A-EC6D-0B479D3FEBE8  user(inst=134)  Yes            
[0000:89:00.1]  :  xilinx_u50_gen3x16_xdma_base_5  44654095-25B4-C06A-EC6D-0B479D3FEBE8  user(inst=133)  Yes            
[0000:88:00.1]  :  xilinx_u50_gen3x16_xdma_base_5  44654095-25B4-C06A-EC6D-0B479D3FEBE8  user(inst=128)  Yes            

* Devices that are not ready will have reduced functionality when using XRT tools
```
5. Additionally, I have a valid UltraScale+ Integrated 100G Ethernet Subsystem license set up in Vivado. The GUI of vlm is as follows:
[image]

Hope your reply, thanks!

mariodruiz, 2022-08-26:
Are you locale variables set to English?

yangxiaomaomao, 2022-08-26:
> Are you locale variables set to English?

Seems it is, the result of `echo $LANG` is as follows:
`en_US.UTF-8`
the result of `locale` is as follows:
```
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=zh_CN.UTF-8
LC_TIME=zh_CN.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=zh_CN.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=zh_CN.UTF-8
LC_NAME=zh_CN.UTF-8
LC_ADDRESS=zh_CN.UTF-8
LC_TELEPHONE=zh_CN.UTF-8
LC_MEASUREMENT=zh_CN.UTF-8
LC_IDENTIFICATION=zh_CN.UTF-8
LC_ALL=
```

mariodruiz, 2022-08-26:
All of them should be set to English, in particular the numeric ones. Decimal comma and thousand separators can make things not to work. 

When you change the locale, you should regenerate everything from scratch to be safe.

yangxiaomaomao, 2022-08-26:
> All of them should be set to English, in particular the numeric ones. Decimal comma and thousand separators can make things not to work.
> 
> When you change the locale, you should regenerate everything from scratch to be safe.

I have change them all to English, the result of `locale` is as follows:
```
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```
and then I run the command `make distcleanall` to remove the previous generated file, and run the command `make all DEVICE=xilinx_u50_gen3x16_xdma_5_202210_1 INTERFACE=0 DESIGN=basic` again. Sadly, I got the same error message information as I mention above. (Switching user to super user, I still get the same result).

mariodruiz, 2022-08-26:
Did you resource Vivado after changing the locale?

Someone else had this problem, but I could not reproduce it locally.

crjake, 2022-09-06:
I'm having the same issue. Setup is the same, except my OS is explicitly supported by Vitis (Ubuntu 20.04.1 LTS).

Output of `locale` (was mostly `en_GB` before):
```
LANG=en_US.UTF-8
LANGUAGE=en_GB:en
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```
I can confirm I ran `make distcleanall`, `source /opt/xilinx/xrt/setup.sh`, `source ~/Xilinx/Vivado/2022.1/settings64.sh` and `source ~/Xilinx/Vitis/2022.1/settings64.sh` after changing my locale. Issue still persists unfortunately.

Thanks!

imzongjian, 2022-09-23:
> I'm having the same issue. Setup is the same, except my OS is explicitly supported by Vitis (Ubuntu 20.04.1 LTS).
>
> Output of `locale` (was mostly `en_GB` before):
>
> ```
> LANG=en_US.UTF-8 [...]
Hi, have you solved this problem?
I met the same thing and also at a 2022.1 IDE. My OS is CentOS.  @yangxiaomaomao @crjake

closed 2024-08-01 by mariodruiz