Xilinx/xup_vitis_network_example #87: Payload size limitation [XRT]

KoalaYuFeng, 2022-09-14:
I am using the host_xrt branch for UDP data transmission. I notice that in  alveo_vnx_krnl_mm2s.cpp file, when size in AlveoVnxKrnlMM2S::transferDataToKrnl excceed 65532, it will exit, return 1. So the largest payload size in XRT version is 65532? 

What causes this limitation? In my perspective, xrt::bo() function can allocate a larger memory size than 65532.

Can @mariodruiz  please have a look?
Thanks for your time.

mariodruiz, 2022-09-15:
Hi @KoalaYuFeng,

I have not worked in that driver. 

@fpgafais any idea?

mariodruiz, 2022-09-15:
@KoalaYuFeng, did you try without the exit condition? If so, what was the outcome?

KoalaYuFeng, 2022-09-15:
> @KoalaYuFeng, did you try without the exit condition? If so, what was the outcome?

I tried this without exit condition, the AlveoVnxKrnlMM2S::sendPacket() function kept running, did not return 0.

fpgafais, 2022-09-15:
@KoalaYuFeng with UDP packets we typically aim to not exceed the maximum packet size which is 65536 per single transfer. Your payload data could have been fragmented into Ethernet frames by lower layers and transmitted out, but please check if you can reconstruct the complete payload on the receiving side

mariodruiz, 2022-09-15:
The mm2s kernel is chunking the data to 1,408 Bytes if the user data is larger than that. 

https://github.com/Xilinx/xup_vitis_network_example/blob/c16cba6598099f87e203659d7f8d6a536d97ef86/Basic_kernels/src/krnl_mm2s.cpp#L78

So, there should be no limitation on the host code. Unless the mm2s kernel has changed.

fpgafais, 2022-09-15:
@mariodruiz right, but 1.4k frames originating from the same transfer should be tagged with UDP offset to be reconstructed into a single packet by the receiving side. The size field in UDP headers is 4 hex so one can build offsets up to 65535. Further fragmentation of larger payloads should be handled on another, upper layer

mariodruiz, 2022-09-15:
That's right. Thank you!

KoalaYuFeng, 2022-09-16:
> @KoalaYuFeng, did you try without the exit condition? If so, what was the outcome?

I just try without exit and meanwhile reconstruct the receive code, it works in a right condition. now it can support larger payload size.  Thanks for your time and help.

KoalaYuFeng, 2022-09-16:
Thanks for your time and help.

closed 2022-09-16 by KoalaYuFeng