Xilinx/xup_vitis_network_example #90: Alveo U50 stat_rx_status always 192

victornvq2, 2022-10-04:
Hi All, 
Completed build project on Alveo U50, however when on run ./ping_fpga  it always returns 192 of cmac stat_rx_status, no matter how many time I try to test,  thus the link never become ready, thus the local ping failed  !
Please advise 

Whishing you guys having a good day !
Victor

victornvq2, 2022-10-10:
The test tool is ping_fpga that was compiled from xrt_host_api directory. It seems that cmac uplus uplus IP is not ready yet. Any hint would be appreciated !

mariodruiz, 2022-10-17:
Hi @victornvq2,

You may need to provide more information.
What is your setup? Where did you connected the QSFP28 port to?

Mario

victornvq2, 2022-10-17:
Hi Mario, 

We have 2 development PC that have Alveo U50 on each of them. 2 PCs are connected via a Cisco QSFP+ DAC cable
1. We've build open-nic-shell and open-nic-driver on each PC, testing network card on Alveo U50. Everything works fine, we are to  ping, send + receive TCP and UDP packets at rate 15 Gbps
2. Now on one  PC, we rebuild xup_vitis_networking by using Vitis 2022.1 with modifying scripts for Alveo U50. the Basic design built went fine with xclbin were generated successfully
3. We have built the xrt_host_api client then run ./ping_fpga <basic.xclbin file>  but it was unsuccessful to ping the localhost
While trying to find the problem, we found that start_rx_status always 192, it never goes down to 3
When I try to modify to ping another PC also unsuccessfully

During the testing we always kept the connection between 2 PCs with Cisco QSFP+ DAC cable that 's working with Open NIC build.
 
I guest there's could  be some different settings between Vivado flow and Vitis kernel flow, but don't know how extractly

Please advise !

Victor

mariodruiz, 2022-10-17:
Hi Victor,

As you are using the switch, you need to read this part of the FAQ https://github.com/Xilinx/xup_vitis_network_example#alveo-card-connected-to-a-100g-capable-switchnic-is-not-getting-link 

FEC and Auto negotiation aren't enable in VNx, so the switch has to have these capabilities disabled as well. If you connect the two Alveo directly, you will get link. This happens all the time, hence it is included in the FAQ. Previous case here https://github.com/Xilinx/xup_vitis_network_example/issues/32#issuecomment-926483193

Mario

victornvq2, 2022-10-18:
Mario, thank you.  I will try that
Best,

mariodruiz, 2023-02-27:
Hi @victornvq2, as part of a community contribution the RS-FEC is enabled by default in the kernel, but not by software. This should make it easier to connect to other network equipment.

Please, check the latest commit and associated documentation

victornvq2, 2023-03-02:
hi Mario, thank I will be trying the lastest commit
Thanks,