Xilinx/xup_vitis_network_example #92: Where is the definition of the "_start()" in the DataMover class?

Ossic, 2022-10-25:
Hi, I would like to ask just one questions for vnx_utils.py.
Where is the definition of the function _start() referred in the class DataMover of vnx_utils.py. Please note just one "_" in front of the "start".
I assume that the function is defined in DefaultIP as the parent class in the overlay.py file. However, in pynq2.7, there are only _start_sw(...) and _start_ert(...) defined which seems to be some kind of connection. 

Intriguingly, vnx-basic.ipynb does work completely.

Wishing you guys having a good day ! 

Run Time
Card:			Alveo U50
OS version: 		Ubuntu 18.04.6 LTS/5.4.0-122-generic
pynq version: 	2.7.0
Git branch: 		master
XRT Version: 	2.11.634
XRT Branch:		2021.1

mariodruiz, 2022-10-25:
Hi @Ossic,

You are right, there is a bug in the code. This driver is working because the `start()` method of the `DefaultIP` is being called.

Mario

closed 2022-10-25 by mariodruiz

Ossic, 2022-10-25:
Hi @mariodruiz 
Thanks for your reply! It still confuses me that what makes python interpreter bypass the "start()" function in DataMover class and thus makes the "start()" function in DefaultIP work priorly. Please forgive my poor grammar skills about python.

mariodruiz, 2022-10-25:
I think the problem is that the `.start` method is assigned in the `__init__` function of `DefaultIP` this could potentially skip the redefinition of the `start` method in the derived class.

commit 87d5f223 (2022-10-25): Bind  to the proper parent's methods. Fix #92
commit 095b12b8 (2022-10-25): Bind .start() to the proper parent's methods. Fix #92